betaflight/betaflight #11655: Motors go spinning eratically fast just after arming at zero throttle on HGLRC F722 mini

mauriciobridge, 2022-06-14:
### Describe the bug

On BF4.3 defaults, after arming motors go spinning like crazy (no props), if I let the quad armed for more than 2 seconds motors may burn!  Going back to BF4.2.11 solves the problem and motors runs smooth.

### To Reproduce

On flight controller HGLRC F722mini when after flashing BF4.3 motors spin erratically and fast just after arming at zero throttle.

### Expected behavior

expect the motors to run smooth and cool

### Flight controller configuration

```
# version
# Betaflight / STM32F7X2 (S7X2) 4.3.0 Jun 14 2022 / 00:50:37 (229ac66) MSP API: 1.44
# config: manufacturer_id: HGLR, board_name: HGLRCF722, version: 836acaf9, date: 2021-10-31T19:48:38Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name HGLRCF722
manufacturer_id HGLR
mcu_id 001900064e53500e2030304e
signature 

# name: BRIDGE

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature TELEMETRY
feature OSD

# beacon
beacon RX_LOST
beacon RX_SET

# aux
aux 0 0 0 1800 2100 0 0
aux 1 13 1 1300 2100 0 0
aux 2 35 2 1800 2100 0 0
aux 3 36 3 1500 2100 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 14 20 26 36
vtxtable powerlabels 25 100 400 MAX

# master
set dyn_notch_count = 1
set dyn_notch_q = 500
set acc_calibration = 44,35,118,1
set mag_hardware = NONE
set baro_hardware = NONE
set serialrx_provider = CRSF
set min_throttle = 1070
set dshot_bidir = ON
set use_unsynced_pwm = OFF
set motor_pwm_protocol = DSHOT300
set motor_pwm_rate = 480
set yaw_motors_reversed = ON
set small_angle = 180
set pid_process_denom = 4
set osd_warn_batt_not_full = OFF
set osd_warn_batt_warning = OFF
set osd_warn_batt_critical = OFF
set osd_warn_visual_beeper = OFF
set osd_warn_core_temp = OFF
set osd_warn_rc_smoothing = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_vbat_pos = 14391
set osd_rssi_pos = 88
set osd_link_quality_pos = 2455
set osd_tim_2_pos = 2433
set osd_remaining_time_estimate_pos = 4472
set osd_throttle_pos = 2401
set osd_vtx_channel_pos = 53
set osd_current_pos = 4183
set osd_mah_drawn_pos = 4152
set osd_motor_diag_pos = 4449
set osd_craft_name_pos = 14348
set osd_power_pos = 4216
set osd_warnings_pos = 14601
set osd_avg_cell_voltage_pos = 4097
set osd_battery_usage_pos = 4113
set osd_esc_tmp_pos = 378
set osd_esc_rpm_pos = 160
set osd_esc_rpm_freq_pos = 184
set osd_flip_arrow_pos = 14574
set osd_core_temp_pos = 377
set osd_stat_rtc_date_time = ON
set osd_stat_max_spd = OFF
set osd_stat_min_batt = OFF
set osd_stat_battery = ON
set osd_stat_bbox = OFF
set osd_stat_max_g_force = ON
set osd_stat_max_esc_rpm = ON
set osd_stat_min_link_quality = ON
set osd_stat_max_fft = ON
set vtx_band = 5
set vtx_channel = 2
set vtx_power = 2
set vtx_freq = 5695
set vcd_video_system = NTSC
set name = BRIDGE

profile 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set thr_mid = 35
set thr_expo = 20
set rates_type = BETAFLIGHT
set roll_rc_rate = 85
set pitch_rc_rate = 85
set yaw_rc_rate = 85
set roll_expo = 15
set pitch_expo = 15
set yaw_expo = 15
set roll_srate = 64
set pitch_srate = 62
set yaw_srate = 59
set tpa_rate = 70
set tpa_breakpoint = 1250

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
# 
```
```
resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: FREE
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: LED 2
A14: LED 1
A15: FREE
B00: MOTOR 3
B01: MOTOR 4
B02: GYRO_CS 1
B03: FREE
B04: MOTOR 1
B05: MOTOR 2
B06: FREE
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: FREE
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: PINIO 1
C09: PINIO 2
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: BEEPER
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FLASH_CS
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1 : DSHOT_BITBANG 2
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# 
```

### Flight controller

HGLRC F722 mini, have been using it for over a year on BF4.2 

### Other components

FC: HGLRC F722 mini 20x20
ESC: HGLRC Seuz 45a 20x20
VTX: TBS Sixty9
RX: Crossfire Sixty9
Motors: FlyWoo NIN 2207.5

### How are the different components wired up

sdf

### Add any other context about the problem that you think might be relevant here

What i already have tried to solve the problem (with no succes):
- turn off and on bidirectional dshot
- gyro filter multiplier slider set at "0" and "0.5"
- set Dshot300 and Dshot600
- set pidloop fq to 2K 4k and 8k
- dynamic idle is set to 0

haslinghuis, 2022-06-14:
No problem with SpeedyBee F7 mini stack and TBS69.

Please report TBS firmware used and try updating to latest version.
Please report ESC firmware used.

mauriciobridge, 2022-06-14:
TBS 69 on latest V6.17
Problem doesnt appear on BF 4.2.11
Problem appear just on this Quad.

BlackBox Logs on BF 4.2 and 4.3 (log is just arming motors)
https://drive.google.com/drive/folders/1smLo4mGqOtLR1udiseT8-uGeerFMpRUZ?usp=sharing

mauriciobridge, 2022-06-14:
ESC tried BlHeli32 32.9 , 32.8 and 32.7 versions... problem still happens

mauriciobridge, 2022-06-14:
just tried setting PID values the same on BF 4.2 and 4.3, result is the same:
works smooth on BF4.2 but erratically on BF4.3

McGiverGim, 2022-06-15:
In the Blackbox logs I can see the gyro moving like crazy under Betaflight 4.3. I suppose you are in the bench, so the craft is not moving at all.
If that is the case, something in the new version makes the gyro values crazy.

McGiverGim, 2022-06-15:
@mauriciobridge can you check the gyro values without arming the motors? to see if they go crazy only after arming or in the bench too.

McGiverGim, 2022-06-15:
Yes, I simply wanted to discard the gyro as origin of problem. The gyro is ok before arming?

mauriciobridge, 2022-06-15:
Yeah, gyro loks good to me under the motors tab
[image]

mauriciobridge, 2022-06-15:
problem does get solved after disabling Airmode

pbaltha, 2022-06-22:
I'm having the same issue with two different Quads. One is F722 Mini with a 45A ESC also from HGLRC and another one is SpeedyBee F7 V2 with a 50A Skystars ESC, both using 16.71 BLHELI_S. The things I've tried was to use all default, with and without RPM Filtering. The quad using SpeedyBee F7 even burned a motor RCInpower 2306. I run both on 4S batteries and had no issue until the RC6 and RC7 and the Final Release. The only way to the quad behave normal is to turn off airmode always on on configuration Tab. One thing that occurs to me is that Betaflight maybe is confusing Barometer or Acelerometer settings with the gyro or something like that making the airmode to come in erraticaly and spinning the motors like crazy. 
People that say that this is a expected behavior on the bench, is wrong, yes you should get a little spin from the motors on the bench based on the vibration, but not to the point this issues brings ( burning a motor ). Hope to have a fix on this. 

Tried every single PID Loop,Tune,Filter ( on and off ), Bluejay, BLHELI_S 16.71 ( coming back to 16.70 either ) none of this helped, just turning off AIRMODE stopped this issue. 

@mauriciobridge I think I have the exactly same issue, and the reason I suspect on the barometer ( maybe trying to deactivate it ) is that those two quads are the only ones I have with a baro built in the FC.

mauriciobridge, 2022-06-22:
I'm usung Blheli32, also tryied with different versions, but same result.  When this happened to me first time, was also very close to burn a motor... I know what is the normal behavior of the quad onbtje bench with no props caused by airmode.  And again.. ut works fine on BF 4.2.11 also on default.

iforce2d, 2022-07-26:
I'm having this problem too (MAMBAF405US). With BF 4.2 the quad flies great. With BF 4.3 it acts like the throttle value has a small amount of random noise added to it. Just arming and sitting at zero throttle the motor speed will surge erratically, although fortunately not enough to lift off. When I disabled airmode as suggested above the problem went away, but now the response feels a bit sloppy and not locked in.

iforce2d, 2022-07-26:
Actually in angle mode, the throttle surges are actually enough to lift the quad off the ground sometimes. 
https://youtu.be/MfQU-tWlb1I
I have not done any tuning other than to set the PID loop frequency to 4kHz as suggested in the release notes, but with 8k and 2k I get the same results. I tried using ESC protocols other than DSHOT600, and tried setting Throttle Boost to zero, both had no effect. So far only disabling airmode seems to make any difference.

pbaltha, 2022-07-26:
@iforce2d I've start all over with my quad that was presenting this sympton. One thing that could have caused the error was a miss calibrated accelerometer and it was not right mounted on the soft-mounts ( the rubbers ). I really can't say if that was the cause, but after a full rebuild ( I dont mean soldering ). But simpling putting, and make some arrangements on the build, taking really good care of the soft mounts and the positioning, full erase and flash the 4.3.1 BF, it's stopped these symptons even with airmode on. Check for this kind of stuff on your build. MAYBE could be the cause, gyros are very, very senstitive.....Have a good one!

mauriciobridge, 2022-07-26:
Still having issues on BF 4.3.1 Stock settings. so im back to BF 4.2 Stock settings, the quad flies good here. 
I'm pointing that my fc has a hardware problem, like a small problem on the Gyro, that BF4,2 can handle better than BF4.3.

pbaltha, 2022-07-26:
@mauriciobridge Try what I said. Try to reposition every component. Maybe trying to put the FC upside down, taking care no wires touchs the gyro. IN my case a full rebuild solved the problem.

mauriciobridge, 2022-07-26:
Mu point is, there is something that BF4.2 can handle here, where BF4.3 can not.  For sure I will end up switching FC for other reasons too.

d00lar, 2022-08-10:
try 
```
set dterm_lpf1_dyn_min_hz = 80
set dterm_lpf1_dyn_max_hz = 110
set dterm_lpf1_dyn_expo = 8
set dterm_lpf1_type = BIQUAD
set dterm_lpf1_static_hz = 0
set dterm_lpf2_static_hz = 0
```

without this kinf of filter my 4.3 also after arm spins as crazy and flyaway itself.

iforce2d, 2022-08-14:
@d00lar thanks for the suggestion, unfortunately it makes no difference for me. I will just use BF 4.2 which was working great, I can't even remember why I wanted to try 4.3 anyway.

closed 2022-09-21 by github-actions[bot]