betaflight/betaflight #11657: BF 4.3 Crazeebee F4 SX1280 SPI ELRS - RCLink unstable

JimB40, 2022-06-15:
### Describe the bug

On battery plug RC link (LQ) is dropping to 0 and back to 100 every 2 seconds until first arm. When first armed RC link becomes stable

https://youtu.be/8wzVxHn2W24

### To Reproduce

Findings

We did debug session with ELRS support guys. Equalized ELRS firmwares & setups
On my both FC bug shows while on two of deadbyte it doesn't.

I have flashed all 4.3 Release Candidates applying included in issue minimal diff
Problem starts with BF 4.3 RC7. Up to RC6 RC link is solid and does not drops
Once FC is armed link is solid too.

### Expected behavior

RCLink should be solid

### Flight controller configuration

# 

# diff all

# version
# Betaflight / CRAZYBEEF4SX1280 (HAMO) 4.3.0 Dec 24 2021 / 17:13:20 (norevision) MSP API: 1.44

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name CRAZYBEEF4SX1280
mcu_id 0053003c3435510636393439
signature 

# feature
feature -AIRMODE
feature MOTOR_STOP

# serial
serial 1 2048 115200 57600 0 115200

# aux
aux 0 0 0 1700 2100 0 0

# master
set acc_calibration = -15,-7,-130,1
set motor_pwm_protocol = DSHOT300
set yaw_motors_reversed = ON
set small_angle = 180
set osd_link_quality_pos = 2443
set vtx_band = 5
set vtx_channel = 8
set vtx_power = 1
set expresslrs_uid = 139,4,68,227,54,146
set expresslrs_switch_mode = WIDE

profile 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
# 

### Flight controller

CRAZYBEEF4SX1280

### Other components

Setup
FC -> CrazeebeeSX1280,  BF4.3 Final Release, 
TX module -> HM ES24TX Pro, ELRS  2.5.0 (06f1e9)
VTX -> on board
RX -> on board

driedger, 2022-06-15:
I am also experiencing the same issue on my setup:
### Flight controller
DarwinFPV ELRS AIO - target BETAFPVF4SX1280

### Other components
### Setup
TX module -> HM ES24TX, ELRS 2.0.0 (5c4808)
VTX -> Happymodel OVX303
RX -> elrs spi

### Add any other context about the problem that you think might be relevant here
Went through most of the same testing as @JimB40, but was also troubleshooting a problem with vtx not able to switch off R8 and RCE power level. 
I had forgotten that I had vtx administrator setup to set channel and power level. I turned off elrs vtx administrator, and this RX LOSS stopped happening.
Digging in further, it doesnt occur when only setting the Pwr level. When I set the BAND in vtx administrator, it starts the 4 second loops of rx loss.
I spoke with Deadbyte in the ELRS discord and he mentioned it may be MSP related.

RipperGOT, 2022-06-15:
I see in the video you are using 500Hz Packet Air Rate,, no? Try 150Hz. 
I don't have that problem with 150Hz

JimB40, 2022-06-16:
> I had forgotten that I had vtx administrator setup to set channel and power level. I turned off elrs vtx administrator, and this RX LOSS stopped happening.
That was the clue @driedger. 
I've set Band to Off in VTX Administrator and RX LOSS is is not happening.
So bug is probably about TX re-sending constantly MSP Command. Probably EEPROM WRITE that resets RC link for a moment 

> I see in the video you are using 500Hz Packet Air Rate,, no? Try 150Hz.
Tested with different speeds. This bug is speed independent.

JimB40, 2022-06-16:
BTW guys. EEPROM is rated 10000-100000 writes. So sending WRITE command constantly you kill FC slowly.

AlessandroAU, 2022-06-17:
Looks like a reboot loop caused by changing VTX settings, possibly caused by this PR? https://github.com/betaflight/betaflight/pull/11597

Aeromarine, 2022-06-27:
Yes I'm having exactly the same problem on my brand new Crazeebee F4 SX1280 board as well.... same as this video described: https://youtu.be/8wzVxHn2W24 from the @JimB40

cregjohnstone87, 2022-07-02:
Also having same problem on 4.3.0 release.

We have a few people testing mspvtx version From PR which is also based on 4.3.0 release.

Those guys with F4SX1280 all experiencing issues until first arm

SteveCEvans, 2022-07-02:
The intent was that such issues be fixed by https://github.com/betaflight/betaflight/pull/11597 as this will now only do an EEPROM (actually FLASH) write if the config has changed. Is the VTX administrator writing the same data each time or does it vary?

zwilmes, 2022-08-13:
BF4.3.1 on Happymodel F4SX1280 hdzero quad. Can confirm the issue, it seemed to work ok for a while but was getting random failsafes. Then it progressively got worse until it was twitching all over the place as I was flying and then would eventually failsafe. Turning off the band setting on vtx admin did help in my case.

SteadyeddieFPV, 2022-09-29:
I am having the same issue on both the above listed Happymodel board and 2 BETAFPVF4SX1280 boards running BF 4.3.1.  all 3 running HDZero.  turning off the band in the VTX administrator solves the problem.

Prajwal-Koirala, 2023-12-26:
This is an article about a bug in Betaflight 4.3 that causes the RC link to drop. It discusses the bug being present on the Crazeebee F4 SX1280 flight controller and how it can be fixed by turning off the band setting in VTX Administrator. The bug is likely caused by the VTX administrator sending too many MSP write requests, which can damage the EEPROM. The issue was closed on GitHub on August 9, 2022, but was reopened on August 10, 2022.

Let me know if you would like to learn more about this bug or Betaflight in general.

haslinghuis, 2024-05-12:
> I am having the same issue on both the above listed Happymodel board and 2 BETAFPVF4SX1280 boards running BF 4.3.1. all 3 running HDZero. turning off the band in the VTX administrator solves the problem.

Seems to be fixed

closed 2024-05-12 by haslinghuis