betaflight/betaflight #11673: yaw axis becoming unresponsive on the BetaFPV F4 1S ELRS board. Please help..

Aeromarine, 2022-06-18:
### Describe the bug

Hello, I have tried so many different ways to re-flash the old firmware (from RC5 to RC7 to the official 4.3), clean install and re-install different version of the configurtor as well, got no luck at all. My meteor65 was working just fine until I updated to the v4.3.

My gyro X & Y working just fine, there is no YAW moment at all. I’ve been searching around google for some answers for the passed week, seems some of other people at www.racedayquads.com are having the same issue as well: https://imgur.com/C9eKlpa

Here is the video link about no Yaw axis on my whoop: https://youtu.be/zXzezRadho0

Is my FC dead? Please advice.. 

### To Reproduce

yaw axis becoming unresponsive on the BetaFPV F4 1S ELRS board after flash to v4.3

### Expected behavior

should be no problem at all, time to enjoy the new v4.3

### Flight controller configuration

```
 # diff

# version
# Betaflight / BETAFPVF4SX1280 (BHER) 4.3.0 Jun 14 2022 / 00:30:19 (229ac66) MSP API: 1.44

# start the command batch
batch start

board_name BETAFPVF4SX1280
manufacturer_id BEFH

# name: Meteor65-ELRS

# feature
feature -LED_STRIP
feature -AIRMODE
feature -ANTI_GRAVITY

# serial
serial 0 1 115200 57600 0 115200
serial 30 2048 115200 57600 0 115200

# aux
aux 0 0 0 1900 2100 0 0
aux 1 1 1 900 1100 0 0
aux 2 2 1 1400 1600 0 0
aux 3 28 3 1900 2100 0 0
aux 4 35 2 1900 2100 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 0 1 2 3
vtxtable powerlabels 25 100 200 350

# master
set dyn_notch_count = 1
set dyn_notch_q = 500
set acc_calibration = 50,-49,88,1
set max_check = 2000
set dshot_idle_value = 900
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set motor_poles = 12
set vbat_max_cell_voltage = 440
set vbat_min_cell_voltage = 300
set current_meter = NONE
set ibata_scale = 179
set small_angle = 180
set pid_process_denom = 4
set osd_warn_core_temp = OFF
set osd_warn_rc_smoothing = OFF
set osd_warn_fail_safe = OFF
set osd_warn_launch_control = OFF
set osd_warn_no_gps_rescue = OFF
set osd_vbat_pos = 2433
set osd_rssi_pos = 2241
set osd_link_quality_pos = 2273
set osd_rssi_dbm_pos = 2305
set osd_tim_1_pos = 386
set osd_tim_2_pos = 2455
set osd_remaining_time_estimate_pos = 33
set osd_flymode_pos = 2423
set osd_throttle_pos = 2401
set osd_vtx_channel_pos = 2080
set osd_crosshairs_pos = 2253
set osd_current_pos = 343
set osd_craft_name_pos = 2441
set osd_home_dist_pos = 98
set osd_flight_dist_pos = 130
set osd_warnings_pos = 2378
set osd_avg_cell_voltage_pos = 289
set osd_disarmed_pos = 2315
set osd_esc_tmp_pos = 163
set osd_esc_rpm_pos = 150
set osd_core_temp_pos = 278
set vtx_band = 5
set vtx_channel = 8
set vtx_power = 3
set vtx_low_power_disarm = ON
set vtx_freq = 5917
set vcd_video_system = NTSC
set expresslrs_uid = 78,177,59,2,118,241
set expresslrs_rate_index = 1
set name = Meteor65-ELRS

profile 0

rateprofile 0

# rateprofile 0
set rates_type = BETAFLIGHT
set roll_rc_rate = 100
set pitch_rc_rate = 100
set yaw_rc_rate = 100
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70

# end the command batch
batch end
```
```
  # resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: MOTOR 2
A01: ADC_BATT
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: SERIAL_TX 11
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: RX_SPI_EXPRESSLRS_BUSY
A14: BEEPER
A15: RX_SPI_CS
B00: FREE
B01: FREE
B02: RX_SPI_BIND
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: GYRO_EXTI
B07: MOTOR 4
B08: MOTOR 1
B09: RX_SPI_EXPRESSLRS_RESET
B10: MOTOR 3
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: RX_SPI_EXTI
C14: LED 1
C15: LED
D02: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1 : SERIAL_TX 11
TIM2:
    CH1 : MOTOR 2
    CH3 : MOTOR 3
TIM3: FREE
TIM4:
    CH2 : MOTOR 4
    CH3 : MOTOR 1
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: MOTOR 3
DMA1 Stream 2: FREE
DMA1 Stream 3: MOTOR 4
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: MOTOR 2
DMA1 Stream 6: FREE
DMA1 Stream 7: MOTOR 1
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

```

### Flight controller

BetaFPV F4 1S 5A Whoop Flight Controller - 2.4GHz ELRS

haslinghuis, 2022-06-18:
Please also provide `status` output in cli

Aeromarine, 2022-06-18:
# status
MCU F411 Clock=108MHz (PLLP-HSE), Vref=3.31V, Core temp=30degC
Stack size: 2048, Stack address: 0x2001fff0
Configuration: CONFIGURED, size: 3777, max available: 16384
Devices detected: SPI:1
Gyros detected: gyro 1 locked dma
GYRO=ICM20689, ACC=ICM20689
OSD: MAX7456
System Uptime: 14 seconds, Current Time: 2022-06-18T22:41:18.301+00:00
CPU:28%, cycle time: 125, GYRO rate: 8000, RX rate: 15, System rate: 9
Voltage: 10 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
Arming disable flags: RXLOSS CLI MSP RPMFILTER

haslinghuis, 2022-06-18:
Curious here. Does it work again if going back to 4.2.11?

Also show gyroregisters output

Aeromarine, 2022-06-18:
# gyroregisters

# Gyro 1
# WHO_AM_I    0x98
# CONFIG      0x0
# GYRO_CONFIG 0x18

Aeromarine, 2022-06-18:
is ELRS whoop board, I have to use 4.3...

haslinghuis, 2022-06-18:
Yes you might have been affected with a bad batch. We are still investigating with TDK if it's a hardware of configuration error.
FYI: https://github.com/betaflight/betaflight/pull/11584

Aeromarine, 2022-06-18:
yes, I have read the https://github.com/betaflight/betaflight/pull/11584 as well.. I am hoping for the fix from the dev... otherwise I will just have to learn to fly ARCO..

haslinghuis, 2022-06-18:
Recommend using flight simulator to learn acro for about one or two months building muscle memory.

SteveCEvans, 2022-06-18:
@Aeromarine  it’s the gyro, not the acc that has the issue, so acro mode doesn’t help. I’m pressing TDK for a fix, which I know is possible as I was able to recover my gyro with some prodding of undocumented registers.

Aeromarine, 2022-06-18:
@SteveCEvans ohhh, that’s a good news! I am hoping for a fix! Is a almost brand new meteor65 whoop… I don’t want it just die this way…

closed 2022-07-27 by github-actions[bot]

nikybiasion, 2022-10-05:
There are news about bricked gyro?