betaflight/betaflight #11677: Use bidirectional DSHOT telemetry repeated values to codify telemetry data

damosvil, 2022-06-19:
### Is your feature request related to a problem? Please describe

At this time there are more than 7 * 256 bidirectional DSHOT codificable telemetry values repeated, that aren't being used. Please check attached spreadsheet.

[DSHOT repeated codification values.ods](https://github.com/betaflight/betaflight/files/8935905/DSHOT.repeated.codification.values.ods)

### Describe the solution you'd like

At this time there are 7 ranges of 256 values repeated. I propose to make an extension with:
- One range to codify temperature, voltage, and all that fits into it
- Two ranges for debugging
- Leave 4 ranges for a future extension

### Describe alternatives you've considered

I made a previous proposal using 0x0FFF telemetry value as a escape telemetry value, but it implied much more logic in dshot. This proposal only affects to bbUpdateStart function, where dshot response telemetry value can easily used to update the corresponding telemetry data value.

### Other information

This will need to be implemented as a bidir dshot extension.

This work would lead to apply throttle limit depending on the esc temperature. At this time the implementations do this in the esc, but this is not the correct place to do it, because one of the escs can lose power while the others are ok, rendering betaflight unable to control the kwad.

Depending on the esc layout a temperature correction value should be added.

At this time I am improving a version of Bluejay where I am improving telemetry data. Now I am working in temperature control where I am limiting the throttle, but as I said this is not the optimal place to do. Instead, it is better to do it in Betaflight.

I will have extended telemetry versions of Bluejay and Betaflight in a few weeks.

damosvil, 2022-06-20:
Updated spreadsheet with the correct binary values at eee dshot telemetry coding field

damosvil, 2022-06-23:
The ongoing work is being carried on the following two repos:
https://github.com/damosvil/bluejay/tree/feature/dshot_extended_telemetry
https://github.com/damosvil/betaflight/tree/feature/dshot_extended_telemetry_on_unused_ranges

At this moment this is alpha code, but it helps to have an idea on how to extend DSHOT telemetry. Changes are compatible with previous versions of Betaflight.

damosvil, 2022-06-25:
Draft pull request created:
https://github.com/betaflight/betaflight/pull/11694

damosvil, 2022-06-26:
Flight footage with additional temperature telemetry working:
https://youtu.be/CxIo7ElG3JA

haslinghuis, 2022-10-07:
Fixed in #11694

closed 2022-10-07 by haslinghuis