betaflight/betaflight #11685: shutdown quad at a minimum voltage

jawz101, 2022-06-21:
### Is your feature request related to a problem? Please describe

I am a new flyer and getting used to lipo battery power doesn't seem to difficult but it would be nice if my quad would just fall out of the sky when the voltage is so low to cause battery damage.  Sometimes I can simply be flying a tiny whoop indoors line of sight or letting someone else fly line of sight and not see where the voltage stands.

### Describe the solution you'd like

An option to kill everything if the voltage falls below some defined value.

### Describe alternatives you've considered

Always fly with goggles and monitor voltage.

jawz101, 2022-06-21:
It's a tinyhawk 3.  I haven't glued anything and the remote doesnt come with alarms

jawz101, 2022-06-23:
thank you for the information.  Closing

closed 2022-06-23 by jawz101

etracer65, 2022-06-23:
@jawz101  

> The forcefulness of a hard stop on a tiny whoop sounds great. Like in a home where there's little chance of losing it.

You may think that simply disarming is a good solution, but it's not what you really want. The reality is that battery voltage sags significantly as the battery drains. So even a mild throttle raise can cause the voltage to drop momentarily below the cutoff voltage. The result is that you'd have many unexpected and undesirable disarms and it'd be much more likely to damage your quad. The disarm by voltage has been tried and doesn't work out and that's why the current system is in place. The gradual loss of throttle is better to allow you to maintain control and quickly realize that you're battery is done. If you want it to behave similarly to a disarm, then set the value to 0. Within a couple of seconds of hitting the voltage cutoff you'll be on the ground and unable to fly.

jawz101, 2022-06-23:
Thank you.  I guess I'll try 75% out

etracer65, 2022-06-23:
50% or lower would be better. Setting it to 75 will mean it will only reduce your throttle 25%. You may not feel that right away.