betaflight/betaflight #11690: Zeez F7 write blackbox with gaps on BF 4.3

mituritsyn, 2022-06-23:
### Describe the bug

The log files are written with gaps. at the same time the quad with elrs has micro freezes in flight just like rx_loss. Initially i thought i had a problem with the flight controller and replaced it with hobbiwing, but recently i was sent another quadcopter with a zeez flight controller for tuning and it has the same problem.
[Downloads_1.zip](https://github.com/betaflight/betaflight/files/8966674/Downloads_1.zip)

[image]
[image]

### To Reproduce

Flash zeez f7 fc with BF 4.3 and write flight log

### Expected behavior

Flight log without gaps. No freezes duaring flight

### Flight controller configuration

```
  # version
# Betaflight / STM32F7X2 (S7X2) 4.3.0 Jun 14 2022 / 00:50:37 (229ac66) MSP API: 1.44
# config: manufacturer_id: ZEEZ, board_name: ZEEZF7, version: 309c0f10, date: 2021-06-14T21:40:58Z

# start the command batch
batch start

board_name ZEEZF7
manufacturer_id ZEEZ

# feature
feature GPS
feature TELEMETRY

# serial
serial 4 2 115200 57600 0 115200

# led
led 0 5,8::CO:2
led 1 6,8::CO:2
led 2 7,8::CO:2
led 3 8,8::CO:2
led 4 9,8::CO:2
led 5 10,8::CO:2
led 6 11,8::CO:2
led 7 0,0::C:0
led 8 0,0::C:0
led 9 0,0::C:0
led 10 0,0::C:0

# aux
aux 0 0 0 1750 2100 0 0
aux 1 46 2 1775 2100 0 0
aux 2 13 1 1300 1700 0 0
aux 3 15 2 1325 1700 0 0
aux 4 35 1 1750 2100 0 0
aux 5 40 0 900 2100 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 14 20 27 36
vtxtable powerlabels 25 200 500 800

# master
set dyn_notch_count = 2
set dyn_notch_q = 500
set acc_calibration = -170,-38,152,1
set max_check = 2000
set rc_smoothing_auto_factor = 80
set rc_smoothing_setpoint_cutoff = 25
set rc_smoothing_feedforward_cutoff = 25
set min_throttle = 1070
set dshot_bidir = ON
set use_unsynced_pwm = OFF
set motor_pwm_rate = 480
set failsafe_procedure = GPS-RESCUE
set yaw_motors_reversed = ON
set small_angle = 180
set gps_sbas_mode = AUTO
set gps_auto_baud = ON
set gps_ublox_use_galileo = ON
set gps_rescue_angle = 35
set gps_rescue_descent_dist = 100
set gps_rescue_ground_speed = 2500
set gps_rescue_throttle_max = 1800
set gps_rescue_throttle_hover = 1300
set gps_rescue_min_dth = 50
set gps_rescue_allow_arming_without_fix = ON
set gps_rescue_alt_mode = FIXED_ALT
set ledstrip_race_color = GREEN
set ledstrip_beacon_color = BLUE
set osd_link_quality_pos = 2081
set osd_tim_2_pos = 2519
set osd_throttle_pos = 2508
set osd_gps_speed_pos = 2233
set osd_gps_sats_pos = 2105
set osd_home_dir_pos = 2127
set osd_home_dist_pos = 2093
set osd_avg_cell_voltage_pos = 2497
set osd_stat_max_dist = ON
set osd_stat_min_batt = OFF
set osd_stat_endbatt = ON
set osd_stat_min_rssi = OFF
set osd_stat_bbox = OFF
set osd_stat_min_link_quality = ON
set osd_stat_total_dist = ON
set debug_mode = GYRO_SCALED
set vtx_band = 5
set vtx_channel = 3
set vtx_power = 2
set vtx_freq = 5732
set rpm_filter_q = 450

profile 0

# profile 0
set vbat_sag_compensation = 100
set anti_gravity_gain = 4500
set p_pitch = 56
set i_pitch = 100
set d_pitch = 49
set f_pitch = 142
set p_roll = 53
set i_roll = 95
set d_roll = 43
set f_roll = 136
set p_yaw = 53
set i_yaw = 95
set f_yaw = 136
set d_min_roll = 32
set d_min_pitch = 36
set thrust_linear = 20
set feedforward_smooth_factor = 30
set feedforward_jitter_factor = 10
set dyn_idle_min_rpm = 30
set simplified_master_multiplier = 120
set simplified_d_gain = 90
set simplified_feedforward_gain = 95

rateprofile 0

# rateprofile 0
set thr_mid = 30
set thr_expo = 30
set roll_expo = 20
set pitch_expo = 20
set yaw_expo = 20
set roll_srate = 60
set pitch_srate = 60
set yaw_srate = 60

# end the command batch
batch end
```
```
 # resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: SERIAL_TX 4
A01: SERIAL_RX 4
A02: FREE
A03: FREE
A04: FLASH_CS
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: CAMERA_CONTROL
A09: FREE
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: OSD_CS
B00: LED_STRIP
B01: FREE
B02: BEEPER
B03: FREE
B04: MOTOR 4
B05: SPI_MOSI 3
B06: MOTOR 3
B07: MOTOR 2
B08: MOTOR 1
B09: FREE
B10: SERIAL_TX 3
B11: PINIO 1
B12: GYRO_CS 1
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: ADC_BATT
C03: ADC_CURR
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: GYRO_EXTI
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SERIAL_TX 5
C13: FREE
C14: LED 1
C15: LED 2
D00: FREE
D01: FREE
D02: SERIAL_RX 5
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1 : CAMERA_CONTROL
    CH2N: LED_STRIP
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1 : DSHOT_BITBANG 2
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: ADC
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: LED_STRIP
DMA2 Stream 7: FREE
```

### Flight controller

Zeez F7 2020 v1 and Zeez F7 3030 v1

### Other components

First quad:
HM elrs ep2RX, fw: 2.2
Zeez esc 32.9
ClearTX VTX(SA 2.0)
Second:
Crossfire RX
HW esc 32.9
RUSH tank vtx(SA 2.1)

### Add any other context about the problem that you think might be relevant here

# flash_info
Flash sectors=1024, sectorSize=131072, pagesPerSector=64, pageSize=2048, totalSize=134217728
Paritions:
  0: BBMGMT    1003 1023
  1: FLASHFS   0 1002
FlashFS size=131465216, usedSize=0

# flash_scan
Verifying
Success

Also noticed that on the second quad the rpm filters are letting in more noise than usual - could this be related to the main problem? Could it be scheduler issue?

RX12, 2022-06-23:
I am experiencing the same issue (as in #11458 )

haslinghuis, 2022-06-23:
Might be missing a setting in the ZEEZF7 unified target config file.

Please check and report:
```
get gyro_1_bustype
```

Try
```
set gyro_1_bustype = SPI
```

RX12, 2022-06-23:
> Might be missing a setting in the ZEEZF7 unified target config file.
> 
> Please check and report:
> 
> ```
> get gyro_1_bustype

Bustype is set to GYROAUTO by default

Could this cause the gaps in the log and the twitching in flight as well???

haslinghuis, 2022-06-23:
Probably yes. please also check status output before setting bustype to SPI

RX12, 2022-06-23:
What do I have to check for? 

# status
MCU F722 Clock=216MHz, Vref=3.29V, Core temp=45degC
Stack size: 2048, Stack address: 0x20010000
Config size: 3962, Max available config: 16384
Gyros detected: gyro 1
GYRO=MPU6000, ACC=MPU6000
OSD: MAX7456
System Uptime: 17 seconds, Current Time: 2022-06-23T12:36:07.400+00:00
CPU:3%, cycle time: 125, GYRO rate: 8000, RX rate: 33, System rate: 9
Voltage: 0 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
SD card: Not configured
Arming disable flags: RXLOSS CLI MSP RESCUE_SW RPMFILTER

mituritsyn, 2022-06-23:
```
# status
MCU F722 Clock=216MHz, Vref=3.29V, Core temp=63degC
Stack size: 2048, Stack address: 0x20010000
Configuration: CONFIGURED, size: 4034, max available: 16384
Devices detected: SPI:1, I2C:0
Gyros detected: gyro 1 locked dma
GYRO=MPU6000
OSD: MAX7456
System Uptime: 119 seconds, Current Time: 2022-06-23T12:37:41.318+00:00
CPU:53%, cycle time: 125, GYRO rate: 8000, RX rate: 15, System rate: 9
Voltage: 0 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
SD card: Not configured
Arming disable flags: RXLOSS CLI MSP RPMFILTER

# set gyro_1_bustype
gyro_1_bustype = GYROAUTO
Allowed values: NONE, I2C, SPI, SLAVE, GYROAUTO

# set gyro_1_spibus
gyro_1_spibus = 2
Allowed range: 0 - 4

```

RX12, 2022-06-23:
In 4.2.x it's also set to GYROAUTO and everything is working fine

haslinghuis, 2022-06-24:
As I don't have the board obvious it worked on previous firmware.
Just trying to isolate the problem, Maybe there is bug with the new firmware so please try to set it SPI instead of GYROAUTO to see if it helps.

mituritsyn, 2022-06-24:
> As I don't have the board obvious it worked on previous firmware. Just trying to isolate the problem, Maybe there is bug with the new firmware so please try to set it SPI instead of GYROAUTO to see if it helps.

There will be a race this weekend so I have no opportunity to test now. I will try to get the zeez back on my copter and test next week. Thanks for help.

haslinghuis, 2022-06-27:
Check this firmware build: https://dev.azure.com/Betaflight/Betaflight%20Nightlies/_build/results?buildId=6724&view=artifacts&pathAsName=false&type=publishedArtifacts

https://github.com/betaflight/betaflight/pull/11697

RX12, 2022-06-28:
I would have to use the unified target for the zeez. I'm sry but I 'm not too sure what I all have to set up in this case. It's more than just the resources. right?

haslinghuis, 2022-06-28:
Now that the build is merged you can just flash latest 4.3.1 firmware

RX12, 2022-06-28:
I don't see 4.3.1 in the configurator yet

haslinghuis, 2022-06-28:
Nope you're right, in that case flash as usual using these steps;

- Download the artifact from the link above.
- Detect your board and use `Load Firmware [Local]` to select the firmware.
- This way the config is automatically added.

Also try the `set gyro_1_bustype = SPI` setting.

mituritsyn, 2022-07-01:
zero gaps with #11697 and set gyro_1_bustype = SPI
I apologise, but I was too lazy to test individually
[image]

RX12, 2022-07-01:
> zero gaps with #11697 and set gyro_1_bustype 
Did you feel any difference with filtering after trying this Betaflight build?? 

I tested it 2 days ago and it didn't have any motor stutters. (Haven't logged yet)
But... I used the same PIDs and filtering as in 4.2.9, but on this version, the motors were very close to burning. I had to use an extreme amount of filtering for it to work. Seems strange
Not sure if it's related to 4.3 though

mituritsyn, 2022-07-01:
> I tested it 2 days ago and it didn't have any motor stutters. (Haven't logged yet) But... I used the same PIDs and filtering as in 4.2.9, but on this version, the motors were very close to burning. [...]

you shouldn't use 4.2 filter settings on 4.3 build. at least dynamic notches was completely reworked

RX12, 2022-07-01:
> you shouldn't use 4.2 filter settings on 4.3 build. at least dynamic notches was completely reworked

Sry for not clarifying. Ofcourse I only copied over the dterm and gyro lowpass values to match the quads noise profile and nothing more. I'm going to log and test it first before saying anything further. I only flew 2 packs

RX12, 2022-07-02:
Tested it today again and experienced an issue regarding the logging.
Through the OSD settings I see that it's recording, but after unplugging the files are gone and there's only a few KB big empty logs.

I'm using Blackbox on a switch. Could the issue be related to this?

SteveCEvans, 2022-07-02:
Please try the latest nightly as https://github.com/betaflight/betaflight/pull/11697 will probably help.

RX12, 2022-07-03:
Adding to my last comment, logs are deleted after unplugging when using `blackbox_mode = normal` and setting it on a switch
However it's working fine when `blackbox_mode = always`

closed 2022-07-14 by mituritsyn