betaflight/betaflight #11692: BF 4.3 won't save "set motor_pwm_protocol = DSHOT600" after upgrading from 4.2 Both in the GUI and CLI

cmanley, 2022-06-24:
### Describe the bug

In BF 4.2 I used DSHOT600. 
After upgrading to 4.3, I set "set motor_pwm_protocol = DSHOT600" in the CLI and the response was
motor_pwm_protocol set to DSHOT600
Then I saved, but after rebooting, it was set back to DSHOT300.

Relevent CLI output:
```
# Betaflight / STM32F411 (S411) 4.3.0 Jun 14 2022 / 00:48:04 (229ac66) MSP API: 1.44
# config: manufacturer_id: MTKS, board_name: MATEKF411RX, version: a074838b, date: 2021-06-14T21:36:54Z
# set motor_pwm_protocol = DSHOT600
motor_pwm_protocol set to DSHOT600
# get motor_pwm_protocol
motor_pwm_protocol = DSHOT600
Allowed values: PWM, ONESHOT125, ONESHOT42, MULTISHOT, BRUSHED, DSHOT150, DSHOT300, DSHOT600, PROSHOT1000, DISABLED
Default value: DISABLED
```

### To Reproduce

Do what I did :)

### Expected behavior

Either saving works, or if something went wrong or if there is some kind of setting or CPU load that DSHOT600 is incompatible with, then display that error/reason.

### Flight controller configuration

```
  # REPLACE THIS LINE BY THE OUTPUT OF YOUR `diff`
```
```
  # REPLACE THIS LINE BY THE OUTPUT OF YOUR `resource show all`
```

### Flight controller

MATEKF411RX (TinyHawk 2 Freestyle board)

### Add any other context about the problem that you think might be relevant here

Since BF 4.2 on this board, my ESCs were flashed with JESC and RPM filtering was enabled. After upgrading to BF 4.3 I enabled dynamic idle too as part of a uavtech preset.

cmanley, 2022-06-24:
Anyway thanks for pointing that documentation out.

KarateBrot, 2022-06-24:
I'll ask if we can maybe display a general error message after unsaved changes, thanks.

haslinghuis, 2022-06-24:
This is by design. 

Details: https://github.com/betaflight/betaflight/pull/11222