betaflight/betaflight #11704: GPS RESCUE slow and weird behaviour

simoccc, 2022-06-28:
### Describe the bug

I'm writing here because I'm noticing a weird behaviour in BF 4.2 and 4.3 GPS Rescue mode. I didn't had this problems till BF 4.1 

Tested with 3 quads with Beitan BN880 and Matek m8q-5883 Betaflight 4.2 and 4.3.

My GPS rescue mode is Current alture not maximum .

When Rescue starts ( from Radio switch ) the quad correctly rotates fast to home direction, but it drifts for some seconds and sometimes it looses altitude a lot , I mean  20/30/40 meters, than it raise up but everything seems extremely slow and random. 

Is rescue  in current altitude mode changed from 4.1.x to 4.2/4.3 ? can I compensate changing some parameters in CLI? I'd like the quad to rotate and gain altitude ( initial altitude ?  ) fast , even if I'm not in maximum altitude mode, is it possibile?

here a quick video https://youtu.be/QUgudSAJcdk

### To Reproduce

Enable Rescue mode with a switch on the radio. GPS rescue mode set in Current Altitude Mode

### Expected behavior

Quad rotate in. to the home direction but it waits some seconds doing nothing and loose altitude, after that it starts the rescue but it's not gaining altitude at all.

### Diff configuration

```txt
2022-06-28 @ 23:54:03 -- Running - OS: MacOS, Chrome: 80.0.3987.116, Configurator: 10.7.2

2022-06-28 @ 23:54:03 -- Using cached release information for configurator releases.

2022-06-28 @ 23:54:03 -- You are using an outdated version of the Betaflight Configurator.
Using a newer version of the firmware with an outdated version of Configurator means that changing some settings will result in a corrupted firmware configuration and a non-working craft. Furthermore, some features of the firmware will only be configurable in CLI.
Betaflight Configurator version  is available for download online, please visit this page to download and install the latest version with fixes and improvements.
Please close the configurator window before updating.

2022-06-28 @ 23:54:10 -- Serial port successfully opened with ID: 1

2022-06-28 @ 23:54:10 -- MultiWii API version: 1.43.0

2022-06-28 @ 23:54:10 -- Flight controller info, identifier: BTFL, version: 4.2.11

2022-06-28 @ 23:54:10 -- Running firmware released on: Nov 9 2021 20:29:32

2022-06-28 @ 23:54:10 -- Board: CLRA/CLRACINGF7(STM32F7X2), version: 0

2022-06-28 @ 23:54:10 -- Unique device ID: 0x3800503338510e39393832

2022-06-28 @ 23:54:10 -- Craft name:

2022-06-28 @ 23:54:10 -- Arming Disabled

2022-06-28 @ 23:57:24 -- CLI mode detected

Entering CLI Mode, type 'exit' to return, or 'help'

# 
# Building AutoComplete Cache ... Done!
# 
# diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.2.11 Nov  9 2021 / 20:29:32 (948ba6339) MSP API: 1.43
# config: manufacturer_id: CLRA, board_name: CLRACINGF7, version: c2b8c0d3, date: 2021-06-18T03:42:42Z

# start the command batch
batch start

board_name CLRACINGF7
manufacturer_id CLRA

# feature
feature GPS
feature TELEMETRY

# beeper
beeper -ON_USB

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# serial
serial 1 64 115200 57600 0 115200
serial 4 2 115200 57600 0 115200
serial 5 1 115200 57600 0 115200

# aux
aux 0 0 0 1850 2100 0 0
aux 1 1 3 1400 1600 0 0
aux 2 46 3 1875 2100 0 0
aux 3 13 1 1900 2100 0 0
aux 4 35 2 1875 2100 0 0
aux 5 40 4 1700 2100 0 0

# master
set gyro_to_use = FIRST
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set dyn_notch_min_hz = 80
set dyn_notch_max_hz = 350
set acc_calibration = -75,44,-93,1
set mag_hardware = NONE
set baro_hardware = NONE
set rssi_channel = 10
set rc_smoothing_auto_smoothness = 20
set serialrx_provider = CRSF
set blackbox_p_ratio = 64
set dshot_idle_value = 600
set dshot_bidir = ON
set failsafe_procedure = GPS-RESCUE
set vbat_min_cell_voltage = 300
set vbat_warning_cell_voltage = 335
set vbat_scale = 160
set ibata_scale = 154
set gps_provider = UBLOX
set gps_sbas_mode = EGNOS
set gps_auto_baud = ON
set gps_ublox_use_galileo = ON
set gps_set_home_point_once = ON
set gps_rescue_angle = 35
set gps_rescue_ground_speed = 1800
set gps_rescue_throttle_hover = 1300
set gps_rescue_sanity_checks = RESCUE_SANITY_FS_ONLY
set gps_rescue_allow_arming_without_fix = ON
set gps_rescue_alt_mode = CURRENT_ALT
set deadband = 6
set yaw_deadband = 4

set debug_mode = GYRO_SCALED
set gyro_rpm_notch_min = 80

profile 1

# profile 1
set anti_gravity_gain = 5000
set feedforward_transition = 20
set iterm_relax = RPY
set iterm_relax_cutoff = 9
set throttle_boost = 10
set p_pitch = 77
set i_pitch = 108
set d_pitch = 65
set f_pitch = 114
set p_roll = 71
set i_roll = 102
set d_roll = 60
set f_roll = 108
set p_yaw = 76
set i_yaw = 108
set f_yaw = 108
set d_min_roll = 0
set d_min_pitch = 0

rateprofile 0
```

### Resources configuration

```txt
# 

# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: PINIO 1
A15: OSD_CS
B00: LED 1
B01: FREE
B02: FREE
B03: CAMERA_CONTROL
B04: BEEPER
B05: SPI_MOSI 3
B06: MOTOR 1
B07: MOTOR 2
B08: MOTOR 3
B09: MOTOR 4
B10: FREE
B11: FREE
B12: FLASH_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: SERIAL_TX 6
C07: SERIAL_RX 6
C08: FREE
C09: MCO 2
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SERIAL_TX 5
C13: PREINIT 2
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: SERIAL_RX 5
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH2: CAMERA_CONTROL
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 2
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: FREE
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

#
```

### Flight controller

CLRACING F7 V2.1

### Other components

GPS Beitan BN-880 not using SDA SCL

haslinghuis, 2022-06-28:
We are improving behavior. Please look at the first step we made: https://github.com/betaflight/betaflight/pull/11579.
You can test it flashing the latest development. More improvements are worked on.

simoccc, 2022-06-29:
> We are improving behavior. Please look at the first step we made: #11579.

that seems amazing.
Does the Table "Problems and Solutions" applies to BF 4.2 and 4.3 actual GPS behaviour that could help  a lot to tune actual BF Officiale release in the mean time.

> You can test it flashing the latest development. More improvements are worked on.

I will test it but I cannot give you feedback until the end of this month , I will try my best.

haslinghuis, 2022-06-29:
The PR is part of 4.4 which is ready for 31%. As being an improvement the table seems to address the new code.

simoccc, 2022-06-29:
Ok, so there is no quick fix or improvement in the short term I guess , this is the way 4.2 and 4.3 are made I guess.

haslinghuis, 2022-06-29:
The PR has been flight tested - it would be save to test. (Need latest development build).

QuinoFPV, 2022-06-29:
> Ok, so there is no quick fix or improvement in the short term I guess , this is the way 4.2 and 4.3 are made I guess.

Simoccc did you tried to force SET values for Throttle (Hover value) and Flight mode (Angle) during Stage1 ? This works very well for me with stable altitude.

[image]

aowi7280, 2022-06-30:
@simoccc I notice you have these set pretty aggressively.  
set gps_rescue_angle = 35
set gps_rescue_ground_speed = 1800
I fly a heavy 6" with 4S 2200mah for comparison. I fly in very windy conditions and don't need an angle greater than 33. 32 would probably be fine too. The quad is going to pitch to that 35 degree angle and lose altitude fast. Also that ground speed will cause a steep angle requiring a very high throttle to keep from losing altitude. Even at 1300 ground speed, I lose a little altitude and I have gps_rescue_throttle_max = 1750. I don't see that in your diff so it must be at default 1600, you could raise it.

Another thing I change is:
gps_rescue_alt_buffer = 5
Allowed range: 0 - 100
Default value: 15
This will maintain a closer altitude to where your current altitude is. 4.3.0 is working great for me.

simoccc, 2022-07-01:
@aowi7280 you are right about gps_rescue_ground_speed I will check that out. I usually set Rescue angle same as FPV camera. This rigs I'm testing are quite heavy  7" ( AUW 1.3-1.6Kg ) . I haven't touch throttle_max because I already see high amps draw during rescue ( I use Li-ion mainly ), I will need to check this more too. I tried to raise rescue throttle D gain and it seems a little better in using throttle ( otherwise it just start up/down throttle continuously )

About gps_rescue_alt_buffer ( I'm in Current altitude rescue mode ) , I understand that if you go low it will be more locked to altitude, that is good but what I'm searching is that when I'm in rescue mode the quad should be gaining at least some meters of altitude , now it's not, is it normal? isn't growing  gps_rescue_alt_buffer the only way to achieve this ? 

I cannot find deep info on gps rescue behaviour online, ctzsnooze #11579 description is amazing for example but it's for the new GPS rescue implementation as I understand , maybe there is something that is in common with the old one dunno.

simoccc, 2022-07-03:
@QuinoFPV very nice idea ! So, if I understand correctly you forced stage 1 throttle to a good throttle pinch value and gpsrescue and angle mode to activate together , that could be a way to solve the initial issues I will give it a try ! Tnx

aowi7280, 2022-07-03:
I set throttle to Hold so that no matter what battery I am using, the quad maintains it. This stage lasts 1.5 seconds until Rescue kicks in.

QuinoFPV, 2022-07-04:
> @QuinoFPV very nice idea ! So, if I understand correctly you forced stage 1 throttle to a good throttle punch value and gpsrescue and angle mode to activate together , that could be a way to solve the initial issues I will give it a try ! Tnx

Yes, and that's exactly the suggestion indicated in #11579 (may 2nd 2022), that i tested successfully activating GPS RESCUE switch also at less than 1 meter from ground:

''DO NOT FORGET: the GPS Rescue Hover throttle value MUST be set accurately! The correct value should cause the quad to climb a bit while in level mode. You can test the right value by configuring Failsafe Stage 1 to enable Level mode, and set that value so that the quad climbs a little bit. Also, remember to set the Level 1 throttle value to the same Hover value. If these values are set too low, the quad will drop early in the rescue - possibly into the water!''

QuinoFPV, 2022-07-04:
> I set throttle to Hold so that no matter what battery I am using, the quad maintains it. This stage lasts 1.5 seconds until Rescue kicks in.

Interesting but if you are zero throttle or low throttle during a descent you probably continue to go down, i suppose angle mode and ('almost') hovering can be better

closed 2022-08-13 by github-actions[bot]