betaflight/betaflight #11706: FR: Enable 4 Tune Profiles (Either by default, or as a branch option)

TehllamaFPV, 2022-06-29:
### Is your feature request related to a problem? Please describe

Summary would be changing the provided 3 tune profiles to 4 tune profiles (and all variables subsumed into that, and deconflicting changes with migrating TPA values to the tuneProfile blocks).

Very minor edge case problem, but having a fourth tune profile makes it possible to further idiot-proof multi-voltage (racing) setups that can actually fly with a very wide range of input voltages and take advantage of the Cell Count Auto-Profile Switching facility.

The primary manifestation of this 'problem' is that when using Cell Count Auto-Profile to accommodate different cell count batteries, many of these configurations end up with a gap where the profile does not change and operates off the previously selected profile, which in the case of FreedomSpec/TexasSpec tune profiles that use high gains when 3S batteries are selected can result in severe flyaway conditions and motor burning when a higher voltage battery is plugged in not resulting in a new tune profile being selected.
This is an indirect answer to a user error... but this is a user error I have made made times, with mixed results.  
I do realize I am among the few users actually bothering with multi-voltage tunes, but particularly for non-advanced racers, newer pilots, budget-constrained race pilots, or expanding the utility of certain spec racing configurations, this would be a nice benefit.

### Describe the solution you'd like

#define PID_PROFILE_COUNT 3    -->  #define PID_PROFILE_COUNT 4
(in https://github.com/betaflight/betaflight/blob/1f1f846abbc2ab9bb73aadaac2de6d634285c0aa/src/main/target/common_pre.h and also in the unit test here https://github.com/betaflight/betaflight/blob/025ee87a7aca068e3659fd066b8a9afbed123361/src/test/unit/platform.h)

(Configurator integration would also be required if this becomes a default behavior - although making this an edge case parallel compiled branch and relying extensively on presets could reduce the need for this near-term)... I don't know enough to fully grasp what that change would entail.

### Describe alternatives you've considered

The limitations on this are likely to be total memory space allocation adding another set of tune profile parameters stored on flash to use, and that would currently only benefit a small subset of users with a marginal improvement in utility (since this is really to combat one specific type of user error induced problem when using a very specific type of multi-voltage preset, and is only ever applicable with high performance craft in a KV range that allows for four different cell counts to actually be useful)..

Compiling and using a separate branch would work, and integrating a separate configurator and preset system can be achieved, but the intent of this overall package is to enable more pilots to take advantage of a more fully idiot-proofed multi-voltage system that can safely fly well no matter what battery is plugged in (especially with applying motor output limits to specific over-voltage configurations).

### Other information

If this makes sense to push to back burner, especially if flash space constraints are significant, that is fine.  As stated above, this is a very unique edge cases that requires a combination of relatively advanced multi-voltage tunes and egregious user error to produce erroneous functionality, and there is a definite flash space cost to implement.

haslinghuis, 2022-07-31:
As we tune one profile we should be able to calculate settings for different cell values. But we need some data to verify.

TehllamaFPV, 2022-07-31:
@haslinghuis I already tune for three voltages (and run lots of quads with these setups), and even include dynamic motor output cutoffs on these. It's a bit tedious, but affords huge flexibility in various situations.
Where I'm limited is that on 5" race quads, there is a range of KV (1900-2350KV for my fleets), they can fly viably on 3/4/5/6S.  With some larger quads, there are also ranges where 4S/5S/6S/8S are all viable, and I'd like to have auto-select tunes there.

I realize we can add some auto-calculations, but in the interim, this is the fastest solution to achieve what I'm specifically looking for - what I don't know is if this runs the F722 out of flash space to implement.

haslinghuis, 2022-07-31:
@TehllamaFPV please checkout the firmware and configurator part.

PR #11770 (2022-09-16): Raise PID profile count
Fixes: #11706
closed 2022-09-16 by blckmn