betaflight/betaflight #11710: Can't bind frsky D8 mode with crazybeeF4FR running betaflight 4.3.0

ndellagrotte, 2022-06-30:
### Describe the bug

My Mobula7 HD with the CrazybeeF4FR flight controller running the stable release version of Betaflight 4.3.0 will not bind with my Frsky receiver in d8 mode. The flashing lights on the flight controller and the reciever indicate that the pairing process was successful, but the receiver is not detected in the Betaflight configurator's reciever tab, therefore the RX_LOSS flag prevents me from flying. 

What I tried and why it does not work:

1. Binding in d16 mode is not a solution, because of another firmware related bug that I have yet to create a report for (I will do this soon). In short, when i bind and fly in d16 mode, the quad drops out of the sky after a few seconds of flight with no warnings in the OSD. Yes, I am aware of #6172 and I have telemetry disabled. No, it is not a range issue. RSSI values are good. I have no idea what is causing this.

2. Flashing an older firmware version is not possible, because my quad has the BMI270 gyro which does not work with stock betaflight firmware versions below 4.3.0. I believe that there are modified versions of betaflight 4.2.11 for some iFLight quads that have the BMI270, but apparently Happymodel can't be bothered to make them for their quads or even announce that they're using BMI270 gyros (fuck you, happymodel). I tried contacting support but got no response.

### To Reproduce

1. flash betaflight 4.3.0 stable to a mobula 7 hd 2.0 (alternatively, flash to crazybeef4fr on any quad, but I can't guarantee that the issue will replicate)
2. apply custom defaults
3. bind using the bind button / bind_rx

### Expected behavior

I expected the receiver to bind properly and show up in the receiver tab as i moved the sticks

### Diff configuration

```txt
# 

# diff

# version
# Betaflight / STM32F411 (S411) 4.3.0 Jun 14 2022 / 00:48:04 (229ac66) MSP API: 1.44
# config: manufacturer_id: HAMO, board_name: CRAZYBEEF4FR, version: a074838b, date: 2021-06-14T21:36:54Z

# start the command batch
batch start

board_name CRAZYBEEF4FR
manufacturer_id HAMO

# name: Mobula7 HD

# feature
feature -RX_PARALLEL_PWM

# serial
serial 1 2048 115200 57600 0 115200

# led
led 0 4,7::CO:2
led 1 5,7::CO:2
led 2 6,7::CO:2
led 3 7,7::CO:2
led 4 8,7::CO:2

# aux
aux 0 0 0 1300 2100 0 0
aux 1 1 1 1200 1550 0 0
aux 2 28 1 1550 2100 0 0
aux 3 35 2 1550 2100 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 LOWRACE  L FACTORY 5333 5373 5413 5453 5493 5533 5573 5613
vtxtable powerlevels 5
vtxtable powervalues 10 2 10 16 26
vtxtable powerlabels 0 RCE 25 100 400

# master
set acc_calibration = -40,-22,7,1
set mag_hardware = NONE
set baro_hardware = NONE
set align_board_yaw = -90
set small_angle = 180
set pid_process_denom = 1
set osd_warn_esc_fail = OFF
set osd_vbat_pos = 2091
set osd_rssi_pos = 2081
set osd_tim_1_pos = 54
set osd_tim_2_pos = 2102
set osd_flymode_pos = 2456
set osd_g_force_pos = 65
set osd_throttle_pos = 2423
set osd_vtx_channel_pos = 2357
set osd_crosshairs_pos = 2253
set osd_current_pos = 2400
set osd_mah_drawn_pos = 2433
set osd_craft_name_pos = 2442
set osd_warnings_pos = 329
set osd_disarmed_pos = 2346
set osd_core_temp_pos = 2369
set debug_mode = SMARTAUDIO
set vtx_band = 5
set vtx_channel = 1
set vtx_power = 5
set vtx_freq = 5658
set frsky_spi_tx_id = 88,196,2
set frsky_spi_offset = 19
set frsky_spi_bind_hop_data = 200,161,122,83,44,5,201,162,123,84,45,6,202,163,124,85,46,7,203,164,125,86,47,8,204,165,126,87,48,9,205,166,127,88,49,10,206,167,128,89,50,11,207,168,129,91,51,0,0,0
set frsky_x_rx_num = 1
set gyro_1_align_yaw = 900
set name = Mobula7 HD

profile 0

# profile 0
set p_pitch = 50
set i_pitch = 50
set d_pitch = 30
set f_pitch = 60
set p_roll = 50
set i_roll = 45
set d_roll = 27
set f_roll = 60
set p_yaw = 72
set i_yaw = 55
set f_yaw = 100
set d_min_roll = 0
set d_min_pitch = 0

rateprofile 0

# rateprofile 0
set roll_expo = 23
set pitch_expo = 23
set roll_srate = 75
set pitch_srate = 75

# end the command batch
batch end

#
```

### Resources configuration

```txt
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: GYRO_EXTI
A02: SERIAL_TX 2
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: RX_SPI_CS
B00: ADC_BATT
B01: ADC_CURR
B02: RX_SPI_BIND
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: MOTOR 2
B07: MOTOR 3
B08: MOTOR 4
B09: LED
B10: MOTOR 1
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: RX_SPI_EXTI
C15: BEEPER
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH3 : MOTOR 1
TIM3: FREE
TIM4:
    CH1 : MOTOR 2
    CH2 : MOTOR 3
    CH3 : MOTOR 4
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: TIMUP 4
DMA1 Stream 7: TIMUP 2
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Flight controller

CrazybeeF4FR

mmoustache, 2022-07-11:
Same issue! Makes angry, realy. Please fix it ASAP.

sugaarK, 2022-07-13:
> Same issue on crazybeef4fr ! Makes angry, realy. Please fix it ASAP.

makes you really angry.... great way to get support from a bunch of people working for free who have ZERO to do with the hardware you brought...

Alfinto, 2022-07-13:
CRAZYBEEF4FR (happy model diamond shape) betaflight 4.3.1
irx4 lite multiprotocol module (fw 1.3.3.14) tested with D8 an D16
frksy xlite s (isrm 1.1.0 / 2.1.6 EU) tested with X LBT V2

i can get it to bind with D8, D16 and X LBT V2, but none is stable.
(binding needs a few try's, sometimes it manages to bind on first try)

telemetry connect/disconnect warnings go of constantly
watching the betaflight receiver tab in the configurator, while moving the stickts, it looks like the receiver works for a second just fine, then freezes for second, does that in a loop. 

after D8 bind looks like this:
set rx_spi_protocol = FRSKY_D
set frsky_spi_tx_id = 68,178,1
set frsky_spi_offset = -26
set frsky_spi_bind_hop_data = 0,30,60,91,120,150,180,210,5,35,65,95,125,155,185,215,10,40,70,100,130,160,190,221,15,45,75,105,135,165,195,225,20,50,80,110,140,170,200,230,25,55,85,115,145,175,205,0,0,0

SteveCEvans, 2022-07-15:
I can confirm that both CRAZYBEEF4R and BETAFPVF411RX are constantly failsafeing with 4.3.1.

daleckystepan, 2022-07-16:
Sure I can. I can't imagine how 'Fix CDC_Send_FreeBytes' could directly affect FrSky failsafing - there are changes only in VCP driver that is used for CLI over USB.

Problem was reported also for 4.3.0 which does not include that patch. It needs more investigation or I am missing something. If it randomly works and does not work it might be because of optimization, missing volatile or race condition.

ndellagrotte, 2022-07-16:
> 
The first target aka the "last working" target will not bind in d8, and in d16 the connection appears to be stable from the perspective of the betaflight receiver tab.  However, I have not flown it yet.

The second target aka the "first broken" target will not bind in d8, and in d16 it has dropouts frequently. If I had to guess, there is a dropout every second which lasts half a second.

thuehlinger, 2022-07-16:
I can also confirm that using the "last working" firmware build fixes the same issue for me (Meteor 65). It's worth mentioning, however, that the failsafing issues (probably) only show when the FC is connected to Betaflight Configurator through USB.

daleckystepan, 2022-07-16:
Can you please retest with these builds?

[betaflight_4.4.0_STM32F411.zip](https://github.com/betaflight/betaflight/files/9126791/betaflight_4.4.0_STM32F411.zip)
[betaflight_4.4.0_STM32F411SX1280.zip](https://github.com/betaflight/betaflight/files/9126792/betaflight_4.4.0_STM32F411SX1280.zip)

ndellagrotte, 2022-07-17:
> 

which one should use?

haslinghuis, 2022-07-17:
@ndellagrotte please first use the detect button to select the right board config and then use load firmware local to select the first file as the latter is for F411 (SX1280) ELRS boards.

SteveCEvans, 2022-07-17:
@daleckystepan  what’s the change you’ve made? Did you still want me to make timing measurements?

daleckystepan, 2022-07-18:
@ndellagrotte please use the first one. Sorry for confusion.

@SteveCEvans i have removed delay(1ms) from waiting loop. So it will write new data to buffer asap after there is free space. It can speed it up little bit.

Let's wait for confirmation and then we can decide if we need more testing.

joeparachini, 2022-07-19:
Having this issue as well on CRAZYBEEF4FR with Frsky D8. Happy to help test

thuehlinger, 2022-07-19:
Not sure if maybe we should split this issue in two: I only see the failsafe status flag going on and off and intermittent losses (for a about half a second every second) of the stick position in the displayed timetrace when my Meteor is connected to Betaflight using USB. This happens with both betaflight 4.3.0 and 4.3.1, but not with 4.2.x. I also needed a few attempts to bind using FrSky D16 LBT, but this seemed to be the same both for 4.3.x and 4.2.x. So the original title of this issues does not really apply to me, but the reported losses of transmitter connectivity described here happend to me as well. Should I file a separate bug report? (btw: I can also confirm that removing the 1ms delay in the VCP driver leads to unresponsiveness of the FC in Betaflight and does not solve the issue).

SteveCEvans, 2022-07-19:
@thuehlinger See https://github.com/betaflight/betaflight/pull/11750 which fixes the RC link stability with USB connected.

ndellagrotte, 2022-07-21:
> I've just done a `git bisect` and the outcome, testing on a BETAFPVF411RX is:
> 
> ```
> b6e2790f15cdc27235175b42926c9389ee3c05fe is the first bad commit
> commit b6e2790f15cdc27235175b42926c9389ee3c05fe
> Author: Štěpán Dalecký <daleckystepan@gmail.com>
> Date:   Sat Jul 2 12:03:45 2022 +0200
> 
>     Fix CDC_Send_FreeBytes
> ```
> 
> The preceding commit, which works is:
> 
> ```
> commit 4339e4ffd6c61c87e17dbd515fe77f8079eaf073
> Merge: d46b337 6e04a6b
> Author: J Blackman <blckmn@users.noreply.github.com>
> Date:   Tue Jul 5 12:46:35 2022 +1000
> 
>     Merge pull request #11716 from daleckystepan/fix-mcu-types
>     
>     Fix mcu types
> ```
> 
> Below are the two image, the first being the last good one, and the second being the first broken one. [betaflight_STM32F411_4339e4ff.hex.zip](https://github.com/betaflight/betaflight/files/9123587/betaflight_STM32F411_4339e4ff.hex.zip) [betaflight_STM32F411_b6e2790f.hex.zip](https://github.com/betaflight/betaflight/files/9123588/betaflight_STM32F411_b6e2790f.hex.zip)
> 
> @ndellagrotte Could you please confirm that the first of these images works for you in D8 mode, and the second does not.
> 
> @daleckystepan Could you please investigate.

I would like to reiterate that the first of these two builds does not have the problem with intermittent disconnections every 1.7 seconds. With the first image, the connection appears stable in the Betaflight receiver tab However, I can only bind in d16 mode because d8 binding does not work at all, and in d16 my quad will still failsafe within a few seconds of flying (despite the improvement with build [STM32F411_4339e4ff.hex.](https://github.com/betaflight/betaflight/files/9123587/betaflight_STM32F411_4339e4ff.hex.zip) The issues of d8 connectivity (or lack thereof) and unstable d16 connectivity are prevalent throughout all of the builds I have tested and should probably be separated into 2 separate issues, with this thread focused on the inability of betaflight versions >4.3.0 to bind with FrskyD8, and another thread regarding the instability/failsafes in d16 mode.

joeparachini, 2022-07-24:
@haslinghuis At least during my testing of the assets in #11750, the original issue of unable to bind in d8 mode is not fixed

haslinghuis, 2022-07-24:
Yes aware of the two issues. Please report findings to continue the hunt.

joeparachini, 2022-07-24:
Thank you. Will do. Let me know what else I can do to assist. Side note: is configurator v10.8.0 correct version to test these FWs?

My experience is that with 4.2.11 I can flash, connect, select proper proto, and bind immediately to SPI FRSKY D8 successfully.

`# diff

# version
# Betaflight / STM32F411 (S411) 4.2.11 Nov  9 2021 / 20:28:23 (948ba6339) MSP API: 1.43
# config: manufacturer_id: HAMO, board_name: CRAZYBEEF4FR, version: a074838b, date: 2021-06-14T21:36:54Z

# start the command batch
batch start

board_name CRAZYBEEF4FR
manufacturer_id HAMO

# feature
feature -RX_PARALLEL_PWM

# master
set rx_spi_protocol = FRSKY_D
set frsky_spi_tx_id = 9,3
set frsky_spi_offset = -22
set frsky_spi_bind_hop_data = 6,9,12,15,18,21,24,27,30,33,36,39,42,45,48,51,54,57,60,63,66,69,72,75,78,81,84,87,91,93,96,99,102,105,108,111,114,117,120,123,126,129,132,135,138,141,144,0,0,0
set gyro_1_align_yaw = 900

profile 0

rateprofile 0

# end the command batch
batch end`

Using 'betaflight_4.4.0_STM32F411_6af16ba.hex' from #11750 , I am unable to bind after 3-4 attempts.
However, there does appear to be _some_ kind of link between the two. Even when I'm not trying to bind, I noticed when I hit my bind button on TX that the board does respond with blinky lights. 
https://user-images.githubusercontent.com/30644596/180668268-76d795d4-28f6-4898-bb90-c0bdfbc8a997.mp4

`# diff

# version
# Betaflight / STM32F411 (S411) 4.4.0 Jul 24 2022 / 15:19:12 (6af16ba) MSP API: 1.45
# config: manufacturer_id: HAMO, board_name: CRAZYBEEF4FR, version: a074838b, date: 2021-06-14T21:36:54Z

# start the command batch
batch start

board_name CRAZYBEEF4FR
manufacturer_id HAMO

# feature
feature -RX_SERIAL

# master
set rx_spi_protocol = FRSKY_D
set frsky_spi_tx_id = 9,3,0
set frsky_spi_offset = -8
set frsky_spi_bind_hop_data = 6,9,12,15,18,21,24,27,30,33,36,39,42,45,48,51,54,57,60,63,66,69,72,75,78,81,84,87,91,93,96,99,102,105,108,111,114,117,120,123,126,129,132,135,138,141,144,0,0,0
set gyro_1_align_yaw = 900

profile 0

rateprofile 0

# end the command batch
batch end`

Hope this helps.

haslinghuis, 2022-07-24:
The latest nightly build of configurator should (always) be ready and (could) include some changes needed while progressing changes for the next release and should be fully backwards compatible.
Did you restore settings on the new firmware as the binding looks ok. Only the frsky_spi_tx_id has an extra number attached.
Assume use full chip erase while flashing has been performed.

@SteveCEvans please re-evaluate as I don't have FrSky hardware for testing any of this.

ndellagrotte, 2022-08-22:
Found a temporary fix after months of not being able to even fly my Mobula 7HD. https://ci.betaflight.tech/job/Betaflight/2114/ This is the first nightly build of betaflight (4.2.0) that supports the BMI270 gyro, and it also binds flawlessly in Frsky D8 mode. However, is it a nightly build and I have not flown it yet, although I plan to tomorrow. On a side note, this issue is definitely not fixed by #11750 and I have no idea why it says that at the top.

ndellagrotte, 2022-08-22:
I may have found the problem. I'm absolutely certain that #9697 is the commit that broke d8 binding for me. My radio is the Betafpv Literadio 2 se frsky (CC2500) if that helps at all.

ndellagrotte, 2022-09-02:
Idiot warning: Sometimes I don't know what I'm talking about and will probably misuse FPV terminology or say something wrong, but I think that the information in this comment is mostly accurate and can help someone in the same situation as me. Take it with a grain of salt.

Closing this issue because it will likely never be fixed due to Frsky being doo-doo and breaking compatibility with 3rd party built-in SPI Rx receivers. However, there are workarounds for the issue. If your radio supports Redpine or Futaba SFHSS protocols then, that would be the best solution in my opinion. Since my Literadio 2 SE does not support Redpine, I went with the only other option that I know of for 3rd party Frsky SPI Rx receivers, which is Futaba SFHSS. It has more latency than Redpine, and I'm not sure about range (although in my experience it has been decent with Futaba SFHHS). There are some caveats to that setup, so I made a guide at the bottom of this comment. If you choose Redpine, the steps shouldn't be that much different at all.

If your radio does not support either protocols and you MUST bind in Frsky d8/d16 mode, then you can downgrade to Betaflight 4.2.11 or lower. However, if you have the BMI270 gyro or other newer hardware that is only supported by Betaflight 4.3.x, you can't use any 4.2.x version unless you have an iFlight quad. In that case, check this out https://github.com/Guidus93/betaflight/releases/tag/v4.2.11-ShortageSolver

Futaba SFHSS setup
1. Check if your radio supports Futaba SFHSS. If so, then you can use that instead of Frsky D8/D16 even if the built-in receiver on your FC is Frsky.
2. If your radio is compatible, then flash Betaflight 4.3.0 (not 4.3.1, because your connection will not be stable). 
3. Apply custom defaults
4.  Go to the receiver tab and make sure "Receiver Mode" is set to "SPI Rx (e.g. built-in Rx)". 
5. Also in the receiver tab, select "SPI Bus Receiver Provider" and change it to SFHSS. Save and reboot.
6. Bind

@mmoustache

closed 2022-09-02 by ndellagrotte

mr-sneezy, 2022-10-04:
@ ndellagrotte 
Thanks for posting the details you have, it's probably saved me another bunch of hours chasing a problem I thought was just mine alone (more needs to be posted in RC forums on this issue).

I have three micro copters with F411 boards that all gave me issues with the in-built Frsky SPI RX after flashing to 4.3.1 from 4.1.0, the same approx 1 to 2 second failsafe dropouts for half a second or so making them unflyable.

After reading your info I found that I can run 4.3.0 fine on them (in D mode receiving), just not 4.3.1 so it seems something got broken in 4.3.1 regarding D8 (overclocking the F411 CPU and dropping the PID loop frequency below 4Khz didn't help).

bigjetty, 2022-10-27:
So I'm having this same issue? is there a fix yet?

heinstein187, 2022-11-11:
This is what Happymodel sent me. I was able to bind and fly, but I had to remap the motor, remap the channels, change the direction of the motors, and configure the VTX. 
[betaflight_4.2.10_CRAZYBEEF4FR_norevision_support BMI270.zip](https://github.com/betaflight/betaflight/files/9992855/betaflight_4.2.10_CRAZYBEEF4FR_norevision_support.BMI270.zip)

ahmadazmi19, 2024-05-19:
I want to ask, the FR D8 receiver is completely dead, where does a short circuit usually occur or what is usually damaged?

kingpig1, 2026-09-28:
Im having same isues as said and i have readed the comments and tried the versions and stil not working. Anyone know if there is a final solution?