betaflight/betaflight #11711: crsf_use_rx_snr does not work on SPI ELRS receivers

dopplet, 2022-07-01:
### Describe the bug

I use crsf_use_rx_snr to display SNR dbm instead of raw dbm power, for me this is a much better variable to determine rx signal quality.  Works with Serial based ELRS receivers with no issue.  Recently I built a new drone with a Happy Model X12 AIO which has SPI based ELRS.  When crsf_use_rx_snr is set to ON, the OSD continues to display the raw power dBm and not SNR dBm.

### To Reproduce

set crsf_use_rx_snr on elrs SPI board and verify that raw dBm is shown on OSD.

### Expected behavior

SNR dBm should be shown on OSD

### Diff configuration

```txt
defaults nosave

# version
# Betaflight / CRAZYBEEF4SX1280 (HAMO) 4.3.0 Jun 14 2022 / 00:31:27 (229ac66) MSP API: 1.44

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name CRAZYBEEF4SX1280
manufacturer_id HAMO
mcu_id 0050003f5547501620333332
signature 

# name: TH

# feature
feature -AIRMODE

# serial
serial 1 2048 115200 57600 0 115200

# beeper
beeper -ON_USB

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# aux
aux 0 0 0 1800 2100 0 0
aux 1 1 1 1800 2100 0 0
aux 2 13 2 1800 2100 0 0
aux 3 19 2 1400 1600 0 0
aux 4 28 1 1400 1600 0 0
aux 5 36 3 1800 2100 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 LOWRACE  L FACTORY 5333 5373 5413 5453 5493 5533 5573 5613
vtxtable powerlevels 5
vtxtable powervalues 10 2 14 20 26
vtxtable powerlabels 0 RCE 25 100 400

# rxfail
rxfail 5 s 2000
rxfail 6 s 1000

# master
set gyro_lpf1_static_hz = 0
set gyro_lpf2_static_hz = 0
set dyn_notch_count = 1
set dyn_notch_q = 500
set gyro_lpf1_dyn_min_hz = 0
set gyro_lpf1_dyn_max_hz = 0
set acc_calibration = -26,75,-37,1
set crsf_use_rx_snr = ON
set dshot_idle_value = 1200
set dshot_bidir = ON
set dshot_bitbang = AUTO
set motor_pwm_protocol = DSHOT300
set motor_poles = 12
set motor_output_reordering = 3,2,0,1,4,5,6,7
set failsafe_delay = 50
set failsafe_off_delay = 100
set failsafe_throttle = 1200
set failsafe_procedure = AUTO-LAND
set vbat_min_cell_voltage = 290
set vbat_warning_cell_voltage = 310
set ibata_scale = 470
set yaw_motors_reversed = ON
set small_angle = 180
set deadband = 1
set yaw_deadband = 1
set pid_process_denom = 3
set simplified_gyro_filter = OFF
set report_cell_voltage = ON
set osd_warn_batt_not_full = OFF
set osd_warn_batt_warning = OFF
set osd_warn_visual_beeper = OFF
set osd_warn_esc_fail = OFF
set osd_warn_core_temp = OFF
set osd_warn_rc_smoothing = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_warn_rssi = ON
set osd_warn_link_quality = ON
set osd_rssi_alarm = 0
set osd_cap_alarm = 450
set osd_alt_alarm = 110
set osd_tim1 = 0
set osd_tim2 = 1
set osd_vbat_pos = 6551
set osd_rssi_pos = 216
set osd_link_quality_pos = 14401
set osd_link_tx_power_pos = 6273
set osd_rssi_dbm_pos = 14433
set osd_tim_1_pos = 6600
set osd_tim_2_pos = 6568
set osd_flymode_pos = 14369
set osd_throttle_pos = 6615
set osd_vtx_channel_pos = 6305
set osd_crosshairs_pos = 14605
set osd_ah_pos = 142
set osd_current_pos = 14784
set osd_mah_drawn_pos = 14754
set osd_craft_name_pos = 14337
set osd_display_name_pos = 394
set osd_gps_speed_pos = 152
set osd_gps_lon_pos = 16403
set osd_gps_lat_pos = 16436
set osd_gps_sats_pos = 120
set osd_home_dir_pos = 238
set osd_home_dist_pos = 183
set osd_flight_dist_pos = 215
set osd_compass_bar_pos = 8
set osd_altitude_pos = 16472
set osd_pid_roll_pos = 353
set osd_pid_pitch_pos = 34
set osd_pidrate_profile_pos = 6337
set osd_warnings_pos = 14730
set osd_avg_cell_voltage_pos = 14721
set osd_pit_ang_pos = 6575
set osd_rol_ang_pos = 6607
set osd_disarmed_pos = 14635
set osd_esc_rpm_pos = 161
set osd_rtc_date_time_pos = 40
set osd_flip_arrow_pos = 310
set osd_core_temp_pos = 6583
set osd_rate_profile_name_pos = 23
set osd_pid_profile_name_pos = 2
set osd_stat_rtc_date_time = ON
set osd_stat_tim_1 = ON
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_min_rssi = OFF
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_max_g_force = ON
set osd_stat_min_link_quality = ON
set osd_stat_total_time = ON
set osd_stat_min_rssi_dbm = ON
set debug_mode = RX_EXPRESSLRS_SPI
set cpu_overclock = 120MHZ
set vtx_band = 5
set vtx_channel = 8
set vtx_power = 5
set vtx_low_power_disarm = UNTIL_FIRST_ARM
set vtx_freq = 5917
set vcd_video_system = PAL
set gyro_1_sensor_align = CW270
set gyro_1_align_yaw = 2700
set expresslrs_uid = 107,108,159,151,125,52
set expresslrs_rate_index = 2
set expresslrs_switch_mode = WIDE
set name = TH

profile 0

# profile 0
set p_pitch = 94
set i_pitch = 168
set d_pitch = 92
set f_pitch = 250
set p_roll = 90
set i_roll = 160
set d_roll = 80
set f_roll = 240
set p_yaw = 90
set i_yaw = 160
set f_yaw = 240
set d_min_roll = 60
set d_min_pitch = 68
set feedforward_smooth_factor = 30
set feedforward_jitter_factor = 9
set simplified_master_multiplier = 200

profile 1

# profile 1
set p_pitch = 70
set i_pitch = 126
set d_pitch = 69
set f_pitch = 187
set p_roll = 67
set i_roll = 120
set d_roll = 60
set f_roll = 180
set p_yaw = 67
set i_yaw = 120
set f_yaw = 180
set d_min_roll = 45
set d_min_pitch = 51
set feedforward_smooth_factor = 30
set feedforward_jitter_factor = 9
set simplified_d_gain = 150
set simplified_pi_gain = 150
set simplified_feedforward_gain = 150

profile 2

# profile 2
set p_pitch = 79
set i_pitch = 142
set d_pitch = 78
set f_pitch = 212
set p_roll = 76
set i_roll = 136
set d_roll = 68
set f_roll = 203
set p_yaw = 76
set i_yaw = 136
set f_yaw = 203
set d_min_roll = 50
set d_min_pitch = 57
set feedforward_smooth_factor = 30
set feedforward_jitter_factor = 9
set simplified_master_multiplier = 170

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set roll_rc_rate = 1
set pitch_rc_rate = 1
set yaw_rc_rate = 1
set roll_expo = 67
set pitch_expo = 67
set yaw_expo = 67
set roll_srate = 80
set pitch_srate = 80
set yaw_srate = 80

rateprofile 1

rateprofile 2

# rateprofile 2
set rates_type = BETAFLIGHT
set roll_rc_rate = 110
set pitch_rc_rate = 110
set yaw_rc_rate = 100
set roll_srate = 75
set pitch_srate = 75
set yaw_srate = 70
set tpa_breakpoint = 1250

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
```

### Resources configuration

```txt
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: GYRO_EXTI
A02: SERIAL_TX 2
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: RX_SPI_EXPRESSLRS_RESET
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: RX_SPI_EXPRESSLRS_BUSY
A14: PREINIT 2
A15: RX_SPI_CS
B00: ADC_BATT
B01: ADC_CURR
B02: RX_SPI_BIND
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: MOTOR 4
B07: MOTOR 2
B08: MOTOR 1
B09: LED
B10: MOTOR 3
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: RX_SPI_EXTI
C15: BEEPER
D02: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1 : DSHOT_BITBANG 2
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: DSHOT_BITBANG 2
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Flight controller

Happy Model X12 ELRS AIO

dopplet, 2022-07-02:
Not a coder, but did some grepping and found the logic in crsf.c that performed this task.  After a couple of tries I came up with the following diff that works by displaying the SNR dB on the OSD now and it aligns with the telemetry data shown on the radio.

```
diff --git a/src/main/rx/expresslrs.c b/src/main/rx/expresslrs.c
index 93815353e..70b79767d 100644
--- a/src/main/rx/expresslrs.c
+++ b/src/main/rx/expresslrs.c
@@ -1010,7 +1010,13 @@ static void handleLinkStatsUpdate(const uint32_t timeStampMs)
             uint16_t rssiScaled = scaleRange(constrain(receiver.rssiFiltered, receiver.rfPerfParams->sensitivity, -50), receiver.rfPerfParams->sensitivity, -50, 0, 1023);
             setRssi(rssiScaled, RSSI_SOURCE_RX_PROTOCOL);
 #ifdef USE_RX_RSSI_DBM
-            setRssiDbm(receiver.rssiFiltered, RSSI_SOURCE_RX_PROTOCOL);
+            if (rxConfig()->crsf_use_rx_snr) {
+                int16_t rssiDbm = 0;
+                rssiDbm = receiver.snr;
+                setRssiDbmDirect(rssiDbm,  RSSI_SOURCE_RX_PROTOCOL);
+            } else {
+                setRssiDbmDirect(receiver.rssiFiltered, RSSI_SOURCE_RX_PROTOCOL);
+            }
 #endif
 #ifdef USE_RX_LINK_QUALITY_INFO
             setLinkQualityDirect(receiver.uplinkLQ);
@@ -1021,7 +1027,11 @@ static void handleLinkStatsUpdate(const uint32_t timeStampMs)
         } else {
             setRssiDirect(0, RSSI_SOURCE_RX_PROTOCOL);
 #ifdef USE_RX_RSSI_DBM
-            setRssiDbmDirect(-130, RSSI_SOURCE_RX_PROTOCOL);
+            if (rxConfig()->crsf_use_rx_snr) {
+                setRssiDbmDirect(-30, RSSI_SOURCE_RX_PROTOCOL);
+            } else {
+                setRssiDbmDirect(-130, RSSI_SOURCE_RX_PROTOCOL);
+            }
 #endif
 #ifdef USE_RX_LINK_QUALITY_INFO
             setLinkQualityDirect(0);
```

I don't know if there are any negative side effects to this, or if this is the right approach.

phobos-, 2022-07-08:
What if users want to see rssi dBm and snr dBm at the same time? It's better to just add a separate element to display snr. The work has already been done to make it possible, see this PR: https://github.com/betaflight/betaflight/pull/11062

closed 2022-08-22 by github-actions[bot]