betaflight/betaflight #11715: Better disclosure of malware from the wiki

kamnxt, 2022-07-02:
### Is your feature request related to a problem? Please describe

As mentioned by @sirherpderp in #11625, someone was trying to share malicious software on the wiki. Someone I know got infected on Jun 11th, which we discovered today as their laptop opened PayPal automatically and tried to log in.

The malware installed Netsupport Client in `%appdata%/SupportWinUp/client32.exe`, also added to autostart in Windows. Netsupport was of course set up to not show any popups, tray icon etc.

### Describe the solution you'd like

IMO Betaflight should issue some kind of announcement telling users to check if they have downloaded the malicious software. Maybe tell a few of the larger FPV related blogs to mention this as well.

### Describe alternatives you've considered

Not applicable here?

joymonkey, 2022-07-03:
I just had this same thing happen yesterday, which I believe may trace back to the same malicious download link for (what I thought was) the Zadig driver fix tool. Someone used the same Netsupport Client malware to open a browser, log into PayPal, and send themselves $250 USD. The recipient's email address was VZpaypal2022 @ gmail.com , and they go by the moniker "Valerii Zimin" , the transaction memo reads "~@~CLD:1657177200~@~" . PayPal has refused to refund me, as it was a Friends & Family transaction.

KarateBrot, 2022-07-07:
Good news guys, the account in question (Pedrocesarg) just got banned :)

closed 2022-08-14 by github-actions[bot]