betaflight/betaflight #11717: Explicit Landing Mode - Reduced Rates, Throttle Limit, Air Mode disabled, Flight Mode

jhancuff, 2022-07-05:
### Is your feature request related to a problem? Please describe

The ability to consistently land larger, heavier quads such as X8 quads and loaded cinelifters.

### Describe the solution you'd like

I think it would be nice to have a configurable landing mode that can be assigned to a switch where the user can configure a different set of rates, throttle limits, and the ability to temporarily disable Air Mode to prevent a heavy quad from bouncing upon landing, especially when it's an X8 configuration with landing legs.

For example, change rates to something super slow and linear, limit throttle range, change throttle curve to enable finer control.  Disable Air Mode to prevent bouncing.  Change flight mode to angle with reduced max angle.  All of these options should be user configurable (e.g. alt rates, throttle limit and curve, max angle, and air mode).

Again, the practical application, especially for me, is landing encumbered cinelifters with pricey payloads and X8 configuration where you can't really get a feeling for when the landing legs are going to contact the landing surface.

### Describe alternatives you've considered

Programming logic into my TX16S to use additional aux channels to disable air mode and change rate profiles.  But this isn't available to everyone (DJI users, for example).  It'd be nice for it to be a standard mode that can be assigned to a switch position.

etracer65, 2022-07-05:
Everything you're asking for can already be accomplished using a single aux channel using existing functionality. There is no need to use multiple additional aux channels or do any programming in your radio. You can set up your "landing mode" that changes the rateprofile to your "low rates" preference, throttle limit is also part of the rateprofile, you can activate angle mode, and disable air mode all using a single switch. And since these settings all already exist you're free to customize them however you want.

sugaarK, 2022-07-06:
@jhancuff  as mr etracer said you can do all of this with rare profiles, a single aux switch and using the adjustment tab. You can stack multiple functions onto a single aux switch

jhancuff, 2022-07-06:
Got it, thanks!