betaflight/betaflight #11720: ELRS RX works in Emu but not in BF. F411.

zheroz00, 2022-07-08:
### Describe the bug

**RX was bound prior to installing on the FC.

I have an iFlight F411 which uses the IFLIGHT_F411_PRO target. A BetaFPV ELRS nano RX receiver is connected to UART2. (I cannot use UART1) The ports and receiver type are configured in BF 4.3, however the TX and RX will not connect. When powered the RX immediately goes into DFU mode. It seems a workaround is to add a resistor to allow it to proceed past DFU during boot, but it doesn't seem necessary as this issue does not exist in Emuflight.

Leaving the FC wired the same as above, I flash Emu 0.4.0 to the FC, set the port and receiver type and reboot. The TX and RX connect immediately. Powering on/off results with it continuing to power up and connect. Switch back to BF 4.3 and perform the same steps and it does not connect. I've repeated this process three times to be sure.

### To Reproduce

1. Flash IFLIGHT_F411_PRO in BF 4.3.
2. Setup the ports to have UART2 use Serial RX.
3. Set the receiver mode to Serial and the provider to CRSF.
4. Save and reboot. No RX connection. Cycle power several times, no connection.
5. Flash FC with Emuflight 0.4.0.
6. Setup the ports to have UART2 use Serial RX.
7. Set the receiver mode to Serial and the provider to CRSF.
8. Save and reboot. RX and TX connect immediately. Cycle power several times and the connection continues to work.
9. Flash back to BF 4.3 and perform steps 1-4 again. No RX connection.
10. Flash back to Emu and perform steps 5-8. RX connection is good.

### Expected behavior

ELRS to not enter DFU mode on boot.

### Diff configuration

```txt
diff

# version
# Betaflight / STM32F411 (S411) 4.3.0 Jun 14 2022 / 00:48:04 (229ac66) MSP API: 1.44
# config: manufacturer_id: IFRC, board_name: IFLIGHT_F411_PRO, version: 714fcb3f, date: 2021-12-07T06:25:44Z

# start the command batch
batch start

board_name IFLIGHT_F411_PRO
manufacturer_id IFRC

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature TELEMETRY

# serial
serial 0 1 115200 57600 0 115200
serial 1 64 115200 57600 0 115200

# master
set dyn_notch_count = 1
set dyn_notch_q = 500
set acc_calibration = -36,46,120,1
set mag_hardware = NONE
set serialrx_provider = CRSF
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set align_board_roll = 180
set align_board_pitch = 180
set yaw_motors_reversed = ON

profile 0

rateprofile 0

# end the command batch
batch end

#
```

### Resources configuration

```txt
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: MOTOR 1
A01: GYRO_EXTI
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FLASH_CS
B00: ADC_BATT
B01: ADC_CURR
B02: BEEPER
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: MOTOR 3
B07: MOTOR 4
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: MOTOR 2
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: LED 2
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH1 : MOTOR 1
    CH3 : MOTOR 2
TIM3: FREE
TIM4:
    CH1 : MOTOR 3
    CH2 : MOTOR 4
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: MOTOR 3
DMA1 Stream 1: MOTOR 2
DMA1 Stream 2: SPI_MISO 3
DMA1 Stream 3: MOTOR 4
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: MOTOR 1
DMA1 Stream 6: FREE
DMA1 Stream 7: SPI_MOSI 3
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Flight controller

iFlight Succex-D F411 AIO v3.2

### Other components

RX = BetaFPV ELRS nano EP1
VTX = Caddx Vista

### How are the different components wired up

RX connected to UART2 with Serial RX enabled in ports. Wiring is correct. i.e. TX > RX, RX > TX.

Caddx Vista connected to UART1.

mituritsyn, 2022-07-14:
this is known issue 
https://www.expresslrs.org/2.0/quick-start/troubleshooting/#i-think-my-receiver-and-transmitter-are-bound-as-the-led-on-the-receiver-is-solid-but-i-do-not-have-stick-inputs-or-rssi-on-radio
https://github.com/ExpressLRS/ExpressLRS/issues/655#issuecomment-1172733275
the new is that in emu it is working fine. previously it was explained by uncontrolled uart activity  before code was performed 
@Quick-Flash

mituritsyn, 2022-07-19:
@JyeSmith @CapnBry

CapnBry, 2022-07-19:
I don't have any hardware that experiences this issue so I have nothing to contribute. We know that pulling the RX line low at the FC (the TX pin of the ESP8285) locks it into the bootloader, and adding a pullup prevents it, but I am not anywhere near the level of Betaflight platform guru needed to know how the pins get initialized and that timing.

mituritsyn, 2022-07-19:
> I don't have any hardware that experiences this issue so I have nothing to contribute. We know that pulling the RX line low at the FC (the TX pin of the ESP8285) locks it into the bootloader, and [...]

i just wanted the elrs team to be aware. it was explained earlier by some BF devs that this problem most likely occurs before BF code loaded and uarts  initialised. obviously this is not entirely true. we have a lot of people in our local elrs chat experiencing this problem as cinewhoop and toothpicks have become very popular. For most of them, simply re-soldering to another uart is difficult enough, let alone adding a 3.3 pull-up

sugaarK, 2022-07-19:
Sounds like the betaFPV RX unit is it built to specification

mituritsyn, 2022-08-02:
> Sounds like the betaFPV RX unit isn't built to specification

No, this problem affects all receivers. The point is that at some moment the tx or rx are pulled down and esp goes into boot mode.
@hydra

SteveCEvans, 2022-08-05:
I've connected a [BetaFPV ELRS Lite](https://betafpv.com/collections/rx/products/elrs-lite-receiver) to a [REVONANO](https://librepilot.atlassian.net/wiki/spaces/LPDOC/pages/26968079/OpenPilot+Revolution+Nano) (for ease of swapping between the ports as they're pluggable) running 4.3.1 and it works without issue on both the Main (UART2) and Flexi (UART1) ports.

SteveCEvans, 2022-08-05:
Also, are you powering from USB or a battery?

SteveCEvans, 2022-08-05:
The boot loader specification can be found at:

https://www.st.com/resource/en/application_note/cd00167594-stm32-microcontroller-system-memory-boot-mode-stmicroelectronics.pdf#page27

In table 3 can be seen the following suggesting that UART 1 and 2 should behave the same.

[image]

Once running the application code, be it Betaflight or something else, the boot loader won’t be entered due to activity on the DFU port. Of course the code can jump back to the boot loader, but that only happens if detected by `checkForBootLoaderRequest()`.

I can create a version of code that catches this for debug I guess, but I doubt it’s being invoked.

CapnBry, 2022-08-07:
The ELRS version doesn't matter. When zheroz00 says "DFU mode" I believe they're mistakenly referring to the ESP bootloader as DFU mode, which I believe has confused the situation. What they're saying is that the ESP8285 is stuck in bootloader mode, which happens before any ELRS code runs. This occurs when the receiver's TX line is low on startup of the receiver MCU. Betaflight is up and running, not in DFU, but the receiver won't boot due to the condition of its TX pin.

The implication is that the Betaflight startup pin init and the EMUflight startup pin init are subtly different, and that the Betaflight implementation somehow has its RX pin low at the critical moment during the ESP's boot process, and EMUflight does not. It could simply be a timing thing, like EMUflight starts up more slowly and _would_ block the receiver's boot process but it happens a couple milliseconds later and thus the receiver has already passed the bootloader pin check.

The only way to know for sure would be to put a scope on the OP's hardware and measure the timing of the RX line going high triggered off the application of power to the FC, then do the same for the other software. That wouldn't suggest any fix but could tell if there's genuinely something different between the two procedures or if the same thing happens just at a more optimal time.

SteveCEvans, 2022-08-07:
@CapnBry The DFU reference threw me. Your explanation makes more sense. As you say, scoping the power on behaviour is the only way to establish what’s going on.

hydra, 2022-08-11:
> No, this problem affects all receivers. The point is that at some moment the tx or rx are pulled down and esp goes into boot mode. @hydra

@mituritsyn Am I being mentioned for a specific reason?  I don't think this is something for me.

However I will say that the boot loader entry requirements  are in the ESP8266 and ESP32 data sheets and agree that any connected device should not immediately trigger them.

To fix this we need:
* links to schematics for the affected receivers.
* scope traces on the ESP IO lines that deal with boot loader entry for BF, and EF so the difference can be visualized.
* analysis of the GPIO/RX startup code in BF.
* ideally a specific git commit that changed/broke the behaviour, find an older working version of BF, then use `git bisect`.

But like I said, this isn't for me.

mituritsyn, 2022-08-13:
> @mituritsyn Am I being mentioned for a specific reason? I don't think this is something for me.

Sorry if I bothered you, I was just following your commits and thought this type of problem was closest to you.

> To fix this we need:
> 
> * links to schematics for the affected receivers.

all uart receivers are affected on some FC. In most it is f411 based controllers, the problem is usually only reproduced on the second uart, but in some cases it is on both

> * ideally a specific git commit that changed/broke the behaviour, find an older working version of BF, then use `git bisect`.

there wasn't working version of BF

> But like I said, this isn't for me.

sorry again, perhaps you know someone who could solve this puzzle

SteveCEvans, 2022-08-13:
I’ll look to see if a pull-up can be configured on UART lines.

zheroz00, 2022-08-22:
> I've connected a [BetaFPV ELRS Lite](https://betafpv.com/collections/rx/products/elrs-lite-receiver) to a [REVONANO](https://librepilot.atlassian.net/wiki/spaces/LPDOC/pages/26968079/OpenPilot+Revolution+Nano) (for ease of swapping between the ports as they're pluggable) running 4.3.1 and it works without issue on both the Main (UART2) and Flexi (UART1) ports.
> 
> Note that this was with ELRS 2.5.0.
> 
> @zheroz00 What version of ELRS are you using?

Apologies for the late reply. I was out of town. I am on 2.5.1 as well.

zheroz00, 2022-08-22:
> Also, are you powering from USB or a battery?

It can be powered via USB or the battery. I've tried both. Neither are working in BF for me unfortunately.

zheroz00, 2022-08-22:
> The ELRS version doesn't matter. When zheroz00 says "DFU mode" I believe they're mistakenly referring to the ESP bootloader as DFU mode, which I believe has confused the situation. What they're [...]

You are correct. Sorry about that. I meant bootloader.

Prajwal-Koirala, 2023-12-27:
This is a bug report about an issue where an ELRS receiver goes into DFU mode when connected to a specific flight controller board. The bug does not occur with Emuflight firmware. The user has tried flashing different versions of both Betaflight and ELRS firmware, but the problem persists. Others have reported similar issues with different ELRS receivers and FC boards.

The cause of the bug is likely related to the way Betaflight initializes the receiver's UART pins. A possible fix could be to add a pull-up resistor to the receiver's TX line.

I hope this summary is helpful! Let me know if you have any other questions.

haslinghuis, 2024-09-10:
Closing as stalled.

closed 2024-09-10 by haslinghuis