betaflight/betaflight #11735: It is not possible to configure the MATEKF405SE controller after applying the factory settings (BF 4.3.0 - 4.3.1).

VMaximI, 2022-07-12:
### Describe the bug

It is not possible to configure the MATEKF405SE controller after applying the factory settings. When trying to save the settings, the configurator does not respond. CLI not working (no connection).
Version 4.2.11 and earlier do not have these issues.
If you flash 4.3.0 and immediately disable i2c2, then there is no further problem.

### To Reproduce

1. Flash target MATEKF405SE
2. Reboot the controller.
3. Return it to the factory settings with the button in the configurator.

From now on, you will not be able to configure it (save changes)

### Expected behavior

After a factory reset, the controller will reboot and be ready for configuration.
Pressing the "Save and Reboot" button saves the settings and reloads the flight controller. All changes are saved.
CLI works and allows you to change the parameters of the controller and save them.

### Diff configuration

```txt
Impossible. CLI not responding
```

### Resources configuration

```txt
Impossible. CLI not responding
```

### Flight controller

Matek F405 SE

### Add any other context about the problem that you think might be relevant here

If after the firmware you refuse to reset to the factory settings and apply these commands, then there is no further problem.

# Betaflight / STM32F405 (S405) 4.0.4 Jun 30 2019 / 14:58:15 (f3a95efa3) MSP API: 1.41

board_name MATEKF405SE
manufacturer_id MTKS

# resource
resource BEEPER 1 C15
resource MOTOR 1 B07
resource MOTOR 2 B06
resource MOTOR 3 B00
resource MOTOR 4 B01
resource MOTOR 5 C08
resource MOTOR 6 C09
resource MOTOR 7 B14
resource MOTOR 8 B15
resource PPM 1 A03
resource LED_STRIP 1 A15
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 C10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 C12
resource SERIAL_TX 6 C06
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 C11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 D02
resource SERIAL_RX 6 C07
resource I2C_SCL 1 B08
resource I2C_SCL 2 NONE
resource I2C_SDA 1 B09
resource I2C_SDA 2 NONE
resource LED 1 A14
resource LED 2 A13
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 B03
resource SPI_MISO 1 A06
resource SPI_MISO 2 C02
resource SPI_MISO 3 B04
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 C03
resource SPI_MOSI 3 B05
resource ADC_BATT 1 C00
resource ADC_RSSI 1 C05
resource ADC_CURR 1 C01
resource SDCARD_CS 1 C14
resource OSD_CS 1 B12
resource GYRO_EXTI 1 C04
resource GYRO_CS 1 A04
resource USB_DETECT 1 C13

# timer
timer B07 AF2
# pin B07: TIM4 CH2 (AF2)
timer B06 AF2
# pin B06: TIM4 CH1 (AF2)
timer B00 AF2
# pin B00: TIM3 CH3 (AF2)
timer B01 AF2
# pin B01: TIM3 CH4 (AF2)
timer C08 AF3
# pin C08: TIM8 CH3 (AF3)
timer C09 AF3
# pin C09: TIM8 CH4 (AF3)
timer B14 AF9
# pin B14: TIM12 CH1 (AF9)
timer B15 AF9
# pin B15: TIM12 CH2 (AF9)
timer A08 AF1
# pin A08: TIM1 CH1 (AF1)
timer A15 AF1
# pin A15: TIM2 CH1 (AF1)
timer A03 AF3
# pin A03: TIM9 CH2 (AF3)
timer A02 AF2
# pin A02: TIM5 CH3 (AF2)

# dma
dma SPI_TX 3 0
# SPI_TX 3: DMA1 Stream 5 Channel 0
dma ADC 1 0
# ADC 1: DMA2 Stream 0 Channel 0
dma pin B07 0
# pin B07: DMA1 Stream 3 Channel 2
dma pin B06 0
# pin B06: DMA1 Stream 0 Channel 2
dma pin B00 0
# pin B00: DMA1 Stream 7 Channel 5
dma pin B01 0
# pin B01: DMA1 Stream 2 Channel 5
dma pin C08 1
# pin C08: DMA2 Stream 4 Channel 7
dma pin C09 0
# pin C09: DMA2 Stream 7 Channel 7
dma pin A08 0
# pin A08: DMA2 Stream 6 Channel 0
dma pin A15 0
# pin A15: DMA1 Stream 5 Channel 3
dma pin A02 0
# pin A02: DMA1 Stream 0 Channel 6

# feature
feature RX_SERIAL
feature TELEMETRY
feature OSD
feature SOFTSERIAL

# serial
serial 1 64 115200 57600 0 115200

# master
set serialrx_provider = SBUS
set mag_bustype = I2C
set mag_i2c_device = 1
set mag_hardware = AUTO
set baro_bustype = I2C
set baro_i2c_device = 1
set baro_i2c_address = 118
set baro_hardware = AUTO
set blackbox_device = SDCARD
set dshot_burst = OFF
set current_meter = ADC
set battery_meter = ADC
set ibata_scale = 179
set beeper_inversion = ON
set beeper_od = OFF
set sdcard_mode = SPI
set sdcard_spi_bus = 3
set system_hse_mhz = 8
set max7456_spi_bus = 2
set gyro_1_bustype = SPI
set gyro_1_spibus = 1
set gyro_1_sensor_align = CW270
set camera_control_mode = HARDWARE_PWM
set osd_vbat_pos = 2467
set osd_current_pos = 2434
set osd_disarmed_pos = 2282

haslinghuis, 2022-07-12:
Did you use full chip erase while flashing?

From the config it should be `set mag_i2c_device = 2`
```
set mag_bustype = I2C
set mag_i2c_device = 2
set mag_hardware = AUTO
set baro_bustype = I2C
set baro_i2c_device = 1
set baro_i2c_address = 118
set baro_hardware = AUTO
```

Please also provide `status` command output from cli.

VMaximI, 2022-07-13:
> Вы использовали полное стирание чипа при перепрошивке?
> 
> Из конфига должно быть`set mag_i2c_device = 2`
> 
> ```
> set mag_bustype = I2C
> set mag_i2c_device = 2
> set mag_hardware = AUTO
> set baro_bustype = I2C
> set baro_i2c_device = 1
> set baro_i2c_address = 118
> set baro_hardware = AUTO
> ```
> 
> Пожалуйста, также предоставьте `status`вывод команды из cli.

Yes, of course, I use a full chip erasure.

# status
MCU F40X Clock=168MHz (PLLP-HSI), Vref=3.28V, Core temp=37degC
Stack size: 2048, Stack address: 0x1000fff0
Configuration: UNCONFIGURED, size: 4100, max available: 16384
Devices detected: SPI:0, I2C:0
Gyros detected: locked

OSD: NONE
System Uptime: 46 seconds, Current Time: 2022-07-13T07:09:30.946+00:00
CPU:0%, cycle time: 0, GYRO rate: 0, RX rate: 0, System rate: 10
Voltage: 0 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
SD card: Not configured
Arming disable flags: NOGYRO BOOTGRACE CLI MSP

I have now completely flashed the controller with a full chip wipe, but I did not do a factory reset before using the "status" command, otherwise I will lose the ability to use the CLI until a full flashing.

VMaximI, 2022-07-13:
I think the problem is
resource I2C_SCL 2 B10
resource I2C_SDA 2 B11
which originated in BF 4.3
if I2C2 is disabled, then everything will be fine. But this cannot be done after resetting the controller to factory settings, it will have to be reflashed again and, refusing to reset, apply the necessary commands.