betaflight/betaflight #11736: LQ always 0%

bebenoeg, 2022-07-12:
### Describe the bug

Hello I'm Bebe, I'm a Betaflight user on my quad, and I have a problem after upgrading the firmware to 4.3 via betaflight 10.8 i.e. LQ is always 0% and flashing. whereas for RSSI can appear normal.
Previously I used firmware 4.2.11 and everything worked normally.
Please help me how to fix it.
Thanks.

### To Reproduce

normal

### Expected behavior

normal LQ

### Configurator version

10.8.0

### Flight controller configuration

```txt
# diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.3.0 Jul  3 2022 / 23:19:02 (b3b2e5ae7) MSP API: 1.44
# config: manufacturer_id: DIAT, board_name: MAMBAF722_I2C, version: edcd244b, date: 2022-06-29T02:24:46Z

# start the command batch
batch start

board_name MAMBAF722_I2C
manufacturer_id DIAT

# name: nug

# feature
feature -RX_PARALLEL_PWM
feature GPS

# serial
serial 1 2 115200 57600 0 115200
serial 2 8192 115200 57600 0 115200

# beacon
beacon RX_LOST
beacon RX_SET

# aux
aux 0 0 0 900 1075 0 0
aux 1 1 7 1400 2100 0 0
aux 2 46 7 1925 2100 0 0
aux 3 13 1 1925 2100 0 0
aux 4 35 2 1925 2100 0 0
aux 5 36 1 1925 2100 0 0

# adjrange
adjrange 0 0 5 900 2100 12 5 0 0
adjrange 1 0 6 900 2100 29 6 0 0

# rxrange
rxrange 0 987 2011
rxrange 1 987 2011
rxrange 2 987 2011
rxrange 3 987 2011

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 3
vtxtable powervalues 25 200 400
vtxtable powerlabels 25 200 400

# rxfail
rxfail 8 s 1000

# master
set gyro_lpf1_static_hz = 0
set gyro_lpf2_static_hz = 750
set dyn_notch_count = 1
set dyn_notch_q = 500
set gyro_lpf1_dyn_min_hz = 0
set acc_calibration = -67,-23,-56,1
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1010
set max_check = 1990
set rssi_channel = 9
set dshot_bidir = ON
set failsafe_delay = 4
set failsafe_procedure = GPS-RESCUE
set vbat_max_cell_voltage = 435
set ibata_scale = 81
set beeper_dshot_beacon_tone = 5
set small_angle = 180
set gps_provider = UBLOX
set gps_sbas_mode = MSAS
set gps_set_home_point_once = ON
set gps_rescue_angle = 33
set gps_rescue_initial_alt = 60
set gps_rescue_descent_dist = 100
set gps_rescue_ground_speed = 1500
set gps_rescue_throttle_min = 1180
set gps_rescue_sanity_checks = RESCUE_SANITY_FS_ONLY
set gps_rescue_min_sats = 6
set gps_rescue_min_dth = 50
set simplified_gyro_filter_multiplier = 150
set tlm_inverted = ON
set osd_warn_batt_not_full = OFF
set osd_warn_link_quality = ON
set osd_cap_alarm = 5000
set osd_alt_alarm = 400
set osd_rssi_pos = 6201
set osd_link_quality_pos = 14425
set osd_tim_1_pos = 2454
set osd_flymode_pos = 6177
set osd_throttle_pos = 6505
set osd_current_pos = 6496
set osd_mah_drawn_pos = 2441
set osd_gps_speed_pos = 6227
set osd_gps_lon_pos = 4496
set osd_gps_lat_pos = 4464
set osd_gps_sats_pos = 6190
set osd_home_dir_pos = 6224
set osd_home_dist_pos = 6217
set osd_altitude_pos = 6209
set osd_pidrate_profile_pos = 2086
set osd_avg_cell_voltage_pos = 14721
set osd_stat_tim_1 = ON
set osd_stat_tim_2 = OFF
set osd_stat_max_dist = ON
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_min_rssi = OFF
set osd_stat_max_alt = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_min_link_quality = ON
set vtx_band = 5
set vtx_channel = 8
set vtx_power = 3
set vtx_low_power_disarm = UNTIL_FIRST_ARM
set vtx_freq = 5917
set vcd_video_system = NTSC
set gyro_1_align_yaw = 1800
set name = nug

profile 0

# profile 0
set dterm_lpf1_dyn_min_hz = 112
set dterm_lpf1_dyn_max_hz = 225
set dterm_lpf1_static_hz = 112
set dterm_lpf2_static_hz = 225
set p_pitch = 65
set i_pitch = 117
set d_pitch = 47
set f_pitch = 175
set p_roll = 63
set i_roll = 112
set d_roll = 42
set f_roll = 168
set p_yaw = 63
set i_yaw = 112
set f_yaw = 168
set d_min_roll = 42
set d_min_pitch = 47
set simplified_master_multiplier = 140
set simplified_dmax_gain = 0
set simplified_dterm_filter_multiplier = 150

rateprofile 0

# rateprofile 0
set roll_rc_rate = 15
set pitch_rc_rate = 15
set yaw_rc_rate = 20
set roll_expo = 76
set pitch_expo = 76
set yaw_expo = 64
set roll_srate = 91
set pitch_srate = 91
set yaw_srate = 80

# end the command batch
batch end
```

matyas-brenner, 2022-07-17:
Duplicity or related to this issue?
https://github.com/betaflight/betaflight/issues/11600

haslinghuis, 2022-07-17:
Duplicate of #11600

closed 2022-07-17 by haslinghuis