betaflight/betaflight #11743: SPI RX RX_Failsafe due to RQly = 0 handling

avrgandy, 2022-07-16:
### Describe the bug

Since updating to Betaflight 4.3 from 4.2.11 on a standard original EMAX TinyHawk whoop and a standard EMAX TinyHawy Freestyle, OSD (and MSP, when connected) give constant RX_FAILSAFE / LINK QUALITY errors and Jumper T-Lite speaks "Telemetry Lost/Recovered" persistently). 
Could be that BF 4.3 handles RQly differently (original TinyHawk SPI (FRsky D8) always send RQly = 0, but this didn't trigger warnings/errors in BF 4.2.11)?
Have tried FRSkyX D16, telemetry on/off (Tinyhawk always provides basic telemetry including RQly even when NO_TELEMETY selected in BF).
Downgrading to BF 4.2.11 resolves the issue.

### To Reproduce

Upgrade TinyHawk or TinyHawk Freestyle to BF 4.3. Gives "LINK QUALITY" error in OSD and  constant 'telemetry lost' audio on Jumper T-Lite.

### Expected behavior

Expect RX RQly=0 handling to be the same in BF 4.2.11 and 4.3. Appears to be handled differently.

### Diff configuration

```txt
# 

# diff all

# version
# Betaflight / STM32F411 (S411) 4.2.11 Nov  9 2021 / 20:28:23 (948ba6339) MSP API: 1.43
# config: manufacturer_id: MTKS, board_name: MATEKF411RX, version: edcd244b, date: 2022-06-29T02:24:46Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name MATEKF411RX
manufacturer_id MTKS
mcu_id 005b00183439510836383737
signature 

# name: Tinyhawk

# feature
feature -RX_PARALLEL_PWM

# beacon
beacon RX_LOST
beacon RX_SET

# serial
serial 0 2048 115200 57600 0 115200
serial 1 64 115200 57600 0 115200

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 3 900 1300 0 0
aux 2 2 3 1300 1700 0 0
aux 3 27 1 1300 2100 0 0
aux 4 13 2 1700 2100 0 0
aux 5 35 2 1700 2100 0 0

# rxrange
rxrange 0 987 2011
rxrange 1 987 2011
rxrange 2 987 2011
rxrange 3 987 2011

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5655 5665 5885 5905 5905 5905
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 1
vtxtable powerlabels 25

# master
set gyro_lowpass2_hz = 400
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set dyn_notch_min_hz = 140
set dyn_notch_max_hz = 400
set dyn_lpf_gyro_min_hz = 320
set dyn_lpf_gyro_max_hz = 800
set acc_calibration = 5,-3,63,1
set mag_hardware = NONE
set baro_hardware = NONE
set fpv_mix_degrees = 15
set rx_spi_protocol = FRSKY_D
set dshot_idle_value = 350
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set motor_poles = 12
set bat_capacity = 500
set vbat_max_cell_voltage = 445
set vbat_min_cell_voltage = 315
set vbat_warning_cell_voltage = 325
set current_meter = VIRTUAL
set ibata_scale = 117
set ibatv_scale = 117
set beeper_dshot_beacon_tone = 4
set small_angle = 180
set deadband = 5
set yaw_deadband = 5
set osd_warn_rssi = ON
set osd_warn_link_quality = ON
set osd_warn_over_cap = ON
set osd_rssi_pos = 2106
set osd_tim_2_pos = 2081
set osd_flymode_pos = 2412
set osd_g_force_pos = 2425
set osd_throttle_pos = 2393
set osd_vtx_channel_pos = 2097
set osd_current_pos = 2400
set osd_mah_drawn_pos = 2434
set osd_craft_name_pos = 2444
set osd_pidrate_profile_pos = 2417
set osd_warnings_pos = 14474
set osd_avg_cell_voltage_pos = 2369
set osd_disarmed_pos = 2283
set osd_flip_arrow_pos = 2254
set osd_core_temp_pos = 2456
set osd_stat_battery = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set vtx_band = 4
set vtx_channel = 4
set vtx_power = 1
set vtx_freq = 5800
set vcd_video_system = NTSC
set frsky_spi_tx_id = 165,62
set frsky_spi_offset = -38
set frsky_spi_bind_hop_data = 0,30,60,91,120,150,180,210,5,35,65,95,125,155,185,215,10,40,70,100,130,160,190,221,15,45,75,105,135,165,195,225,20,50,80,110,140,170,200,230,25,55,85,115,145,175,205,0,0,0
set gyro_1_align_yaw = 1800
set gyro_rpm_notch_harmonics = 2
set name = Tinyhawk

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 112
set dyn_lpf_dterm_max_hz = 272
set dterm_lowpass2_hz = 240
set iterm_relax_cutoff = 12
set p_pitch = 83
set i_pitch = 117
set d_pitch = 69
set f_pitch = 124
set p_roll = 77
set i_roll = 111
set d_roll = 64
set f_roll = 117
set p_yaw = 82
set i_yaw = 117
set f_yaw = 117
set level_limit = 60
set d_min_roll = 42
set d_min_pitch = 46

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
#
```

### Resources configuration

```txt
# 

# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: GYRO_EXTI
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: RX_SPI_CC2500_TX_EN
A09: SERIAL_TX 1
A10: FREE
A11: USB
A12: USB
A13: RX_SPI_CC2500_LNA_EN
A14: RX_SPI_CC2500_ANT_SEL
A15: RX_SPI_CS
B00: ADC_BATT
B01: FREE
B02: RX_SPI_BIND
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: MOTOR 2
B07: MOTOR 3
B08: MOTOR 4
B09: LED
B10: MOTOR 1
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: RX_SPI_EXTI
C15: BEEPER
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1: DSHOT_BITBANG 2
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: DSHOT_BITBANG 2
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

#
```

### Flight controller

Standard original EMAX TinyHawk / TinyHawkFS

egonl, 2022-07-26:
My Tinyhawk Freestyle II does something similar in 4.3.1: RX reception is 'jumpy' and telemetry lost/recovered. Sometimes it works for a (couple of) second(s), then jumps to 1500us. I'm using the internal XJT module of a Taranis Q X7 in D8 mode. 
A log with `blackbox_mode = ALWAYS` and `debug_mode = RX_FRSKY_SPI` is attached, where I pitch up most of the time.
[LOG163 - pitching up.TXT](https://github.com/betaflight/betaflight/files/9193617/LOG163.-.pitching.up.TXT)

egonl, 2022-07-30:
This seems to be resolved in the latest 4.4.0. I guess https://github.com/betaflight/betaflight/commit/3c4a23ef55f25e7689d069767255c854cb02a40f fixed it.

haslinghuis, 2022-07-31:
Fixed in #11764

closed 2022-07-31 by haslinghuis