betaflight/betaflight #11748: Gyro fails to initialize randomly

JohnDGodwin, 2022-07-18:
### Describe the bug

I can reproduce this issue on two separate flight controllers. The first is a happymodel aio and the second is a mamba 20x20. I can mostly connect to betaflight as normal, but after a few reboot cycles just with GUI commands like hitting "Save and Reboot" will sometimes fail to initialize the gyro and I will get the no gyro found errors shown below.
[image]
[image]

### To Reproduce

plug in flight controller and use gui commands that reboot the flight controller such as enabling GPS or changing motor protocol.

### Expected behavior

the gyro to initialize. 

### Diff configuration

```txt
# diff all

# version
# Betaflight / STM32F405 (S405) 4.3.1 Jul 13 2022 / 03:32:11 (8d4f005) MSP API: 1.44
# config: manufacturer_id: DIAT, board_name: MAMBAF405US, version: edcd244b, date: 2022-06-29T02:24:46Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name MAMBAF405US
manufacturer_id DIAT
mcu_id 004100385553501120323156
signature 

# feature
feature -RX_PARALLEL_PWM

# serial
serial 0 2 115200 57600 0 115200
serial 1 64 115200 57600 0 115200

# master
set acc_calibration = -259,38,-112,1

profile 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
#
```

### Resources configuration

```txt
resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: SERIAL_TX 4
A01: SERIAL_RX 4
A02: FREE
A03: SERIAL_RX 2
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: MOTOR 2
A09: MOTOR 1
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: PREINIT 2
B00: PINIO 1
B01: FREE
B02: FREE
B03: FREE
B04: FREE
B05: SPI_MOSI 3
B06: FREE
B07: FREE
B08: CAMERA_CONTROL
B09: FREE
B10: FREE
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: INVERTER 1
C01: ADC_BATT
C02: FREE
C03: ADC_CURR
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: MOTOR 4
C09: MOTOR 3
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: FREE
C13: BEEPER
C14: LED 2
C15: LED 1
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1 : MOTOR 2
    CH2 : MOTOR 1
TIM2: FREE
TIM3: FREE
TIM4:
    CH3 : CAMERA_CONTROL
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH3 : MOTOR 4
    CH4 : MOTOR 3
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 3
DMA2 Stream 1: TIMUP 8
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: FREE
DMA2 Stream 5: TIMUP 1
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

#
```

### Flight controller

happymodel elrs 2g4 aio

mituritsyn, 2022-07-19:
another way to reproduce - go to cli and back on any tab

JohnDGodwin, 2022-07-20:
> Looks like ICM20689 troubles. I have the same problems but if I wait few minutes gyro start works ok. #11753

One of the flight controllers uses an MPU6000 gyro though.

MJ666, 2022-08-08:
You should be sure to keep the FC absolutely steady during initialization and gyro calibration. As long the gyro is not calibrated the response in the configurator is wonky and calibration may even fail and the gyro will not work. Initialization and calibration is finished when LED is flashing 3 times and if a beeper is present after 3 beeps. At this point you can start moving the FC.

JaceRyczak, 2022-08-28:
I am experiencing this issue on my Mamba Mini F722 Mk2. has an MPU6000, This issue is not present in BF4.2.x but happens every second reboot of my FC, I lose the gyro and Acc.

SteveCEvans, 2022-08-28:
Is this when powered from a battery/USB or both?

JaceRyczak, 2022-08-29:
Hi,

I have yet to try with a battery since the update, this is just powered via
USB. I will try tonight and see what the result is powering it with a LiPo.

Regards,
Jace.

JaceRyczak, 2022-08-29:
> Is this when powered from a battery/USB or both?

I have done some further testing and have confirmed that the gyro is not found whether or not a battery is connected. It seems though, that it happens ONLY when told to reboot by BF. I can unplug and re-plug the battery and it gets detected without fail. Likewise I can disconnect from BF, then replug usb and connect and the gyro is fine every time. 

In summary: seems the BF "Save & Reboot" function is causing the issue?

closed 2022-10-07 by github-actions[bot]