betaflight/betaflight #11752: BF 4.4 nightly: i have issue with gps arrow not pointing to home point but it keeps slowly rotating on dji osd, but when i switch my RTH swittch quad fly back to home point no problem..not test this issue with failsafe switch though.. and this only happen with dji osd quad no issue on analog. with 4.3 theres no gps home arrow issue at all with my dji quads.

dnlkrnwn, 2022-07-20:
i have issue with gps arrow not pointing to home point but it keeps slowly rotating on dji osd, but when i switch my RTH swittch quad fly back to home point no problem..not test this issue with failsafe switch though.. and this only happen with dji osd quad no issue on analog. with 4.3 theres no gps home arrow issue at all with my dji quads.

_Originally posted by @dnlkrnwn in https://github.com/betaflight/betaflight/issues/11579#issuecomment-1189323219_

lavaed, 2022-08-11:
I have a similar problem with the arrow on osd dji. This was a problem with older versions of betaflight as well. And I noticed one feature, the farther you fly away from the take-off point, the greater the deviation of the arrow from the correct course, but my quad, unlike yours, flies in the direction of the arrow, which leads to the loss of the quad.

lavaed, 2022-08-12:
I turned off the compass and the arrow began to correctly show the direction home.

ctzsnooze, 2022-08-30:
Thanks @toddferrante for identifying the build that caused the problem.
 
It was a GPS PR https://github.com/betaflight/betaflight/pull/11579 which, amongst many changes, increased the resolution of the GPS Heading data by a factor of 10, so that where previously it might have been reported as 179 for 179 degrees, it would now be reported as 1793, meaning 179.3.
 
The analog OSD code is within Betaflight was adjusted at the same time.

I don't know who maintains the code that displays the arrow for DJI OSD users?

There are two possible solutions.

One is for DJI to update their code to the higher resolution we now use.

The alternative is to modify line 1475 of msp.c from

```
        sbufWriteU16(dst, GPS_directionToHome);
```
to
```
        sbufWriteU16(dst, GPS_directionToHome / 10);
```
I don't know what's best at this point.

kristjanbjarni, 2022-09-20:
Since the DJI Digital FPV is basically an abandoned project at this point it means that DJI will not update their code. So this has to be fixed in BF either by going back to using the previous resolution or have a CLI flag to use the previous resolution. I know the BF developers hate hacks like these, but the DJI hardware base is pretty big at this point.

simoccc, 2022-10-02:
I have this issue, random, on all my BF ( 4.4/4.3 ) dji quad with gps ( BN880 and matek 5883) , is it possible that this can depend on DJI code into betaflight ? 
I'm saying this because on analog I never ever got one problem with the arrow. 
In my case the quad follow the arrow even if it's wrong home directions. I have other friends that got the same random problem.

toddferrante, 2022-10-02:
We tracked down the cause of this issue and ctzsnooze fixed it.  See the PR he posted above.

haslinghuis, 2022-10-03:
Fixed in #11814

closed 2022-10-03 by haslinghuis