betaflight/betaflight #11753: Gyro doesn't work first few minutes after power connection (battery or PC connection)

igor-lemon, 2022-07-20:
### Describe the bug

When I connect power to FC the gyro of FC doesn't work first few minutes.
After 2-5 minutes gyro works correctly. But if I disconnect power and connect back again the gyro doesn't work again but after few minutes start works again.

I hear about ICM-20689 problems but I never flash RC versions.
Also I'm 100% sure that FC is ok because I have 2 similar FC and one of it I never use before, I had Betaflight v3 on new FC but when I upgraded to betaflight v4.3 I have the same problem on this FC.

[image]

### To Reproduce

Just connect FC to PC, connect to betafight, check Sensors.

### Expected behavior

FC gyro should work correctly immediately

### Diff configuration

```txt
# diff

# version
# Betaflight / STM32F745 (S745) 4.3.1 Jul 13 2022 / 03:35:21 (8d4f005) MSP API: 1.44
# config: manufacturer_id: HBRO, board_name: KAKUTEF7, version: cc8e118f, date: 2020-03-04T02:37:20Z

# start the command batch
batch start

board_name KAKUTEF7
manufacturer_id HBRO

# master
set gyro_1_align_yaw = 2700

profile 0

rateprofile 0

# end the command batch
batch end
```

### Resources configuration

```txt
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: LED 1
A03: FREE
A04: SDCARD_CS
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FREE
B00: FREE
B01: FREE
B02: FREE
B03: FREE
B04: FREE
B05: FREE
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: FREE
B10: FREE
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: ADC_CURR
C03: ADC_BATT
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: SDCARD_DETECT
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: BEEPER
E00: FREE
E01: GYRO_EXTI
E02: SPI_SCK 4
E03: FREE
E04: GYRO_CS 1
E05: SPI_MISO 4
E06: SPI_MOSI 4
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: SPI_MISO 4
DMA2 Stream 1: SPI_MOSI 4
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

#
```

### Flight controller

Hollybro KAKUTE F7 AIO v1.3

### Add any other context about the problem that you think might be relevant here

1. Whean I call `gyroregisters` command sometimes I have

```
# gyroregisters

# Gyro 1
# WHO_AM_I    0x98
# CONFIG      0x0
# GYRO_CONFIG 0x0
```

sometimes

```
# gyroregisters

# Gyro 1
# WHO_AM_I    0x98
# CONFIG      0x0
# GYRO_CONFIG 0x18
```

2. When I away from CLI tab the betaflight loose connection with FC and disconnecting, I need to connect again.

igor-lemon, 2022-07-22:
Up

I found interesting behaviour. If I connect, put on the table and don't shake the FC gyro start works ok almost instantly.
[image]

haslinghuis, 2022-08-08:
@igor-lemon closing as duplicate of https://github.com/betaflight/betaflight/issues/11748

Let's continue there.

closed 2022-08-08 by haslinghuis