betaflight/betaflight #11754: throttle jumps between 1000 and 885

wedwincode, 2022-07-20:
### Describe the bug

When transmitter is on, throttle always falls to failsafe value (885) and returns to normal (1000). Other channels do the same, but jump to the middle(1500). I have tried to re-bind my receiver but there is the same problem. Check the video for better understanding: [link](https://youtu.be/VlDgtYSOIfE).

### To Reproduce

Connect transmitter, check receiver tab

### Expected behavior

Channels should work correctly

### Diff configuration

```txt
# version
# Betaflight / STM32F411 (S411) 4.3.1 Jul 13 2022 / 03:32:52 (8d4f005) MSP API: 1.44
# config: manufacturer_id: HAMO, board_name: CRAZYBEEF4FR, version: a074838b, date: 2021-06-14T21:36:54Z

# start the command batch
batch start

board_name CRAZYBEEF4FR
manufacturer_id HAMO

# resources
resource MOTOR 1 B07
resource MOTOR 2 B08
resource MOTOR 3 B10
resource MOTOR 4 B06

# feature
feature -RX_PARALLEL_PWM
feature TELEMETRY
feature RSSI_ADC

# serial
serial 20 5 115200 57600 0 115200

# map
map TAER1234

# master
set acc_calibration = -68,0,38,1
set align_board_yaw = 90
set small_angle = 180
set osd_warn_batt_not_full = OFF
set osd_cap_alarm = 450
set osd_vbat_pos = 2081
set osd_rssi_pos = 2137
set osd_flymode_pos = 2506
set osd_g_force_pos = 2497
set osd_crosshairs_pos = 2253
set osd_mah_drawn_pos = 2104
set osd_craft_name_pos = 2538
set osd_display_name_pos = 361
set osd_avg_cell_voltage_pos = 2113
set osd_stat_total_time = ON
set frsky_spi_tx_id = 111,138,7
set frsky_spi_offset = -31
set frsky_spi_bind_hop_data = 6,104,204,69,169,34,136,234,99,199,64,164,29,129,229,94,194,59,159,24,124,224,89,189,54,154,19,119,219,84,184,49,149,14,114,214,79,179,44,144,9,109,209,74,174,39,139,0,0,0
set gyro_1_sensor_align = CW90FLIP
set gyro_1_align_pitch = 1800
set gyro_1_align_yaw = 900

profile 2

rateprofile 3

# end the command batch
batch end
```

### Resources configuration

```txt
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: GYRO_EXTI
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: RX_SPI_CS
B00: ADC_BATT
B01: ADC_CURR
B02: RX_SPI_BIND
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: MOTOR 4
B07: MOTOR 1
B08: MOTOR 2
B09: LED
B10: MOTOR 3
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: RX_SPI_EXTI
C15: BEEPER
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH3 : MOTOR 3
TIM3: FREE
TIM4:
    CH1 : MOTOR 4
    CH2 : MOTOR 1
    CH3 : MOTOR 2
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: TIMUP 4
DMA1 Stream 7: TIMUP 2
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Flight controller

Crazybee f4fr v3.0

KarateBrot, 2022-07-20:
Are your receiver and transmitter right next to each other? FrSky seems to like failsafing when they are too close to one another.

wedwincode, 2022-07-20:
@KarateBrot I took my transmitter to another room and the problem still exists