betaflight/betaflight #11755: VBAT/ADC not reading correct value on G4 when using VREF+ @3.3v

x4FF3, 2022-07-22:
### Describe the bug

Hi,

i try to get the ADC/VBAT reading working on a G4 board, which uses VREF+ @3.3V and B02 as vbat input. I tried multiple ADCs for pin PB2/B02. Voltage stays at value of about 12v regardless of voltage. 
Tested on 4.3.1 and current master.

KR
David

### To Reproduce

use G4 board and use PB2/B02 as vbat input pin. connect VREF+ to 3.3v

### Expected behavior

correct reading of voltage

### Diff configuration

```txt
# Betaflight / BOARDG4 (BG4) 4.4.0 Jul 20 2022 / 10:27:17 (norevision) MSP API: 1.45

# start the command batch
batch start

board_name BOARDG4

# master
set adc_device = 1

profile 0

rateprofile 0

# end the command batch
batch end
```

### Resources configuration

```txt
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: MOTOR 1
A01: MOTOR 2
A02: MOTOR 3
A03: MOTOR 4
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: OSD_CS
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: I2C_SCL 1
A14: I2C_SDA 1
A15: BEEPER
B00: GYRO_CS 2
B01: ADC_CURR
B02: ADC_BATT
B03: FREE
B04: FREE
B05: FREE
B06: FREE
B07: FREE
B08: FREE
B09: FREE
B10: FREE
B11: FREE
B12: FREE
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: FLASH_CS
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: FREE
C14: LED 2
C15: LED 1
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1 : DSHOT_BITBANG 1
TIM15: FREE
TIM16: FREE
TIM17: FREE
TIM20: FREE

Currently active DMA:
--------------------
DMA1 Channel 1: DSHOT_BITBANG 1
DMA1 Channel 2: ADC 1
DMA1 Channel 3: SPI_MOSI 1
DMA1 Channel 4: SPI_MISO 1
DMA1 Channel 5: SPI_MOSI 2
DMA1 Channel 6: SPI_MISO 2
DMA1 Channel 7: FREE
DMA1 Channel 8: FREE
DMA2 Channel 1: FREE
DMA2 Channel 2: FREE
DMA2 Channel 3: FREE
DMA2 Channel 4: FREE
DMA2 Channel 5: FREE
DMA2 Channel 6: FREE
DMA2 Channel 7: FREE
DMA2 Channel 8: FRE
```

### Flight controller

BOARDG4

DieHertz, 2022-08-08:
I think I'm experienced same issue here on another G4 in the works

SteveCEvans, 2022-08-08:
I’ll be able to look at this in a week’s time.

DieHertz, 2022-08-08:
No worries, in the meantime I'm soldering a debugger myself and trying to remember how to use it.

DieHertz, 2022-08-13:
That's not just PB2, try typing `status` and you'll see Vref and Core Temp values are also bogus.

closed 2022-09-21 by github-actions[bot]