betaflight/betaflight #11760: ExpressLRS Link on SPI board (HM) unstable in 4.3.1

FabianIsensee, 2022-07-24:
### Describe the bug

Hey there, it seems like the RC link of ExpressLRS on AIOs (SPI RX) is unstable in Betaflight 4.3.1. Upon arming the LQ drops from 100 to 75, then 50, then 25 and finally drops out completely. DIsarming brings it back to 100 instantaneously.

I have a video detailing the problem here: https://www.youtube.com/watch?v=9MxbzhAm2S4
(in the vid its just a naked board without camera or motors)
In the video I am sitting right next to the board (~1m) with my TX in hand, as you can see from the dBm value. The board shows the same behavior when in a quad - its not related to it being 'naked'.

I am quite certain that is has to be a software problem on the betaflight build for this board because:
- downgrading to BF 4.3.0 solves the issue
- other RX (EP2) work fine with my TX
- multiple AIO I have show the same problem (and they are all working on 4.3.0)

Tested hardware & software:
- ELRS v2.0.1 and 2.5.1 on my TX
- BF 4.3.1 target CRAZYBEEF4SX1280 (does not work)
- BF 4.3.0 target CRAZYBEEF4SX1280 (works)

Best,
Fabian

### To Reproduce

Install BF 4.3.1 on a whoop AIO with SPI ELRS receiver, try to fly. LQ drops step by step on arm until RX loss.

### Expected behavior

LQ should stay at 100 or close to 100

### Diff configuration

```txt
# diff

# version
# Betaflight / STM32F411SX1280 (S4SX) 4.3.1 Jul 13 2022 / 03:33:32 (8d4f005) MSP API: 1.44
# config: manufacturer_id: HAMO, board_name: CRAZYBEEF4SX1280, version: 7df66093, date: 2022-06-12T13:41:42Z

# start the command batch
batch start

board_name CRAZYBEEF4SX1280
manufacturer_id HAMO

# serial
serial 1 2048 115200 57600 0 115200

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# aux
aux 0 0 0 1700 2100 0 0
aux 1 13 1 1700 2100 0 0
aux 2 35 2 1700 2100 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 3
vtxtable powervalues 0 1 2
vtxtable powerlabels 25 100 200

# master
set gyro_lpf1_static_hz = 412
set gyro_lpf2_static_hz = 825
set dyn_notch_count = 1
set dyn_notch_q = 500
set gyro_lpf1_dyn_min_hz = 412
set gyro_lpf1_dyn_max_hz = 825
set acc_hardware = NONE
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set motor_poles = 12
set beeper_dshot_beacon_tone = 3
set yaw_motors_reversed = ON
set simplified_gyro_filter_multiplier = 165
set osd_vbat_pos = 2135
set osd_link_quality_pos = 2167
set osd_rssi_dbm_pos = 2199
set osd_tim_2_pos = 2103
set osd_stat_max_esc_rpm = ON
set osd_stat_max_fft = ON
set vtx_band = 5
set vtx_channel = 1
set vtx_power = 1
set vtx_freq = 5658
set expresslrs_uid = 121,114,34,87,187,109
set rpm_filter_harmonics = 2
set rpm_filter_min_hz = 125

profile 0

# profile 0
set dterm_lpf1_dyn_min_hz = 123
set dterm_lpf1_dyn_max_hz = 247
set dterm_lpf1_static_hz = 123
set dterm_lpf2_static_hz = 247
set pidsum_limit = 1000
set pidsum_limit_yaw = 1000
set p_pitch = 94
set i_pitch = 75
set d_pitch = 68
set f_pitch = 149
set p_roll = 90
set i_roll = 72
set d_roll = 60
set f_roll = 143
set p_yaw = 90
set i_yaw = 72
set f_yaw = 143
set d_min_roll = 60
set d_min_pitch = 68
set simplified_master_multiplier = 200
set simplified_i_gain = 45
set simplified_dmax_gain = 0
set simplified_feedforward_gain = 60
set simplified_dterm_filter_multiplier = 165

rateprofile 0

# rateprofile 0
set rates_type = BETAFLIGHT
set roll_rc_rate = 100
set pitch_rc_rate = 100
set yaw_rc_rate = 100
set roll_srate = 58
set pitch_srate = 58
set yaw_srate = 58

# end the command batch
batch end
```

### Resources configuration

```txt
# 
# Building AutoComplete Cache ... Done!
# 
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: LED_STRIP
A01: GYRO_EXTI
A02: SERIAL_TX 2
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: RX_SPI_EXPRESSLRS_RESET
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: RX_SPI_EXPRESSLRS_BUSY
A14: PREINIT 2
A15: RX_SPI_CS
B00: ADC_BATT
B01: ADC_CURR
B02: RX_SPI_BIND
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: MOTOR 2
B07: MOTOR 3
B08: MOTOR 4
B09: LED
B10: MOTOR 1
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: RX_SPI_EXTI
C15: BEEPER
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH3 : MOTOR 1
TIM3: FREE
TIM4:
    CH1 : MOTOR 2
    CH2 : MOTOR 3
    CH3 : MOTOR 4
TIM5:
    CH1 : LED_STRIP
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: MOTOR 2
DMA1 Stream 1: MOTOR 1
DMA1 Stream 2: LED_STRIP
DMA1 Stream 3: MOTOR 3
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: MOTOR 4
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Flight controller

Happymodel ExpressLRS AIO V1.0

### Other components

Happymodel ES24TX Module in a FrSky X9D Plus radio

### How are the different components wired up

everything is as it should be

FabianIsensee, 2022-07-24:
[image]

TX configuration, no sync on arm could be relevant!?

LoakAudio, 2022-08-08:
Hi, I confirm the same behavior on the same card. On my side "no_sync_on_arm" is NOT activated in  the TX module.
4.3.0 = OK
4.3.1 = bad ELRS link

I use the same diff on both : 

`defaults nosave

# version
# Betaflight / CRAZYBEEF4SX1280 (HAMO) 4.3.0 Jun 14 2022 / 00:31:27 (229ac66) MSP API: 1.44

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name CRAZYBEEF4SX1280
mcu_id 0066003b4e56500820363955
signature 

# name: Crux3 NLR

# feature
feature GPS

# serial
serial 0 2 115200 57600 0 115200
serial 1 2048 115200 57600 0 115200

# beeper
beeper -ARMING_GPS_FIX
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -READY_BEEP
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 2 1300 1700 0 0
aux 2 27 3 1700 2100 0 0
aux 3 46 2 1700 2100 0 0
aux 4 13 1 900 1300 0 0
aux 5 35 1 1700 2100 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 3
vtxtable powervalues 0 1 2
vtxtable powerlabels 25 100 200

# rxfail
rxfail 3 s 1575
rxfail 6 s 1500

# master
set dyn_notch_count = 5
set acc_trim_pitch = -4
set acc_trim_roll = 6
set acc_calibration = -55,-37,-53,1
set rc_smoothing_auto_factor = 35
set rc_smoothing_auto_factor_throttle = 28
set rc_smoothing_setpoint_cutoff = 12
set rc_smoothing_feedforward_cutoff = 12
set rc_smoothing_throttle_cutoff = 20
set serialrx_provider = CRSF
set blackbox_device = NONE
set dshot_idle_value = 600
set motor_pwm_protocol = DSHOT300
set motor_poles = 12
set motor_output_reordering = 3,2,1,0,4,5,6,7
set failsafe_delay = 5
set failsafe_procedure = GPS-RESCUE
set align_board_yaw = 180
set bat_capacity = 2000
set vbat_max_cell_voltage = 420
set vbat_min_cell_voltage = 280
set vbat_warning_cell_voltage = 310
set ibata_scale = 1232
set yaw_motors_reversed = ON
set mixer_type = LINEAR
set small_angle = 180
set gps_provider = UBLOX
set gps_ublox_use_galileo = ON
set gps_set_home_point_once = ON
set gps_rescue_initial_alt = 25
set gps_rescue_descent_dist = 30
set gps_rescue_ground_speed = 800
set gps_rescue_throttle_max = 1950
set gps_rescue_ascend_rate = 200
set gps_rescue_descend_rate = 100
set gps_rescue_throttle_hover = 1575
set gps_rescue_min_sats = 5
set gps_rescue_min_dth = 50
set gps_rescue_allow_arming_without_fix = ON
set deadband = 6
set yaw_deadband = 6
set osd_warn_batt_not_full = OFF
set osd_warn_launch_control = OFF
set osd_warn_rssi_dbm = ON
set osd_rssi_dbm_alarm = -107
set osd_cap_alarm = 1800
set osd_core_temp_alarm = 75
set osd_link_quality_pos = 14699
set osd_link_tx_power_pos = 2379
set osd_rssi_dbm_pos = 2348
set osd_tim_2_pos = 14709
set osd_flymode_pos = 2390
set osd_throttle_pos = 2307
set osd_vtx_channel_pos = 363
set osd_crosshairs_pos = 14477
set osd_current_pos = 2337
set osd_mah_drawn_pos = 2371
set osd_gps_speed_pos = 14422
set osd_gps_lon_pos = 2113
set osd_gps_lat_pos = 2081
set osd_gps_sats_pos = 2096
set osd_home_dir_pos = 14446
set osd_home_dist_pos = 2198
set osd_flight_dist_pos = 2166
set osd_altitude_pos = 22582
set osd_warnings_pos = 14569
set osd_avg_cell_voltage_pos = 14690
set osd_disarmed_pos = 14507
set osd_flip_arrow_pos = 14606
set osd_core_temp_pos = 2357
set osd_stat_max_dist = ON
set osd_stat_battery = ON
set osd_stat_min_rssi = OFF
set osd_stat_max_alt = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_min_link_quality = ON
set osd_stat_flight_dist = ON
set osd_stat_min_rssi_dbm = ON
set vtx_band = 1
set vtx_channel = 8
set vtx_power = 3
set vtx_low_power_disarm = ON
set vtx_freq = 5725
set vcd_video_system = NTSC
set expresslrs_uid = 85,36,152,190,203,181
set expresslrs_rate_index = 3
set expresslrs_switch_mode = WIDE
set name = Crux3 NLR

profile 0

# profile 0
set p_pitch = 75
set i_pitch = 100
set d_pitch = 85
set f_pitch = 140
set p_roll = 80
set i_roll = 100
set d_roll = 85
set f_roll = 140
set p_yaw = 115
set i_yaw = 100
set f_yaw = 140
set d_min_roll = 40
set d_min_pitch = 45
set feedforward_smooth_factor = 20
set feedforward_jitter_factor = 15
set feedforward_boost = 5
set dyn_idle_min_rpm = 25
set simplified_pids_mode = OFF

profile 1

# profile 1
set dterm_lpf1_dyn_min_hz = 90
set dterm_lpf1_dyn_max_hz = 180
set dterm_lpf1_static_hz = 90
set dterm_lpf2_static_hz = 180
set p_pitch = 65
set i_pitch = 90
set d_pitch = 85
set f_pitch = 140
set p_roll = 70
set i_roll = 90
set d_roll = 85
set f_roll = 140
set p_yaw = 105
set i_yaw = 90
set f_yaw = 140
set d_min_roll = 40
set d_min_pitch = 45
set feedforward_smooth_factor = 20
set feedforward_jitter_factor = 15
set feedforward_boost = 5
set dyn_idle_min_rpm = 25
set simplified_pids_mode = OFF
set simplified_dterm_filter_multiplier = 120

profile 2

# restore original profile selection
profile 2

rateprofile 0

# rateprofile 0
set roll_rc_rate = 2
set pitch_rc_rate = 2
set yaw_rc_rate = 2
set roll_expo = 50
set pitch_expo = 50
set yaw_expo = 50
set roll_srate = 38
set pitch_srate = 38
set yaw_srate = 38
set tpa_breakpoint = 1550

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration`

LoakAudio, 2022-08-11:
In the Discord chanel V-22 asked me the results of "DMA show" and "Timer show".
Here they are. We can see some differences between version 4.3.0 (which works normally) and version 4.3.1 : 

Betaflight 4.3.0 : 

`# dma show

Currently active DMA:
--------------------
DMA1 Stream 0: MOTOR 2
DMA1 Stream 1: MOTOR 1
DMA1 Stream 2: SPI_MISO 3
DMA1 Stream 3: MOTOR 3
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: MOTOR 4
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# timer show

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH3 : MOTOR 1
TIM3: FREE
TIM4:
    CH1 : MOTOR 2
    CH2 : MOTOR 3
    CH3 : MOTOR 4
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE `

Betaflight 4.3.1 : 

`# dma show

Currently active DMA:
--------------------
DMA1 Stream 0: MOTOR 2
DMA1 Stream 1: MOTOR 1
DMA1 Stream 2: LED_STRIP
DMA1 Stream 3: MOTOR 3
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: MOTOR 4
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# timer show

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH3 : MOTOR 1
TIM3: FREE
TIM4:
    CH1 : MOTOR 2
    CH2 : MOTOR 3
    CH3 : MOTOR 4
TIM5:
    CH1 : LED_STRIP
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE`

Hoping it will help.

SteveCEvans, 2022-08-11:
Turn off LED_STRIP. You weren’t using it with 4.3.0.

LoakAudio, 2022-08-11:
Hi @SteveCEvans, as I'm using the same backup with BF 4.3.0 and 4.3.1, LED_STRIP is activated by default in 4.3.1

LoakAudio, 2022-08-12:
@FabianIsensee @SteveCEvans I confirm that disabling LED_STRIP in the BF Configurator solves the problem.

As it is the update to BF 4.3.1 that activates the LED_STRIP, not the user, can it be disabled by default on BF side ?

LoakAudio, 2022-08-12:
Thank you gentlemen.

FabianIsensee, 2022-08-16:
Thanks everyone :-)

closed 2022-08-16 by FabianIsensee

cynfewl, 2022-11-16:
I ran across this bug on my DiamondF4 ELRS with built in LEDs and the drone dropping out of the sky, even at close range.

Is there any way to have ELRS SPI and LEDs at the same time, because I would like to run the LEDs for night flying?

I am happy to wait for 4.4 if this is fixed in that release.

lesharodin, 2023-01-20:
UP @haslinghuis 
Is enabling LED strip with ELRS SPI will be allowed in 4.4?