betaflight/betaflight #11763: Frsky_X Link on SPI unstable since 4.3.x on CrazyF411 AIO F4 FC

onkelalex, 2022-07-25:
### Describe the bug

Hello,

I am using a Happymodel CrazyF411 AIO F4FC (http://www.happymodel.cn/index.php/2020/10/28/crazyf411-aio-f4-flight-controller-built-in-20a-esc-osd-frsky-receiver-for-toothpick/) with latest BF4.3.1.

Conencted the buildin FrSky X receiver to a Jumper T-Pro 4in1 board.

Binding is successful, but connection is lost after round about 1 minute. Then it circles between telemetry lost/ telemetry recoverd. Stickmoves are not recognized smoothly (the bars are moving in big steps after a stick was moved)

Other FrSky Receiver are acting normally. With BF4.2.11 connection is stable.

### To Reproduce

Bind Jumper T-Pro with AIO to CrazyF411 FC with BF >=4.3.0

### Expected behavior

Stable connection between Transmitter and Receiver

### Diff configuration

```txt
# diff

# version
# Betaflight / STM32F411 (S411) 4.3.1 Jul 13 2022 / 03:32:52 (8d4f005) MSP API: 1.44
# config: manufacturer_id: MTKS, board_name: MATEKF411RX, version: edcd244b, date: 2022-06-29T02:24:46Z

# start the command batch
batch start

board_name MATEKF411RX
manufacturer_id MTKS

# feature
feature -RX_PARALLEL_PWM
feature TELEMETRY

# master
set rc_smoothing = OFF
set frsky_spi_tx_id = 102,89,2
set frsky_spi_offset = -20
set frsky_spi_bind_hop_data = 6,25,44,63,82,101,120,139,158,177,196,215,234,18,37,56,75,94,113,132,151,170,189,208,227,11,30,49,68,87,106,125,144,163,182,201,221,4,23,42,61,80,99,118,137,156,175,0,0,0
set gyro_1_align_yaw = 1800

profile 0

rateprofile 0
```

### Resources configuration

```txt
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: GYRO_EXTI
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: RX_SPI_CC2500_TX_EN
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: RX_SPI_CC2500_LNA_EN
A14: RX_SPI_CC2500_ANT_SEL
A15: RX_SPI_CS
B00: ADC_BATT
B01: ADC_CURR
B02: RX_SPI_BIND
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: FREE
B07: FREE
B08: FREE
B09: LED
B10: FREE
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: RX_SPI_EXTI
C15: BEEPER
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Flight controller

Happymodel CrazyF411 AIO (MATEKF411RX)

### Other components

Jumper T-Pro/T-Lite with 4in1 multimodul, actual release

moparmaddnessman, 2022-08-04:
I ran into the same bug with the crazybee x12

RPassanha, 2022-08-11:
The same bug with the F4 FC and TBS Tracer.

Islefan00, 2022-08-17:
I'm still having this issue with my FC. Turned telemetry off, running in FrSky_D, and my PID values are set to 4Mhz. Was this issue resolved?

ChadyG, 2022-09-22:
Same behavior on a Tinyhawk II STM32F411.  Receiver connection unstable on 4.3.1, but good on 4.2.11.

howels, 2022-09-27:
Can you please test the nightly builds and see if this issue persists? I am running the 4.4 nightly and cannot reproduce on a CrazyBee F4 Lite with FrSky_D and telemetry off. I am starting with RSSI ~75 and flying around with RSSI 30-40 without link dropouts which is actually better than I used to see on 4.2.
Please check that blackbox is set to "NONE" to ensure your CPU is not overloaded - noticed the default is "SERIAL" in recent builds. I am running 4K PID loop with DSHOT300 and RPM filters enabled.

howels, 2022-09-27:
Tested nightly with FrSky-D, FrSky-X (telem on and off) and FrSky-X2.

FrSky-X and FrSky-X2 bind but drop link very quickly at around 50-60 RSSI - TBH this is the same behaviour as I have seen on most previous versions of BetaFlight (see https://www.rcgroups.com/forums/showthread.php?3244881-Dropouts-Lockups-In-D16-Mode-With-Telemetry-On-In-SPI-Receiver-Flight-Controllers).

FrSky-D works reliably down to 30 RSSI, and supports voltage telemetry (do not even need to enable the telemetry feature for voltage) so I suggest that you use FrSky-D for now.

@onkelalex please confirm the results on FrSky-D.

onkelalex, 2022-09-27:
Hi @howels,

I will be able to test on the upcoming weekend. Will report as soon as possible.

Best regards
Alex

aQuaplug, 2022-10-04:
4.3.0 is little better than 4.3.1 to me.
I tested multiple times with 4.3.0 and 4.3.1 with CrazybeeF4FR (1S Whoop),
and Crazyf4 pro 12A diversity SPI receiver (MATEKF411RX target)

radio - Horus X12S / X9 Lite S
I used redpine protocols before, and I just want do not swap external module if X2 or X is acceptably flying.
and finally X2 for SPI bf4.3, I want to flashing my XJT(ISRM) to new versions.
but result is  not so good. I flashed internal XJT module multiple times between v1 and v2 for testing.

Frsky-X2 - poor range both 4.3.0 and 4.3.1, connection unstable with 4.3.1
Frsky-X - not so good range with 4.3.0 / poor range with 4.3.1, connection unstable with 4.3.1
Redpine - range is much better even in redpine fastmode with 4.3.1

so I downgrade all SPI receiver based FC from 4.3.1 to 4.3.0

all ACCESS radios do not have D option :( even only few has X1 option.
It would very happy to me X2 for SPI is flyable :)

rickgode, 2022-11-03:
Came here to say the same thing. On a URUAV UZ80 on BF 4.3.1 I cannot get a stable connection with my FRSKY_D built-in Rx.  On 4.3 it's fine, but I can reproduce the constant "telemetry lost" by upping the PID loop from 4kHz to 8kHz.  I downgraded to BF 4.3 with a 4kHz PID loop and that seems to have a stable Rx connection.

aQuaplug, 2022-11-15:
TinyHawk II (MATEK F411RX) BF4.3.0 also lost signal within 10meters.
So sad... not just 4.3.1 but also 4.3.0

howels, 2022-11-16:
I've seen this since 4.0, recommend FrSky-D for anyone using this hardware.

closed 2022-12-24 by github-actions[bot]

KarateBrot, 2022-12-24:
Try 4.3.2 or 4.4.0

onkelalex, 2022-12-24:
Can confirm that a stable link on 4.3.2 with FrSky_D is possible 👍 

(Multi 4in1 Module, V1.3.3.20 AETR; FrSky D8).

Thanks for you support guys

rickgode, 2022-12-24:
Yay!  4.3.2 works on my URUAV UZ80 with the CRAZYBEEF4FR board.

(Multi 4in1 Module, V1.3.3.20 AETR; FrSky D8)

Ricky

TheStapl3r, 2022-12-30:
Im having the same issue with the jumper t pro with happymodel x12 frsky board betaflight 4.3.2.  It works fine with Emax controller which came with tinyhawk 3.
Ive tried d8 and X. Just isn't reliable.