betaflight/betaflight #11766: Decimal motor output limit

LeadFingers, 2022-07-27:
### Is your feature request related to a problem? Please describe

motor output limit can only be a whole integer number.  When trying to KV match motors we frequently have motors that need ex: 88.7% motor output limit. We are currently forced to make those pilots leave 0.7% of power on the table.  Being able to set their motor output limit to the first decimal place would reduce this to within the error bars of our testing methods.

### Describe the solution you'd like

allow the motor output limit number to be set to the first decimal place. XY.Z% instead of just XY%

### Describe alternatives you've considered

our current method is to always round down to the nearest integer % motor output limit

### Other information

This feature would be useful for all spec racing that allows KV matching with motor output limit.

daleckystepan, 2022-07-27:
So simply round up. There is no need to make it with such precision.

LeadFingers, 2022-07-27:
To make the racing as fair as physically possible we cant allow ANY pilots to have more power than the "ideal" spec motor (1250KV).  So by allowing some pilots to round up we are explicitly giving them up to 1% more power than their competitors.  I would agree with you if we were just racing for fun on the weekends, but when we are having big races with significant sums of money on the line a 1% advantage is to much to ignore.

daleckystepan, 2022-07-27:
I don't understand it perfectly but I am not sure if it is possible to compensate kV by motor output limit with such precision. Even if we allow decimals.

LeadFingers, 2022-07-27:
We have had a lot of success normalizing motors by using the existing motor output limit.  In general you are correct that an exact KV match is not possible but we have found that is mostly due to how different ESC's and motor stator configurations produce non linear power curves.  We are transitioning to an under load RPM/power limit and designing tests to measure that for this reason.  Sayin that we are "KV matching" is not strictly true anymore but it is an easy short hand for what we are doing.  This feature would allow many different manufacturers to compete in spec racing without forcing them to go through the expensive R&D process of making a new motor specifically for one racing league and is something pilots have been frequently asking for at Street League events.

sugaarK, 2022-07-27:
Manufacturers love making new motors so I don’t see any issue there… I agree with @daleckystepan  what your asking for is unnecessary. Don’t forget what’s printed on the motor is only within 5-10% of the KV your motor actually has

sugaarK, 2022-07-27:
The only way to make a spec event truly ‘fair’ is to use only one make and model of motors

LeadFingers, 2022-07-27:
I'm aware that the KV on the side of the motor is mostly just marketing.  We have forked the betaflight configurator and turned it into a motor KV testing tool by using the bi directional Dshot RPM values and a modified motors tab.  We have tested the results from the field tool against Oscilloscope measurements and have found it to be be accurate to within tenths of a KV.  At check in every pilot has their quad run through this tool and it finds the actual KV of each motor and has their motor output limit set for the event.  Even if every pilot uses the "same" motor manufacturing differences can lead to significant KV differences, especially when motors have been through multiple production runs.  The best tool we currently have for spec racing is motor output limiting and we have found that our ability to measure what motor output the quad should have is more precise than what motor output limits are currently available to us.

ImperiusFPV, 2022-07-27:
As a competitive racer, this feature is actually pretty important to me. It doesn't sound important, but .7 percent more RPM is pretty important to the most competitive racers. If there was a manufacturer who could produce enough motors for everyone to use who is competing, that would be awesome, but that's not our current reality. Motor output limiting is the only thing we have to make the playing field  level with different motors, and we are testing true RPM not sticker KV. If everyone could have the same max RPM with a motor output limit to a .1% margin, myself and many other racers would be much happier.

skyfpv, 2022-07-31:
Motor output limit should be an integer between 1000 and 2000 in my opinion. It represents the maximum command allowed by the ESC. There is no reason to limit it to the somewhat arbitrary resolution of 100.

closed 2022-09-01 by haslinghuis