betaflight/betaflight #11797: Allow easy exporting of Filter settings

evilC, 2022-08-14:
### Is your feature request related to a problem? Please describe

No

### Describe the solution you'd like

As a User
I want to be able to export filter settings
So that I can easily transfer them from one quad to another

Currently, there is no filter equivalent to `dump profile` - suggest adding `dump filters`
With profiles, even without using the `dump profile` command, I could do a `diff all`, then just navigate to the `#Profile 1` section and copy it somewhere, whereas with filters, they are just munged in with othet settings in the `#Master` section. (I presume? Maybe some are somewhere else?)
So at the very least, if we could just have a `#Filters` section, we may not even need a `dump filters` command

### Describe alternatives you've considered

Doing a `diff all` and combing through the `#Master` section trying to find which lines relate to filters, but it's not exactly great UX and user error is certainly a possibility.

zvonler, 2023-09-06:
I looked at the discussion in discord around the time this ticket was created: https://discord.com/channels/868013470023548938/924169080045445120/1008499259844337734

It sounds like it would be helpful if the output of any dump (and diff?) command was aware of the pages/tabs in the configurator enough to group options together on that basis.  E.g. if the filter tab settings always ended up in one "block" of the dump/diff output, it would be easy to inspect/copy them. Does that sound like a useful enough change to be worth pursuing?

evilC, 2023-09-06:
> Does that sound like a useful enough change to be worth pursuing?

Yeah, that would probably suffice IMHO

closed 2026-09-26 by blckmn