betaflight/betaflight #11800: HGLRC FC Zeus f722 mini - upgrade from 2.9/2.11 to 3.1 makes baro don't work and USB connection unstable.

porlock, 2022-08-15:
### Describe the bug

After upgrading my HGLRC FC Zeus f722 mini from  2.9/2.11 to 3.1 i noticed that barometer don't work, also usb connection was unstable, after 1-5 min bf configuratore freezes and i have to replug usb cabble.  I did flashing with full chip erease and applying custom defaults. Realy strange is that after i revert to 2.9 (with full chip erease and custom defaults) it did not solve the issue. But applying my full cli dump did fix it (diff also not worked), baro start work again and USB did not freeze anymore. Its not a coidence, i did full process twice with same resoults.  Please find my full dump that fix isue after reverting to 2.9
[dump_all_BTFL_cli_DOJO_20220812_232746.txt](https://github.com/betaflight/betaflight/files/9340591/dump_all_BTFL_cli_DOJO_20220812_232746.txt)

### To Reproduce

-Install BF 3.2 on HGLRC FC Zeus f722 min  (with full chip erease and custom defaults) 
-check that barometer is not working 
-use bf configurator for until it freezes
- revert firmware to 2.9/2.10 (with full chip erease and custom defaults) 
-check that barometer is not working 
-use bf configurator for until it freezes
- apply my full config dump
- check that barometer is start working again
- bf configurator dont freezes anymore

### Expected behavior

Working barometer
stable usb connection

### Diff configuration

```txt
# 

# diff all

# version
# Betaflight / STM32F7X2 (S7X2) 4.2.9 Apr 27 2021 / 19:34:29 (e097f4ab7) MSP API: 1.43
# config: manufacturer_id: HGLR, board_name: HGLRCF722, version: a074838b, date: 2021-06-14T21:36:54Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name HGLRCF722
manufacturer_id HGLR
mcu_id 001300204e53500f2030304e
signature 

# name: DOJO

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature TELEMETRY
feature OSD

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# beacon
beacon RX_SET

# serial
serial 3 1 115200 57600 0 115200

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 1 1300 1700 0 0
aux 2 13 2 1700 2100 0 0
aux 3 35 1 1700 2100 0 0
aux 4 36 2 1300 1700 0 0

# master
set gyro_lowpass2_hz = 300
set dyn_notch_width_percent = 0
set dyn_notch_q = 200
set dyn_notch_min_hz = 90
set dyn_notch_max_hz = 350
set dyn_lpf_gyro_min_hz = 240
set dyn_lpf_gyro_max_hz = 600
set acc_calibration = 32,42,-102,1
set rssi_channel = 15
set rc_interp = MANUAL
set rc_interp_int = 5
set rc_smoothing_type = INTERPOLATION
set serialrx_provider = CRSF
set dshot_bidir = ON
set motor_poles = 12
set current_meter = NONE
set report_cell_voltage = ON
set osd_rssi_pos = 2456
set osd_flymode_pos = 2532
set osd_current_pos = 36
set osd_craft_name_pos = 2543
set osd_warnings_pos = 329
set osd_avg_cell_voltage_pos = 2486
set osd_disarmed_pos = 2348
set gyro_1_sensor_align = CW0
set gyro_1_align_pitch = 0
set gyro_1_align_yaw = 0
set name = DOJO

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 84
set dyn_lpf_dterm_max_hz = 204
set dterm_lowpass2_hz = 180
set pidsum_limit = 1000
set pidsum_limit_yaw = 1000
set p_pitch = 64
set d_pitch = 36
set f_pitch = 133
set p_roll = 59
set d_roll = 33
set f_roll = 126
set p_yaw = 63
set f_yaw = 126
set d_min_roll = 0
set d_min_pitch = 0
set ff_interpolate_sp = ON

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set rates_type = ACTUAL
set roll_rc_rate = 20
set pitch_rc_rate = 20
set yaw_rc_rate = 20
set roll_expo = 65
set pitch_expo = 65
set yaw_expo = 65
set roll_srate = 81
set pitch_srate = 81
set yaw_srate = 81

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
#
```

### Resources configuration

```txt
# 

# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: SERIAL_TX 4
A01: SERIAL_RX 4
A02: FREE
A03: FREE
A04: BARO_CS
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: LED 2
A14: LED 1
A15: FREE
B00: MOTOR 3
B01: MOTOR 4
B02: GYRO_CS 1
B03: FREE
B04: MOTOR 1
B05: MOTOR 2
B06: FREE
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: FREE
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: PINIO 1
C09: PINIO 2
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: BEEPER
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FLASH_CS
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 2
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: FREE
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

#
```

### Flight controller

HGLRC FC Zeus f722 mini

haslinghuis, 2022-08-16:
2.9 / 2.11 / 3.1 ??? We only support BF >= 4,0.

porlock, 2022-08-16:
@haslinghuis yes, of course, i missed 4 don't know why, sorry i already corrected my initial description and tittle.

haslinghuis, 2022-08-16:
@porlock you might try `HGLRCF722E`.

See also: https://github.com/betaflight/betaflight/issues/11674 (are we experiencing different revisions of the board?)

ojwright845, 2022-09-01:
I have several of these boards. I had this problem during RC period, but not with 4.3.1. Just type the below into cli.
set baro_bustype = SPI
set baro_spi_device = 1
set baro_hardware = DPS310

haslinghuis, 2022-09-01:
@ojwright845 looks like the board needs `HGLRCF722E` target but it's missing `set baro_hardware = DPS310` indicating `AUTO` is not working.

porlock, 2022-09-20:
@haslinghuis , HGLRCF722E solved the issue in my case. thanks

closed 2022-09-20 by porlock