betaflight/betaflight #11803: FOXEERF722V3 Betaflight 4.3.1 Servo output not working

jonasbouchraiet, 2022-08-19:
### Describe the bug

I want to use a Servo to stabilize the roll axis for the Gopro on my 7". In order to improve the FOV in turns for reelsteady go.

I checked if that would work by flashing the board, doing the necessary configuration and connecting a battery. 

But the Servo does not power on.

I checked if the servo is dead, but it works with a servo tester just fine.

### To Reproduce

Flash 4.3.1 Full chip erase

apply custom defaults

cal acc

enable servo_tilt in Configuration

turn on camstab mode

change motor protocol to Dshot300

in CLI:

resource led_strip 1 none
resource servo 1 A15

save

enable live mode in Servo tab

### Expected behavior

To my understanding the servo should center at 1500 at this point and I should feel some resistance

### Diff configuration

```txt
# diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.3.1 Jul 13 2022 / 03:36:10 (8d4f005) MSP API: 1.44
# config: manufacturer_id: FOXE, board_name: FOXEERF722V3, version: 66f6d4df, date: 2022-04-18T09:47:43Z

# start the command batch
batch start

board_name FOXEERF722V3
manufacturer_id FOXE

# resources
resource SERVO 1 A15
resource LED_STRIP 1 NONE

# feature
feature SERVO_TILT

# aux
aux 0 8 0 1300 1700 0 0

# master
set acc_calibration = -6,-18,51,1
set motor_pwm_protocol = DSHOT300
set gyro_1_align_yaw = 1800

profile 0

rateprofile 0

# end the command batch
batch end
```

### Resources configuration

```txt
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: FREE
A04: BEEPER
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: MOTOR 2
A09: MOTOR 1
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: SERVO 1
B00: FREE
B01: FREE
B02: GYRO_CS 1
B03: CAMERA_CONTROL
B04: FREE
B05: SPI_MOSI 3
B06: FREE
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: FREE
B11: FREE
B12: FLASH_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: ADC_BATT
C01: FREE
C02: ADC_CURR
C03: OSD_CS
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: MOTOR 4
C09: MOTOR 3
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: FREE
C13: FREE
C14: FREE
C15: LED 1
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH1 : SERVO 1
    CH2 : CAMERA_CONTROL
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH3 : DSHOT_BITBANG 1
    CH4 : DSHOT_BITBANG 3
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: DSHOT_BITBANG 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: DSHOT_BITBANG 3
```

### Flight controller

FOXEERF722V3

### Other components

Servo is an EMAX ES3054

### How are the different components wired up

Servo RED to 5V Servo Black to GND and Servo Yellow to LED.

GND and VCC to GND Battery and 3s Positive.

[image]

jonasbouchraiet, 2022-08-19:
[image]

haslinghuis, 2022-08-19:
You probably need to power the servo using a bec  directly from power source. Only signal wire should be coming from flight controller.

Rclover106, 2022-08-22:
So I ironically just ran into this issue tonight. Same flight controller, I am using a savox hv servo being powered by a BEC. I tried multiple output points, led motor output 5,6,7,8 etc and nothing caused the servo to react. In betaflight configurator it shows the servo pwm reacting as it should( I'm using a headtracker) but the servo itself is not receiving the command.

Rclover106, 2022-08-24:
> [image]

Alright I figured out the issue and it works now. If you look at your resource dump, you can scroll down and see "currently active timers". If you look at all of them timer 2 is being shared between the servo and cam control. based on an older article I found https://github.com/betaflight/betaflight/wiki/Servos-&-SERVO_TILT-for-3.1

If you read into this it says that the timers for servos cannot be shared with anything, they must be only servos in order for it to work, and you cannot remap timers, so you have to figure out what pad you can use.  Most FC's only have the led strip pad on its own timer which is why it works on others peoples builds. 

Here is how the timers are mapped on the f722 V3 FC.
TIM1:
A08------motor 2
A09------motor 1
TIM2:
A15------led strip 1
B03------Camera_control 1
TIM4:
B07------PPM 1
TIM8:
C06------Serial_tx 6 
C07------Serial_rx 6
C08------motor 4
C09------motor 3

This leaves us with two options.
you can either remap the servo to the ppm pad on the FC, and since it is on its own individual timer it should function fine, or you can use the led strip pad, as I did, but unmap the camera control pad to be nothing ( or technically another servo because it can use the same timer as another servo) The command for this in the CLI is :
resource camera_control 1 none
save

and then the servo will function correctly. Hopefully I can save you and future people the hair-pulling frustration I experienced.

jonasbouchraiet, 2022-08-24:
I can confirm this fixes the issue. The servo is working with Camera Control 1 assigned to none or another servo. Thanks so much for your help!!!

closed 2022-08-24 by jonasbouchraiet