betaflight/betaflight #11809: DPS310 bad readings over time.

4stah, 2022-08-25:
### Describe the bug

I have a problem with the constant baro DPS310 readings. After connecting, the reading continues to increase from eg. 0.2 up to eg 3.0 within 20-30s. and so on. I observe the same behavior (exact) on the two SucceX-D F7 V2.2 TwinG and Beast F7 45A V2 AIO boards. The result is that when I take off, the baro shows zero (after arming) and after landing 5min later, it shows 9-11m.

Tested many times, with different baro settings (eg. lpf), in stable enviroement - no result so far.

Many thanks for any help or suggestion at least.

### To Reproduce

connect any of mentioned boards with DPS310 and look at the sensor readings in Betaflight configurator.

### Expected behavior

constant (more or less) readings when craft lays on stable surface.

### Diff configuration

```txt
# version
# Betaflight / STM32F745 (S745) 4.3.1 Jul 13 2022 / 03:35:21 (8d4f005) MSP API: 1.44
# config: manufacturer_id: IFRC, board_name: IFLIGHT_F745_AIO, version: edcd244b, date: 2022-06-29T02:24:46Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name IFLIGHT_F745_AIO
manufacturer_id IFRC
mcu_id 003800353135511331303738
signature 

# name: Bumblebee HD V3

# feature
feature -RX_PARALLEL_PWM
feature -ESC_SENSOR

# serial
serial 0 1 115200 57600 0 115200
serial 1 0 115200 57600 0 115200
serial 2 64 115200 57600 0 115200
serial 5 0 115200 57600 0 115200
serial 6 0 115200 57600 0 115200

# beeper
beeper -GYRO_CALIBRATED
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -GPS_STATUS
beeper -READY_BEEP
beeper -SYSTEM_INIT
beeper -BLACKBOX_ERASE
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE

# beacon
beacon RX_LOST
beacon RX_SET

# led
led 0 0,0:SW:L:2
led 1 1,0:NW:C:0
led 2 2,0:ES:L:2
led 3 3,0:NE:C:0

# color
color 8 240,0,255

# mode_color
mode_color 6 0 0
mode_color 6 1 1

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 1 900 1300 0 0
aux 2 2 1 1300 1700 0 0
aux 3 13 3 1600 2100 0 0
aux 4 15 6 900 1400 0 0
aux 5 35 4 1700 2100 0 0

# adjrange
adjrange 0 0 2 900 2100 12 2 0 0

# master
set acc_trim_pitch = -8
set acc_trim_roll = 36
set acc_calibration = -6,-19,12,1
set mag_hardware = NONE
set baro_i2c_address = 118
set baro_hardware = DPS310
set rssi_channel = 12
set fpv_mix_degrees = 20
set serialrx_provider = CRSF
set blackbox_device = NONE
set dshot_bidir = ON
set failsafe_procedure = AUTO-LAND
set bat_capacity = 1500
set ibata_scale = 113
set beeper_dshot_beacon_tone = 2
set yaw_motors_reversed = ON
set small_angle = 180
set osd_vbat_pos = 374
set osd_rssi_pos = 2497
set osd_link_quality_pos = 417
set osd_rssi_dbm_pos = 247
set osd_flymode_pos = 2048
set osd_current_pos = 325
set osd_mah_drawn_pos = 2533
set osd_craft_name_pos = 32
set osd_gps_speed_pos = 498
set osd_gps_lon_pos = 305
set osd_gps_lat_pos = 337
set osd_gps_sats_pos = 26
set osd_home_dir_pos = 46
set osd_home_dist_pos = 13
set osd_flight_dist_pos = 57
set osd_compass_bar_pos = 395
set osd_altitude_pos = 2537
set osd_warnings_pos = 2250
set osd_avg_cell_voltage_pos = 2516
set osd_battery_usage_pos = 453
set osd_nheading_pos = 14
set osd_nvario_pos = 2542
set osd_stat_tim_2 = OFF
set osd_stat_max_dist = ON
set osd_stat_endbatt = ON
set osd_stat_max_alt = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set name = Bumblebee HD V3
set position_alt_source = BARO_ONLY

profile 0

# profile 0
set vbat_sag_compensation = 100

profile 1

profile 2

# profile 2
set dterm_lpf1_static_hz = 150
set p_pitch = 66
set i_pitch = 99
set d_pitch = 47
set p_roll = 63
set i_roll = 94
set d_roll = 45
set p_yaw = 53
set i_yaw = 99

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set roll_rc_rate = 16
set pitch_rc_rate = 16
set yaw_rc_rate = 20
set roll_expo = 60
set pitch_expo = 60
set yaw_expo = 60
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 80
set throttle_limit_type = SCALE
set throttle_limit_percent = 80

rateprofile 1

# rateprofile 1
set roll_rc_rate = 15
set pitch_rc_rate = 15
set yaw_rc_rate = 20
set roll_expo = 76
set pitch_expo = 76
set yaw_expo = 64
set roll_srate = 91
set pitch_srate = 91
set yaw_srate = 80
set throttle_limit_type = SCALE
set throttle_limit_percent = 80

rateprofile 2

# rateprofile 2
set roll_rc_rate = 20
set pitch_rc_rate = 20
set yaw_rc_rate = 25
set roll_expo = 76
set pitch_expo = 76
set yaw_expo = 64
set roll_srate = 91
set pitch_srate = 91
set yaw_srate = 80
set throttle_limit_type = SCALE
set throttle_limit_percent = 80

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 1

# save configuration
```

### Resources configuration

```txt
resource BEEPER 1 B04
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 C09
resource MOTOR 4 C08
resource PPM 1 A03
resource LED_STRIP 1 B06
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 6 C06
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 A01
resource SERIAL_RX 6 C07
resource INVERTER 2 C13
resource LED 1 B05
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 C10
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 C11
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource SPI_MOSI 3 C12
resource CAMERA_CONTROL 1 A00
resource ADC_BATT 1 C02
resource ADC_CURR 1 C01
resource PINIO 1 C15
resource PINIO 2 C14
resource FLASH_CS 1 A15
resource OSD_CS 1 B12
resource GYRO_EXTI 1 C04
resource GYRO_CS 1 A04
resource USB_DETECT 1 C05
```

### Flight controller

iFlight Beast F7 45A V2

### Other components

 Caddx Vista, TBS Crossfire nano

### How are the different components wired up

standard wiring done by iFlight BumbleBee V2

RipperGOT, 2022-09-15:
I can confirm this.

4stah, 2022-10-15:
Hm, I think that this problem is not only important thad we have wrong readings in OSD but will affect RTH too.

RipperGOT, 2022-10-15:
> Hm, I think that this problem is not only important for wrong readings in OSD but will affect RTH too.

Not only for RTH. In Europe they are quite strict with altitude restrictions and this helps not to exceed it.

closed 2022-10-17 by KarateBrot