betaflight/betaflight #11815: MSP VTX HDZero osd appears incorrectly, disappears

qwerfpv, 2022-08-31:
### Describe the bug

the LQ is displayed incorrectly, incompletely. the battery display disappears when the limit is reached

### To Reproduce

F411 bf 4.4.0 msp vtx version flash

### Expected behavior

no problem on f722 fc. the osd items appear fine

### Diff configuration

```txt
# diff all
###WARNING: NO CUSTOM DEFAULTS FOUND###

# version
# Betaflight / STM32F411 (S411) 4.4.0 Aug  8 2022 / 16:38:56 (1b723b4) MSP API: 1.45
###ERROR IN diff: NO CONFIG FOUND###
# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name JHEF411
manufacturer_id JHEF
mcu_id 006c00423535510836323431
signature 

# resources
resource BEEPER 1 C14
resource MOTOR 1 A08
resource MOTOR 2 A09
resource MOTOR 3 A10
resource MOTOR 4 B00
resource MOTOR 5 B04
resource PPM 1 A02
resource PWM 1 B01
resource PWM 2 A03
resource LED_STRIP 1 A15
resource SERIAL_TX 1 B06
resource SERIAL_TX 2 A02
resource SERIAL_RX 1 B07
resource SERIAL_RX 2 A03
resource I2C_SCL 1 B08
resource I2C_SDA 1 B09
resource LED 1 C13
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource CAMERA_CONTROL 1 B10
resource ADC_BATT 1 A00
resource ADC_RSSI 1 B01
resource ADC_CURR 1 A01
resource PINIO 1 B05
resource FLASH_CS 1 B02
resource OSD_CS 1 B12
resource GYRO_EXTI 1 B03
resource GYRO_CS 1 A04
resource USB_DETECT 1 C15

# timer
timer A02 AF3
# pin A02: TIM9 CH1 (AF3)
timer A08 AF1
# pin A08: TIM1 CH1 (AF1)
timer A09 AF1
# pin A09: TIM1 CH2 (AF1)
timer A10 AF1
# pin A10: TIM1 CH3 (AF1)
timer B00 AF2
# pin B00: TIM3 CH3 (AF2)
timer B04 AF2
# pin B04: TIM3 CH1 (AF2)
timer B01 AF2
# pin B01: TIM3 CH4 (AF2)
timer A03 AF2
# pin A03: TIM5 CH4 (AF2)
timer B10 AF1
# pin B10: TIM2 CH3 (AF1)
timer A15 AF1
# pin A15: TIM2 CH1 (AF1)

# dma
dma ADC 1 1
# ADC 1: DMA2 Stream 4 Channel 0
dma pin A08 1
# pin A08: DMA2 Stream 1 Channel 6
dma pin A09 1
# pin A09: DMA2 Stream 2 Channel 6
dma pin A10 0
# pin A10: DMA2 Stream 6 Channel 0
dma pin B00 0
# pin B00: DMA1 Stream 7 Channel 5
dma pin B04 0
# pin B04: DMA1 Stream 4 Channel 5
dma pin B01 0
# pin B01: DMA1 Stream 2 Channel 5
dma pin A03 1
# pin A03: DMA1 Stream 3 Channel 6
dma pin B10 0
# pin B10: DMA1 Stream 1 Channel 3
dma pin A15 0
# pin A15: DMA1 Stream 5 Channel 3

# feature
feature TELEMETRY
feature OSD

# serial
serial 0 131073 115200 57600 0 115200
serial 1 64 115200 57600 0 115200

# beacon
beacon RX_SET

# aux
aux 0 0 0 1700 2100 0 0
aux 1 13 1 1700 2100 0 0
aux 2 35 2 1400 2100 0 0

# adjrange
adjrange 0 0 3 900 2100 29 3 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY    0    0    0    0    0    0    0    0
vtxtable band 2 BOSCAM_B B FACTORY    0    0    0    0    0    0    0    0
vtxtable band 3 BOSCAM_E E FACTORY    0    0    0    0    0    0    0    0
vtxtable band 4 FATSHARK F FACTORY    0 5760    0 5800    0    0    0    0
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I FACTORY    0    0    0    0    0    0    0    0
vtxtable powerlevels 3
vtxtable powervalues 14 23 0
vtxtable powerlabels 25 200 0

# master
set dyn_notch_count = 0
set acc_lpf_hz = 10
set acc_calibration = -3,23,48,1
set mag_bustype = I2C
set mag_i2c_device = 1
set mag_hardware = NONE
set baro_bustype = I2C
set baro_i2c_device = 1
set baro_hardware = NONE
set serialrx_provider = CRSF
set blackbox_device = SPIFLASH
set dshot_idle_value = 300
set dshot_burst = ON
set motor_pwm_protocol = DSHOT600
set align_board_yaw = 180
set current_meter = ADC
set battery_meter = ADC
set ibata_scale = 170
set beeper_inversion = ON
set beeper_od = OFF
set beeper_dshot_beacon_tone = 3
set small_angle = 180
set gps_rescue_alt_buffer = 15
set gps_rescue_initial_alt = 50
set gps_rescue_descent_dist = 200
set gps_rescue_landing_dist = 10
set gps_rescue_ground_speed = 2000
set gps_rescue_throttle_p = 150
set gps_rescue_throttle_i = 20
set gps_rescue_throttle_d = 50
set gps_rescue_velocity_p = 80
set gps_rescue_velocity_d = 15
set gps_rescue_yaw_p = 40
set gps_rescue_descend_rate = 150
set gps_rescue_throttle_hover = 1280
set gps_rescue_sanity_checks = RESCUE_SANITY_ON
set gps_rescue_min_dth = 100
set osd_warn_bitmask = 57343
set osd_vbat_pos = 14422
set osd_link_quality_pos = 14403
set osd_link_tx_power_pos = 418
set osd_rssi_dbm_pos = 14371
set osd_tim_2_pos = 14775
set osd_vtx_channel_pos = 12706
set osd_battery_usage_pos = 2505
set osd_flip_arrow_pos = 6378
set osd_stat_bitmask = 8519908
set osd_displayport_device = MSP
set system_hse_mhz = 8
set vtx_band = 5
set vtx_channel = 1
set vtx_power = 2
set vtx_freq = 5658
set max7456_spi_bus = 2
set displayport_msp_serial = 0
set flash_spi_bus = 2
set gyro_1_bustype = SPI
set gyro_1_spibus = 1
set gyro_1_sensor_align = CW180
set gyro_1_align_yaw = 1800
set gyro_2_spibus = 1

profile 0

# profile 0
set d_pitch = 52
set d_roll = 46
set d_min_roll = 34
set d_min_pitch = 39
set simplified_d_gain = 115

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
#
```

### Resources configuration

```txt
resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: ADC_BATT
A01: ADC_CURR
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: MOTOR 1
A09: MOTOR 2
A10: MOTOR 3
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FREE
B00: MOTOR 4
B01: FREE
B02: FLASH_CS
B03: GYRO_EXTI
B04: FREE
B05: PINIO 1
B06: SERIAL_TX 1
B07: SERIAL_RX 1
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: CAMERA_CONTROL
B11: FREE
B12: PREINIT 3
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: BEEPER
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1 : MOTOR 1
    CH2 : MOTOR 2
    CH3 : MOTOR 3
TIM2:
    CH3 : CAMERA_CONTROL
TIM3:
    CH3 : MOTOR 4
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: TIMUP 3
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: ADC 1
DMA2 Stream 5: TIMUP 1
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

#
```

### Flight controller

jhemcu GHF411 PRO-BMI, https://www.jhemcu.com/e_productshow/?48-GHF411-PRO-BMI-40A-Toothpick-F4-2-6S-AIO-BRUSHLESS-Flight-Controller-48.html

### Other components

hdzero whoop vtx mspvtx beta fw, jhemcu elrs sp24s v3.0 rc2

### Add any other context about the problem that you think might be relevant here

[image]
[image]
[image]
[image]
[image]
[image]

P-I-Engineer, 2022-09-12:
having same issue, stacking has been the only way i've got it to work, and turning down all alarms, so blinking related as well?
[image]
[image]
[image]

haslinghuis, 2022-09-12:
Please test with latest firmware development version. The configurator parts are awaiting approval.

P-I-Engineer, 2022-09-12:
this is with flying on master from yesterday (61c17a9)

haslinghuis, 2022-09-12:
@P-I-Engineer https://github.com/betaflight/betaflight/pull/11705 has been merged now. Please verify with this build.

qwerfpv, 2022-09-12:
re. I noticed that the osd does not come back even after changing the battery, only if I restart the vrx as well. I don't even understand how this is possible :/

P-I-Engineer, 2022-09-12:
> @P-I-Engineer #11705 has been merged now. Please verify with this build.

not fixed in this pr either.

i was flying 4.3.1 with elrs 3 and mspvtx merged and no osd issues.

on 4.4/master the osd issues are present.

haslinghuis, 2022-09-12:
@phobos- could you please test LQ (also the comment of having to restart VRX)

P-I-Engineer, 2022-09-12:
this appears to be an msp display issue or osd in general issue, either of those change since 4.3.1?

P-I-Engineer, 2022-09-13:
i just flashed latest development and all is good.

osd looked correct, did not disappear, no overlay clashing

[image]

qwerfpv, 2022-09-13:
re. I tested it too. it works fine now :) thx
[image]

phobos-, 2022-09-13:
@haslinghuis tested from my end on latest master and the latest HDZ code - works fine.

haslinghuis, 2022-09-13:
Thanks for testing.

Closing as the issue is fixed in https://github.com/betaflight/betaflight/pull/11705

closed 2022-09-13 by haslinghuis

qwerfpv, 2022-10-27:
re. now i flashed the above fw on f7 fc, and the osd is displayed incorrectly, and battery indicator disappears from the OSD.
[image]
[image]
[image]

qwerfpv, 2022-10-27:
re. now it has this fw. seems good. I test it. I don't know why he gave him that name :D (hakrc F722)
[image]

qwerfpv, 2022-10-27:
re. unfortunately it's still not good. not with this fw either. :/ (f7)