betaflight/betaflight #11818: SmartPort send packet bug

BacikJan, 2022-09-01:
### Describe the bug

SmartPort packet contains at some situation wrong data and size. This bug was manifested by frequent reports of telemetry lost on my RC transmitter. I have created simple routine to read heading data from FC to analyze this bug and I found the problem in  the file `smartport.c`, function `void smartPortSendByte(uint8_t c, uint16_t *checksum, serialPort_t *port)`.

The problem is in the first condition, which causes that if payload data byte or CRC has values 0x7D or 0x7E (FSSP_DLE and FSSP_START_STOP), one more byte is transmitted. 

`if (c == FSSP_DLE || c == FSSP_START_STOP) {
        serialWrite(port, FSSP_DLE);
        serialWrite(port, c ^ FSSP_DLE_XOR);
    } else {
        serialWrite(port, c);
    }`

Example when turning FC around Z axis:

The right packet (size 8 bytes)
10 40 8 9e 7f 0 0 89
HEADER: 0x10 ID: 0x0840 DATA: 32670 CRC: 0x89

As I moving, the next received packet is wrong (size 9 bytes)
10 40 8 90 **7d 5e** 0 0 98
HEADER: 0x10 ID: 0x0840 DATA: 6192528 CRC: 0x00

In this case 0x5E should not be in the packet. Packet shoud has form 10 40 8 90 7d 0 0 98 -> HEADER: 0x10 ID: 0x0840 DATA: 32100 CRC:0x98

Another example of wrong packet -> CRC has value 0x7D or 0x7E and again packet has one more additional byte.

I am not sure why this condition exists, but in reality breaking communication through SmartPort.

Thank you.

### To Reproduce

Just read data from SmartPort telemetry.

### Expected behavior

Removing of the problematic condition.

### Diff configuration

```txt
# version
# Betaflight / STM32F7X2 (S7X2) 4.3.1 Jul 13 2022 / 03:36:10 (8d4f005) MSP API: 1.44
# config: manufacturer_id: MTKS, board_name: MATEKF722, version: edcd244b, date: 2022-06-29T02:24:46Z

# start the command batch
batch start

board_name MATEKF722
manufacturer_id MTKS

# resources
resource MOTOR 1 C08
resource MOTOR 2 C09
resource MOTOR 3 NONE
resource MOTOR 4 NONE
resource SERVO 1 B04
resource SERVO 2 B05
resource SERVO 3 B00
resource SERVO 4 B01
resource SERVO 5 B10
resource SERVO 6 B11
resource SERVO 7 B06
resource SERVO 8 B07
resource PWM 1 A02
resource LED_STRIP 1 A08
resource SERIAL_TX 6 C06
resource SERIAL_RX 6 C07
resource I2C_SCL 1 B08
resource I2C_SDA 1 B09
resource LED 1 A14
resource LED 2 A13
resource SPI_MOSI 2 C03
resource ADC_BATT 1 C02
resource ADC_RSSI 1 C00
resource ADC_CURR 1 C01
resource ADC_EXT 1 A04
resource SDCARD_CS 1 C15
resource PINIO 1 A15
resource PINIO 2 B03
resource GYRO_EXTI 1 C04
resource GYRO_CS 1 B02
resource USB_DETECT 1 C14

# timer
timer B04 AF2
# pin B04: TIM3 CH1 (AF2)
timer B05 AF2
# pin B05: TIM3 CH2 (AF2)
timer B00 AF2
# pin B00: TIM3 CH3 (AF2)
timer B10 AF1
# pin B10: TIM2 CH3 (AF1)
timer B11 AF1
# pin B11: TIM2 CH4 (AF1)
timer B06 AF2
# pin B06: TIM4 CH1 (AF2)
timer B07 AF2
# pin B07: TIM4 CH2 (AF2)
timer B15 AF9
# pin B15: TIM12 CH2 (AF9)

# dma
dma pin A08 2
# pin A08: DMA2 Stream 3 Channel 6
dma pin B04 0
# pin B04: DMA1 Stream 4 Channel 5
dma pin B05 0
# pin B05: DMA1 Stream 5 Channel 5
dma pin B00 0
# pin B00: DMA1 Stream 7 Channel 5
dma pin B10 0
# pin B10: DMA1 Stream 1 Channel 3
dma pin B11 1
# pin B11: DMA1 Stream 6 Channel 3
dma pin B06 0
# pin B06: DMA1 Stream 0 Channel 2
dma pin B07 0
# pin B07: DMA1 Stream 3 Channel 2

# feature
feature -RX_PARALLEL_PWM

# serial
serial 4 32 115200 57600 0 115200

# master
set acc_calibration = 41,-19,344,1
set ibata_scale = 250
set pid_process_denom = 8
set tlm_inverted = ON
set telemetry_disabled_voltage = ON
set telemetry_disabled_current = ON
set telemetry_disabled_fuel = ON
set telemetry_disabled_mode = ON
set telemetry_disabled_acc_x = ON
set telemetry_disabled_acc_y = ON
set telemetry_disabled_acc_z = ON
set telemetry_disabled_pitch = ON
set telemetry_disabled_roll = ON
set telemetry_disabled_altitude = ON
set telemetry_disabled_vario = ON
set telemetry_disabled_lat_long = ON
set telemetry_disabled_ground_speed = ON
set telemetry_disabled_distance = ON
set telemetry_disabled_temperature = ON
set sdcard_spi_bus = 2
set gyro_1_sensor_align = CW180FLIP
set gyro_1_align_pitch = 1800
set gyro_1_align_yaw = 1800

profile 0

rateprofile 0

# end the command batch
batch end
```

### Resources configuration

```txt
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: SERIAL_RX 2
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: LED 2
A14: LED 1
A15: PINIO 1
B00: FREE
B01: FREE
B02: GYRO_CS 1
B03: PINIO 2
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: FREE
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: OSD_CS
B11: FREE
B12: FREE
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: FREE
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: SPI_MOSI 2
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: SERIAL_TX 5
C13: BEEPER
C14: FREE
C15: SDCARD_CS
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Flight controller

Matek F722 WPX

ledvinap, 2022-09-01:
This is expected behavior. Smartport escapes bytes that match start/stop (0x7e) and DLE (0x7e). This way 0x7e is seen only as packet start/end, no payload/crc byte can confuse decoder.
See https://www.frsky-rc.com/wp-content/uploads/2017/07/Manual/FRSKY%20TELEMETRY%20PROTOCOL.PDF

Other side is buggy

BacikJan, 2022-09-01:
Thank you for your answer. I have Frsky Archer RS receiver and then I can't explain telemetry lost problem to myself.

klutvott123, 2022-09-01:
I wouldn't be surprised if it's a bug in the RX firmware. Packets with bytestuffing being dropped have been an issue forever. We got them to fix it for some RXs like the R-XSR and XSR.

closed 2022-10-10 by github-actions[bot]