betaflight/betaflight #11822: No NTSC or Pal choices in OSD Tab in 4.3.0

77Mick, 2022-09-03:
### Describe the bug

Missing video format menu in configurator OSD Tab, both using USB or LIPO as power supply

Board: JHEMCU GHF411 PRO AIO 

Betaflight Configurator 10.8.0 
Betaflight 4.3.0 

### To Reproduce

Connect USB or Lipo to board to reproduce the bug

### Expected behavior

Betaflight showing OSD PAL/NTSC choices as usual

### Diff configuration

```txt
# diff all

# version
# Betaflight / STM32F411 (S411) 4.3.0 Jun 14 2022 / 00:48:04 (229ac66) MSP API: 1.44
# config: manufacturer_id: JHEF, board_name: JHEF411, version: 7b156dec, date: 2021-07-26T13:19:23Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name JHEF411
manufacturer_id JHEF
mcu_id 003800243036511136383539
signature 

# name: CineMik

# resources
resource MOTOR 1 A10
resource MOTOR 2 B00
resource MOTOR 3 A08
resource MOTOR 4 A09

# feature
feature -RX_PARALLEL_PWM
feature -AIRMODE
feature TELEMETRY

# serial
serial 0 1 115200 57600 0 115200

# map
map TAER1234

# aux
aux 0 0 0 1800 2100 0 0
aux 1 1 1 900 1200 0 0
aux 2 13 5 1800 2100 0 0
aux 3 36 4 1800 2100 0 0

# master
set gyro_lpf1_static_hz = 0
set gyro_lpf2_static_hz = 0
set gyro_lpf1_dyn_min_hz = 0
set gyro_lpf1_dyn_max_hz = 875
set acc_calibration = -45,11,-114,1
set align_mag = CW180FLIP
set mag_align_pitch = 1800
set mag_align_yaw = 1800
set mag_hardware = NONE
set baro_hardware = NONE
set rc_smoothing_auto_factor = 250
set rc_smoothing_auto_factor_throttle = 170
set rc_smoothing_setpoint_cutoff = 12
set rc_smoothing_feedforward_cutoff = 12
set rc_smoothing_throttle_cutoff = 20
set serialrx_provider = CRSF
set blackbox_sample_rate = 1/2
set dshot_bidir = ON
set motor_poles = 12
set yaw_motors_reversed = ON
set small_angle = 180
set simplified_gyro_filter_multiplier = 175
set report_cell_voltage = ON
set osd_warn_rssi_dbm = ON
set osd_rssi_dbm_alarm = -98
set osd_vbat_pos = 389
set osd_rssi_pos = 404
set osd_link_quality_pos = 2452
set osd_link_tx_power_pos = 2388
set osd_rssi_dbm_pos = 2420
set osd_flymode_pos = 344
set osd_crosshairs_pos = 2285
set osd_current_pos = 2403
set osd_mah_drawn_pos = 2437
set osd_craft_name_pos = 2080
set osd_gps_speed_pos = 312
set osd_gps_sats_pos = 216
set osd_home_dir_pos = 142
set osd_home_dist_pos = 249
set osd_altitude_pos = 278
set osd_warnings_pos = 2313
set osd_avg_cell_voltage_pos = 2372
set osd_disarmed_pos = 2251
set osd_displayport_device = MSP
set debug_mode = GYRO_SCALED
set displayport_msp_serial = 0
set gyro_1_sensor_align = CW0FLIP
set gyro_1_align_pitch = 1800
set gyro_1_align_yaw = 0
set rpm_filter_q = 650
set name = CineMik

profile 0

# profile 0
set dterm_lpf1_dyn_min_hz = 78
set dterm_lpf1_dyn_max_hz = 157
set dterm_lpf1_static_hz = 157
set dterm_lpf2_static_hz = 157
set iterm_relax_type = GYRO
set yaw_lowpass_hz = 0
set p_pitch = 51
set i_pitch = 36
set d_pitch = 50
set f_pitch = 171
set p_roll = 49
set i_roll = 35
set d_roll = 44
set f_roll = 165
set p_yaw = 49
set i_yaw = 35
set f_yaw = 165
set angle_level_strength = 100
set d_min_roll = 44
set d_min_pitch = 50
set feedforward_averaging = 2_POINT
set feedforward_smooth_factor = 60
set feedforward_jitter_factor = 12
set simplified_master_multiplier = 110
set simplified_i_gain = 40
set simplified_d_gain = 135
set simplified_dmax_gain = 0
set simplified_feedforward_gain = 125
set simplified_dterm_filter = OFF
set simplified_dterm_filter_multiplier = 105

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set thr_expo = 20
set rates_type = BETAFLIGHT
set roll_rc_rate = 80
set pitch_rc_rate = 80
set yaw_rc_rate = 80
set roll_expo = 30
set pitch_expo = 30
set yaw_expo = 30
set roll_srate = 69
set pitch_srate = 69
set yaw_srate = 69

rateprofile 1

# rateprofile 1
set rates_type = BETAFLIGHT
set roll_rc_rate = 50
set pitch_rc_rate = 50
set yaw_rc_rate = 50
set roll_srate = 0
set pitch_srate = 0
set yaw_srate = 0

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
#
```

### Resources configuration

```txt
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: ADC_BATT
A01: ADC_CURR
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: MOTOR 3
A09: MOTOR 4
A10: MOTOR 1
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FREE
B00: MOTOR 2
B01: FREE
B02: FLASH_CS
B03: GYRO_EXTI
B04: FREE
B05: PINIO 1
B06: SERIAL_TX 1
B07: SERIAL_RX 1
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: CAMERA_CONTROL
B11: FREE
B12: PREINIT 3
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: BEEPER
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1 : DSHOT_BITBANG 1
    CH2 : DSHOT_BITBANG 2
TIM2:
    CH3 : CAMERA_CONTROL
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: DSHOT_BITBANG 1
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: FREE
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Flight controller

JHEMCU GHF411 PRO AIO

haslinghuis, 2022-09-03:
Could you specify the VTX being used?

This configuration is using:
```
set osd_displayport_device = MSP
set displayport_msp_serial = 0
```
as https://github.com/betaflight/betaflight/pull/11705 has not been implemented yet.

77Mick, 2022-09-03:
> Could you specify the VTX being used?
> 
> This configuration is using:
> 
> ```
> set osd_displayport_device = MSP
> set displayport_msp_serial = 0
> ```
> 
> as #11705 has not been implemented yet.

This configuration I am using is related to WTFOS setup, the community framework for modifying the firmware of DJI FPV Goggles and Air Units and enable full OSD. 
To get access to the whole OSD area, in addition to the two CLI input, you're supposed to set the Video Format to PAL or Auto in the OSD tab, hence the reason of my question.

k-trahan, 2022-09-06:
> This configuration I am using is related to WTFOS setup, the community framework for modifying the firmware of DJI FPV Goggles and Air Units and enable full OSD. To get access to the whole OSD area, [...]

When using MSP as the OSD DisplayPort, the video format section is hidden on the OSD tab. To get it back, you can disable MSP DisplayPort using `set osd_displayport_device = MAX7456`, change the video format to PAL or Auto as needed, then change back to MSP using `set osd_displayport_device = MSP`.

DavidJJ, 2022-09-18:
I am seeing this same behavior with IFRC/IFLIGHT_F411_PRO(STM32F411) flight controller with iFlight succx vtx. 

once I went into the cli and set the video type to NTSC there, the video mode showed up on the OSD tab.

haslinghuis, 2022-09-18:
Not working on SPRacingH7Extreme. Once setting `set osd_displayport_device = MSP` the setting `vcd_video_system` is ignored as it seams only implemented in MAX7456 code.

nerdCopter, 2022-10-25:
* confirmed issue on two FC with firmware 4.3.1 
* `set osd_displayport_device = MAX7456` & `save` restores the GUI option.  
* some issue with `= AUTO`, firmware side?

nerdCopter, 2022-10-26:
* found the commit/PR that changed this behavior: commit `f53d812fc` / https://github.com/betaflight/betaflight/pull/9333
* prior commit `728acc336` shows GUI fields correctly.

nerdCopter, 2022-10-26:
it seems the firmware itself works fine, so maybe a solution could easily be made in the configurator.

https://github.com/betaflight/betaflight-configurator/blob/c3a81ca438d8123ebf3e54b47362d83ce7704af8/src/js/tabs/osd.js#L2178 returns `false`, which triggers a hide at https://github.com/betaflight/betaflight-configurator/blob/c3a81ca438d8123ebf3e54b47362d83ce7704af8/src/js/tabs/osd.js#L2874

[image]
[image]

haslinghuis, 2022-10-29:
We might need to add a case for `OSD_DISPLAYPORT_DEVICE_MSP` in firmware and check on a new flag instead of 1.

haslinghuis, 2022-10-29:
@nerdCopter please check firmware build with configurator if this solves the problem.

nerdCopter, 2022-10-29:
in conjunction with https://github.com/betaflight/betaflight/pull/11931
this worked on one (1) tested FC `foxeerf405`.

[image]

closed 2022-10-31 by haslinghuis