betaflight/betaflight #11831: SD Card Fatal Error with BF 4.3.x -- BUT NOT with BF 4.2.x

spatzengr, 2022-09-10:
### Describe the bug

Betaflight 4.3.x fails to recognize the SD Card.

@hydra 

### To Reproduce

Get any Flight Controller with an SD Card reader.  In the CLI, paste in "SET blackbox_mode = ALWAYS" and "SAVE"  [ENTER]
IF - and that's a big IF - the SD card doesn't show the below on plug-in, it will after a few seconds.
[image]

### Expected behavior

The SD Card will show as a fatal error on powerup or plug-in.  Even if by chance it does recognize it, it will fatal error after a few seconds of logging.  Of course, if set as ALWAYS logging, you can watch this in the Configurator.  If you start to log on ARMING, then it will only record 1024kb of data to the card.

The below video shows that with BLACKBOX MODE set to ALWAYS, after multiple power ups, if the card is recognized, it will fatal error after a couple of seconds of recording.
https://youtu.be/cF182Uso4Kg

In this video, it shows that if BLACKBOX MODE is set to NORMAL, after multiple power ups, it may recognize the card, and in that case if not logging to not fatal error.
https://youtu.be/hdDOjGS73dk

Below are the logs from these tests.

**I HAVE TRIED THIS ON MULTIPLE FLIGHT CONTROLLERS WITH SD CARDS**, including:
SpeedyBee F405
Omnibus F4 SD
Radix V1

I have also tried multiple cards.  
SanDisk Extreme
SanDisk Ultra
Both cards have write speeds of 12Mb/sec plus and have and continue to serve me well when NOT used with BF 4.3.x.

**FOR EXAMPLE**If I take the same flight controller and SD CARD and flash it back to BF 4.2.x, it works fine without fail!

### Diff configuration

```txt
2022-09-10 @08:47:18 -- OS: Windows

2022-09-10 @08:47:19 -- Configurator: 10.8.0 (5ec9465)

2022-09-10 @08:47:20 -- Loaded release information for configurator from GitHub.

2022-09-10 @08:47:43 -- Serial port successfully opened with ID: 1

2022-09-10 @08:47:43 -- MultiWii API version: 1.44.0

2022-09-10 @08:47:43 -- Flight controller info, identifier: BTFL, version: 4.3.1

2022-09-10 @08:47:43 -- Running firmware released on: Jul 13 2022 03:32:11

2022-09-10 @08:47:43 -- Board: SPBE/SPEEDYBEEF405V3(STM32F405), version: 0

2022-09-10 @08:47:43 -- Unique device ID: 0x1f003f5646501720353936

2022-09-10 @08:47:43 -- Craft name:

2022-09-10 @08:47:43 -- Arming Disabled

2022-09-10 @08:47:56 -- EEPROM saved

2022-09-10 @08:47:56 -- Serial port successfully closed

2022-09-10 @08:48:00 -- Serial port successfully opened with ID: 2

2022-09-10 @08:48:00 -- MultiWii API version: 1.44.0

2022-09-10 @08:48:00 -- Flight controller info, identifier: BTFL, version: 4.3.1
[... 1690 lines cut ...]
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Flight controller

SPEEDYBEEF405V3

### Other components

CADDX Vista
ELRS Receiver

### How are the different components wired up

Vista to UART6
Receiver to UART1

SteveCEvans, 2022-09-10:
I've tested current master `aa185860e`, using a SanDisk Ultra 32G SD card and I'm seeing the following:

|FC|Result|
|--|------|
| MATEKF405STD | Fails as described in this PR |
| MATEKF722SE | Appears to function correctly |

This then appears to be an F4 issue. Investigating.

hydra, 2022-09-12:
As noted in Slack, this probably is F4 related.  Also require confirmation of testing on an H7 board.

Note that on F7 and H7 the FC might be using SDMMC, and not SPI.  For testing reports please confirm if the FC you're testing on has the SD card connected via SPI or SDMMC.

In the mean time I suggest checking the DMA buffers for F4 are the correct memory region and that the GPIO IO speeds and SPI / SD clock speeds are correct.  

If you're able to do a git bisect that would also help.

aowi7280, 2022-09-13:
I have an Omnibus F405 and it works fine with a  Lexar 16G 633x

Betaflight / STM32F405 (S405) 4.3.1 Jul 13 2022 / 03:32:11 (8d4f005) MSP API: 1.44
# config: manufacturer_id: AIRB, board_name: OMNIBUSF4SD, version: edcd244b, date: 2022-06-29T02:24:46Z
blackbox_sample_rate = 1/1

blackbox_device = SDCARD
Allowed values: NONE, SPIFLASH, SDCARD, SERIAL
I had BB enabled on a mode with ARM, so I removed it. Saved. Unplugged. Plugged. The SD card still works.

# status
MCU F40X Clock=168MHz (PLLP-HSE), Vref=3.31V, Core temp=34degC
Stack size: 2048, Stack address: 0x1000fff0
Configuration: CONFIGURED, size: 4100, max available: 16384
Devices detected: SPI:2, I2C:0
Gyros detected: gyro 1 locked
GYRO=MPU6000, ACC=MPU6000, BARO=BMP280, MAG=HMC5883
OSD: MAX7456
System Uptime: 112 seconds, Current Time: 2022-09-12T22:33:02.234-06:00
CPU:48%, cycle time: 128, GYRO rate: 7812, RX rate: 49, System rate: 9
Voltage: 1 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
SD card: Manufacturer 0x28, 15645696kB, 11/2016, v0.3, '5TCGO'
Filesystem: Ready
Arming disable flags: RXLOSS BADRX THROTTLE NOPREARM CLI MSP RPMFILTER ARMSWITCH

hydra, 2022-09-13:
@spatzengr @aowi7280 @SteveCEvans Its not clear from your posts how the SD card is connected to your FCs in the cases where it works and where it does not.

Can you all please post the output of the cli 'status', 'dump' and 'resource show all' commands for all the FCs you've tested so that we can compare clock speeds and SD card connectivity and DMA usage.

aowi7280, 2022-09-13:
# status
MCU F40X Clock=168MHz (PLLP-HSE), Vref=3.31V, Core temp=26degC
Stack size: 2048, Stack address: 0x1000fff0
Configuration: CONFIGURED, size: 4100, max available: 16384
Devices detected: SPI:2, I2C:0
Gyros detected: gyro 1 locked
GYRO=MPU6000, ACC=MPU6000, BARO=BMP280, MAG=HMC5883
OSD: MAX7456
System Uptime: 71 seconds, Current Time: 2022-09-13T08:55:11.428-06:00
CPU:48%, cycle time: 128, GYRO rate: 7812, RX rate: 49, System rate: 9
Voltage: 1 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
SD card: Manufacturer 0x28, 15645696kB, 11/2016, v0.3, '5TCGO'
Filesystem: Ready
Arming disable flags: RXLOSS BADRX THROTTLE NOPREARM CLI MSP RPMFILTER ARMSWITCH

# dump

# version
# Betaflight / STM32F405 (S405) 4.3.1 Jul 13 2022 / 03:32:11 (8d4f005) MSP API: 1.44
# config: manufacturer_id: AIRB, board_name: OMNIBUSF4SD, version: edcd244b, date: 2022-06-29T02:24:46Z

# start the command batch
batch start

board_name OMNIBUSF4SD
manufacturer_id AIRB

# name: Androne

# resources
resource BEEPER 1 B04
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 A03
resource MOTOR 4 A02
resource MOTOR 5 NONE
resource MOTOR 6 A08
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 B08
resource PWM 1 B08
resource PWM 2 B09
resource PWM 3 C06
resource PWM 4 C07
resource PWM 5 C08
resource PWM 6 C09
resource PWM 7 NONE
resource PWM 8 NONE
resource SONAR_TRIGGER 1 NONE
resource SONAR_ECHO 1 A08
resource LED_STRIP 1 B06
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 NONE
resource SERIAL_TX 3 NONE
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
[... 1262 lines cut ...]
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: LED_STRIP
DMA1 Stream 1: FREE
DMA1 Stream 2: SPI_MISO 3
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: ADC 2
DMA2 Stream 4: DSHOT_BITBANG 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

SteveCEvans, 2022-09-13:
I’m building a debug cable to look into this properly, but alas Amazon are apologising that they didn’t deliver the bits today! A quick look has shown a very short high time in the first SPI clock cycle, so I’ll be looking into that first.

SteveCEvans, 2022-09-16:
@hydra please test.

closed 2022-09-16 by blckmn

hydra, 2022-09-23:
> @hydra please test.

I don't have the FC's mentioned, my current line-up of FCs use SDIO not SPI and as I understand it #11831 is an SPI related fix which is likely why I didn't notice the bug in my testing of the H7EXTREME/H7RF/etc.