betaflight/betaflight #11836: Matek F405

Srmoder12345, 2022-09-13:
### Describe the bug

Matek F405 can't find betaflight 4.3

### To Reproduce

Last version was in 2019.
Matek F405 .
betaflight 4.1.2...

### Expected behavior

I hope I can also enjoy betaflight 4.3
using my cost-effective Matek F405 controller.

### Diff configuration

```txt
#matekf405 not betaflighit 4.3
```

### Resources configuration

```txt
#add betaflighit 4.3
```

### Flight controller

Matek F405  TCMMRC  tyro 129 Aliexpress 

### Other components

Speedbybe

### How are the different components wired up

Solda

### Add any other context about the problem that you think might be relevant here

I hope I can also enjoy betaflight 4.3
using my cost-effective Matek F405 controller.

haslinghuis, 2022-09-13:
Duplicate of #11837

closed 2022-09-13 by haslinghuis

Srmoder12345, 2022-09-13:
[image]

Final versión no.

Srmoder12345, 2022-09-13:
[image]

Maintenace versión yes.