betaflight/betaflight #11842: FC with SPI elrs rx no longer binds

dischinator, 2022-09-19:
### Describe the bug

Flashing a dev build starting from commit 122fb891fd2a544255533b28eaf758a6b19c8012 elrs spi receivers no longer bind to the tx. Flashing an older build, for example the c78e14cc2766ddccf43826d7c5311e405e4c56b2 rx will bind correctly

### To Reproduce

Flash a dev build #2768 (122fb891f) on a board with an integrated elrs spi receiver. It will no longer bind to tx.

### Expected behavior

elrs spi receiver should bind

### Diff configuration

```txt
# diff

# version
# Betaflight / STM32F411SX1280 (S4SX) 4.4.0 Sep 16 2022 / 22:27:16 (122fb891f) MSP API: 1.45
# config: manufacturer_id: BEFH, board_name: BETAFPVF4SX1280, version: 7df66093, date: 2022-06-12T13:41:42Z

# start the command batch
batch start

board_name BETAFPVF4SX1280
manufacturer_id BEFH

# master
set acc_hardware = NONE
set serialrx_provider = CRSF
set motor_pwm_protocol = DSHOT300
set expresslrs_uid = 83,11,55,213,118,59
set expresslrs_rate_index = 1

profile 0

rateprofile 0

# end the command batch
batch end
```

### Resources configuration

```txt
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: MOTOR 2
A01: ADC_BATT
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: RX_SPI_EXPRESSLRS_BUSY
A14: BEEPER
A15: RX_SPI_CS
B00: ADC_CURR
B01: LED_STRIP
B02: RX_SPI_BIND
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: GYRO_EXTI
B07: MOTOR 4
B08: MOTOR 1
B09: RX_SPI_EXPRESSLRS_RESET
B10: MOTOR 3
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: RX_SPI_EXTI
C14: LED 1
C15: LED
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH1 : MOTOR 2
    CH3 : MOTOR 3
TIM3:
    CH4 : LED_STRIP
TIM4:
    CH2 : MOTOR 4
    CH3 : MOTOR 1
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: LED_STRIP
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: TIMUP 4
DMA1 Stream 7: TIMUP 2
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

#
```

### Flight controller

DarwinFPV ExpressLRS ELRS AIO F4

### Add any other context about the problem that you think might be relevant here

122fb891fd2a544255533b28eaf758a6b19c8012 introduces some changes to spi handling, that probably causes the problem.

DLLarson, 2022-09-19:
Hi,

As another data point, I had the same issue with yesterday's nightly so I cherry picked the SPI ELRS 3.0 pull request on top of Betaflight version tag 4.3.1. Then I built target CRAZYBEEF4SX1280, flashed my Mobula 6 and binding worked with my binding phrase. For more info see my issue discussion here:

https://discord.com/channels/596350022191415318/798006228450017290/1021455222909980753

-Dale

dischinator, 2022-09-20:
> I cherry picked the SPI ELRS 3.0 pull request on top of Betaflight version tag 4.3.1. Then I built target
@DLLarson Thanks thats was a great idea, I just did it that also since I ran into other problems with my board using the nightly builds.

haslinghuis, 2022-09-20:
@SteveCEvans After some testing https://github.com/betaflight/betaflight/pull/11839 is causing ELRS not being able to bind anymore. Please have a look.

DLLarson, 2022-09-20:
@dischinator My preference was to stay with 4.3.1 anyway so I was happy this worked. The quad flies fine and I don't have to deal with the uncertainties of development builds. 

When flashing 4.4 I had a bunch of errors when loading my 4.3.1 cli backup. For instance, 4.4 seems to have reduced the number of rate profiles to four rather than six. Many OSD defn's are no longer supported. All in all 4.3.1 is a solid codebase so that's where I want to be if possible.

SteveCEvans, 2022-09-20:
> @SteveCEvans After some testing #11839 is causing ELRS not being able to bind anymore. Please have a look.

Will do.

dischinator, 2022-09-21:
> Will do.

Let me know if you need help with testing.

SteveCEvans, 2022-09-21:
@dischinator See https://github.com/betaflight/betaflight/pull/11845

Image below for testing.

[betaflight_4.4.0_STM32F411SX1280.hex.zip](https://github.com/betaflight/betaflight/files/9619489/betaflight_4.4.0_STM32F411SX1280.hex.zip)

dischinator, 2022-09-21:
@SteveCEvans works for me, nice job.

closed 2022-09-21 by dischinator