betaflight/betaflight #11849: Make Issue template axpandable

mituritsyn, 2022-09-24:
### Is your feature request related to a problem? Please describe

issues are hard to read

### Describe the solution you'd like

hide at least diff section of issue template under details or similar tag
```
<details>
  <summary>diff</summary>
</details>
```

### Describe alternatives you've considered

--

McGiverGim, 2022-10-07:
Are you talking about GitHub issues? If that's true, we are limited to the tools that GitHub allows, and I don't think something like this is available at this moment.

KarateBrot, 2022-10-07:
@McGiverGim 

<details>
  <summary>diff</summary>

```c
// long diff file which is expandable
```

</details>

haslinghuis, 2022-10-07:
Looking for a solution but can't find. We just need a way to add the markdown to the placeholder of the textarea.
https://docs.github.com/en/enterprise-server@3.6/actions/using-workflows/workflow-commands-for-github-actions#example-grouping-log-lines

McGiverGim, 2022-10-08:
Yes, is a good idea, but we are limited to what GitHub issues let to do, more info https://docs.github.com/en/communities/using-templates-to-encourage-useful-issues-and-pull-requests/syntax-for-issue-forms and https://docs.github.com/en/communities/using-templates-to-encourage-useful-issues-and-pull-requests/syntax-for-githubs-form-schema

We can't format it with what they provide. The only solution maybe is to create a Github Action to "edit" any new issue and format it as we want.

KarateBrot, 2022-10-08:
Just to be clear, the dropdown is not markdown itself, but HTML rendered in markdown. If that doesn't make a difference that's a bummer...

McGiverGim, 2022-10-08:
No, there is no difference. We use github yaml forms, and we are limited to the ”components” they provide. At least I don't see any workaround.
The only option is to change the textarea from text to markdown and add instructions for the user to add the html tags itself, but this will end with users breaking the layout. It happened before when we only wanted the three backtricks.

closed 2023-02-16 by mituritsyn