betaflight/betaflight #11852: on HGLRCF411 my PID loop stuck on 2K but on MATEK411 works fine.

porlock, 2022-09-25:
### Describe the bug

I installed BF 4.3.1 on HGLRCF411, and my PID loop stuck on 2K, when i change it to 4K and save, after reboot get back to 2K. 
On MATEK411 i dont have this issue, but target seems the same, only custom settings differ, so what can cause that ? Both MATEK411 and HGLRCF411 are not overclocked by default.

### To Reproduce

Install BF 4.3.1 on HGLRCF411 try changing PID loop to 4K

### Expected behavior

PID loop changed to 4K

### Diff configuration

```txt
# diff all

# version
# Betaflight / STM32F411 (S411) 4.3.1 Jul 13 2022 / 03:32:52 (8d4f005) MSP API: 1.44
# config: manufacturer_id: HGLR, board_name: HGLRCF411, version: a074838b, date: 2021-06-14T21:36:54Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name HGLRCF411
manufacturer_id HGLR
mcu_id 002900533239511334393435
signature 

# master
set acc_calibration = 27,-14,-2,1
set pid_process_denom = 4
set gyro_1_align_yaw = 1800

profile 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
```

### Resources configuration

```txt
resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: GYRO_EXTI
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FLASH_CS
B00: ADC_BATT
B01: ADC_CURR
B02: BEEPER
B03: FREE
B04: MOTOR 1
B05: MOTOR 2
B06: MOTOR 3
B07: MOTOR 4
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: CAMERA_CONTROL
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3:
    CH1 : MOTOR 1
    CH2 : MOTOR 2
TIM4:
    CH1 : MOTOR 3
    CH2 : MOTOR 4
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

#
```

### Flight controller

HGLRCF411 AIO

SteveCEvans, 2022-09-25:
`set pid_process_denom = 4` is setting the PID rate to one quarter on the gyro rate, so you’re getting what you asked for. Try setting it to `2`.

porlock, 2022-09-25:
> `set pid_process_denom = 4` is setting the PID rate to one quarter on the gyro rate, so you’re getting what you asked for. Try setting it to `2`.

check this out: https://youtu.be/oNbdFL_psUY

That is te point, it not work, doesn't matter if i change it from cli or by UI, after reset its goes back to 2k.  I noticed that wen i do CPU overclocking it allows me to change to 4K. But for MATEK411 i can set 4K without overclocking.  Is there anything hard coded in BF code that treat HGLRCF411 "special" ? If yes, than why ? Why MATEK411 can run without overclocking ?

porlock, 2022-09-25:
what is even strange is that pid_process_denom = 4 is not part of custom settings for this FC
https://raw.githubusercontent.com/betaflight/unified-targets/master/configs/default/HGLR-HGLRCF411.config

haslinghuis, 2022-09-25:
What is the `status` of this board?

closed 2022-11-02 by github-actions[bot]