betaflight/betaflight #11853: Anti-gravity I-term coefficient not affected by anti-windup

tbolin, 2022-09-25:
### Describe the bug

The I-term is calculated as follows:
https://github.com/betaflight/betaflight/blob/fba7ccfcd069cf76da74a978836a9fdceeef242e/src/main/flight/pid.c#L1022
where `axisDynCi ` is the anti-windup coefficient, `Ki` is the standard I-term coefficient, and `pidRuntime.itermAccelerator` is the extra I-term coefficient from anti-gravity.
If the motors are nearing saturation `axisDynCi` will be lowered (otherwise it is equal to `pidRuntime.dT`), and in full saturation it will be 0.
Since `pidRuntime.itermAccelerator`,  the extra I-term coefficient from anti-gravity is not multiplied with `axisDynCi ` the anti-windup mechanism is partially bypassed when anti-gravity is active, allowing windup.

This bug(?) appears to  have been introduced when the extra I-term gain from anti-gravity was separated from the normal I-gain, in https://github.com/betaflight/betaflight/pull/9389. 
Before that commit the anti-gravity gain was affected by the anti-windup since it was multiplied by the `DynCi` coefficient here 
https://github.com/betaflight/betaflight/blob/a2aeb0cef4a6fe904fdb992bfe2e453fc776b516/src/main/flight/pid.c#L1313
I can't find any discussion on changing how anti-windup affects anti-gravity in https://github.com/betaflight/betaflight/pull/938 or the related issue, so I assume that the change was unintentional. 

### To Reproduce

N/A

### Expected behavior

Anti-windup should affect the I-gain from anti-gravity the same way it affects the normal I-gain.

### Diff configuration

```txt
N/A
```

### Resources configuration

```txt
N/A
```

### Flight controller

All

Quick-Flash, 2022-10-04:
iterm relax should also effect it. So coding it up so that the changes that iterm relax, and anti-windup both apply to the iterm as a variable you can attenuate this by would be worthwhile.

tbolin, 2022-10-08:
Updated the first post with steps to reproduce.

tbolin, 2022-10-08:
Pushed PR #11878 with a proposed fix.

PR #11878 (2022-10-13): Make anti-windup affect the gain from Anti Gravity
Fixes #11853 
See the issue for description and steps to replicate.
With this PR the extra I term gain from Anti Gravity is affected by anti-windup the same way as the normal I term gain.
closed 2022-10-13 by haslinghuis