betaflight/betaflight #11854: Incorrect throttle adjustment in mixer during saturation when airmode is active

tbolin, 2022-09-25:
### Describe the bug

If the range of the output values from the mixer (the mix range) is larger than 1, the output vector to the motors is scaled so that the range of the scaled vector is 1.
If air mode is active the throttle should then be adjusted to make sure that the values in the output vector with added throttle is in the range of [-0.5, 0.5]. 
The current code adjusts the throttle correctly if the mix range is less than 1. 
If the mix range is greater than 1 however, the throttle will always be set to 0.5, which is often incorrect and can cause the output vector to be clipped.
The code can be found here:
https://github.com/betaflight/betaflight/blob/fba7ccfcd069cf76da74a978836a9fdceeef242e/src/main/flight/mixer.c#L454:L456

### To Reproduce

Slowly force a quadcopter towards saturation. The moment saturation is reached there should be a sudden jump in average motor outputs. 
Here is a black box log I found that displays this behavior, together with calculations that indicate that the throttle is incorrectly adjusted and the motor output vector is clipped when the mixer is saturated,  while the throttle is correctly adjusted and the output vector not clipped when the mixer is not saturated.

In the first two screenshots the mix range is at about 90% of maximum.
[image]
The calculations show that the motor outputs in the log is consistent with about 35% adjusted throttle, and no clipping of the output vector.
[image]

A hundredth of a second later there is a significant jump in the output of the three motors not already at full throttle.
[image]
The calculations show that the mix range should now be above 100%, so the mixer is saturated. 
The observed motor output is also consistent with an adjusted throttle of 50% and the output of motor 2 being clipped.
[image]

### Expected behavior

The formula that is currently used for adjusting the throttle when the mixer is not saturated, should also be used when the mixer is saturated.  

### Diff configuration

```txt
N/A
```

### Resources configuration

```txt
N/A
```

### Flight controller

All

Quick-Flash, 2022-09-25:
The math here that your doing is a little off. Dont look at pidsum, but rather look at pre-airmode clip adjusted motors. Combined movements end up adding and subtracting different motors. So even though a pidsum of 500 is all it takes to cause a fully saturated mixer, just adding up the pidsums doesnt always mean you just can use that as a measurement of saturation.

tbolin, 2022-09-26:
I calculate saturation based on the output of the mixer matrix, not by adding PID-sums. I cropped out the mixer matrix from the screenshots of the spreadsheets, but it is in there.

More importantly, even if my saturation calculations are off, that does not explain the issue.
The issue is that the mixer suddenly jumps from calculating the correct throttle adjustment when the mix range is bellow 1, to always using a throttle adjustment of 0.5 when the mix range is greater than 1, regardless of if the newly adjusted motor mix values is in the range -0.5 to 0.5 or not.

I was also surprised to find a bug there. 
The way I discovered this was that I noticed that all PIDs in the log were at their hard limits, but the motor output range looked far from saturation.  
Normally I would then expect the difference between the motors with highest and lowest output here to be close to 100%, or at least around 90% depending on which limits and mechanisms that that black box viewer accounts for, but it was not even 75%.
The highest motor was on 100%, but the lowest was at 26% which seemed like a bit high to be the  minimum motor output.

I then started recreated the mixer matrix in a spreadsheet just to see how it worked.
Before I thought that the output from the mixer matrix would always be centered on 0 (that is, mix max + mix min = 0), and 0.5 throttle would then always give maximum authority to the PIDs. 
However, when playing around with the spreadsheet I got values that where not centered on 0.
The mean of the output is 0, but mix max + mix min is not always 0.
Assuming that this is correct, 0.5 is not always the throttle that gives the maximum authority to the PIDs and will cause the output sent to the motors to be clipped.

I first thought there was a bug in the spreadsheet, so I triple checked everything there and didn't find any errors.
To be really sure I also recreated the rest of the mixer calculations (with airmode on) from PID signals to the D-shot values in excel to convince myself that nothing was wrong with the spreadsheet.
Given PID-values from the log the spreadsheet outputs D-shot values that are equal to those in the log.
That indicates that the mixer calculations in the spreadsheet are correct, so mixer output is not always centered on 0, so 0.5 is not always the throttle output that gives the greatest authority to the PIDs. In the current code, throttle will always be set to 0.5 if the mixer range is greater than 1.0, so there is a bug.

Also, the D-shot values for motor 2 when the mix range is above 1.0. seen in the log can only be explained by the output being clipped before it is applied to the motors, as seen in the third and fourth screenshot (I've checked that the raw outputs for all four motors are the same in the spreadsheet and in the log).

tbolin, 2022-09-29:
I found an easy way demonstrate the bug on a desk with stock Betaflight

1. Plug a quad (or just an FC with an RX) into a computer. DO NOT PLUG IN A BATTERY.
2. Check that there are no battery plugged in.
3. Backup the quad settings.
4. Go to the settings  tab and turn on AIRMODE (prmanently enable airmode) under *Other features*, click save and reboot.
5.  Go to the PID Tuning tab and set all PID gains except P to 0. Turn off I Tem Relax, Anti gravity, Vbat Sag Compensation Thrust Linearization. Click save and reboot.
6. Go to the motors tab, turn off Bidirectional DShot. Set motor Idle to 0. Click save and reboot.
7. Check that there is no battery plugged in, or take off the props.
8. Go to the motors tab and and click the motor test mode
9. **Check that there is no battery is plugged in to the quad**
10. Arm the quad
11. Give approximately the same amount of Roll, Pitch, and Yaw input so that the motors are just about saturating. throttle doesn't matter.
12. Wiggle one stick just a little bit further. The outputs of three of the motors should jump towards 0.5. 

A similar procedure can be used to verify that airmode off also suffers from a similar bug. Turn airmode off under settings and set throttle to at least 50% in step 11. The bug most notable on 100% throttle.

closed 2022-09-30 by sugaarK