betaflight/betaflight #11855: Flashing from 4.2.11 -> 4.3 on T-Motor VELOX Lite F411 Flight Controller causes Gyro to not work

gregoryw3, 2022-09-25:
### Describe the bug

When flashing from 4.2.11 to 4.3 the FC Gyro stops working. This is the second board this has happend to, both brand new boards.

### To Reproduce

Take brand new FC, flash to 4.3. Gyro no longer works. First board gyro failed but betaflight configurator gave no errors, second board betaflight gave the No Gyro error.

### Expected behavior

Gyro to not break.

### Diff configuration

```txt
# diff all

# version
# Betaflight / STM32F411 (S411) 4.2.11 Nov  9 2021 / 20:28:23 (948ba6339) MSP API: 1.43
# config: manufacturer_id: TMTR, board_name: TMOTORF411, version: edcd244b, date: 2022-06-29T02:24:46Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name TMOTORF411
manufacturer_id TMTR
mcu_id 003e00113236510135383430
signature 

# feature
feature -RX_PARALLEL_PWM
feature -DYNAMIC_FILTER

# master
set mag_bustype = I2C
set baro_bustype = I2C
set motor_pwm_protocol = DSHOT300
set align_board_roll = 180
set align_board_yaw = -90

profile 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
#
```

### Resources configuration

```txt
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: ADC_CURR
A01: ADC_BATT
A02: FREE
A03: SERIAL_RX 2
A04: PREINIT 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: LED_STRIP
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FLASH_CS
B00: MOTOR 1
B01: MOTOR 2
B02: BEEPER
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: MOTOR 3
B07: MOTOR 4
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: FREE
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1: LED_STRIP
TIM2: FREE
TIM3:
    CH3: MOTOR 1
    CH4: MOTOR 2
TIM4:
    CH1: MOTOR 3
    CH2: MOTOR 4
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: TIMUP 3
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: TIMUP 4
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: LED_STRIP
DMA2 Stream 7: FREE

#
```

### Flight controller

T-Motor VELOX Lite F411 Flight Controller from GetFPV

### Other components

XILO AXII Stubby SMA 5.8GHz Antenna (RHCP)
Radiomaster R81 8CH Frsky D8 / D16 Compatible Nano Receiver w/Sbus
XILO STAX 5.8GHz FPV Video Transmitter (25-600mW) w/ Smart Audio
T-Motor VELOX Lite F411 Flight Controller
Foxeer Razer Micro 1200TVL 1.8mm FPV Camera

All from GetFPV

### How are the different components wired up

Wired correctly, everything worked until updating FC.

KarateBrot, 2022-09-25:
The correct target for your board is [TMVELOXF411](https://github.com/betaflight/unified-targets/blob/master/configs/default/TMTR-TMVELOXF411.config). Select it in the target dropdown before flashing. Comment if the gyro works again (or if not).

closed 2022-09-25 by KarateBrot

gregoryw3, 2022-09-25:
Thank you, the gyro now works. Auto Select did not correctly select the board.

KarateBrot, 2022-09-25:
Nice!

That's because auto select just puts into the box what target is currently on your board. Since it was the wrong target, the wrong target also got auto selected.

haslinghuis, 2022-09-25:
@gregoryw3 that's because the target is new and it probably was shipped with former target. Auto-detect just reads the target from the board.

gregoryw3, 2022-09-25:
@haslinghuis Then why did it select the wrong target in the first place? I used auto to update it from its stock 4.2.11 to 4.3.

haslinghuis, 2022-09-25:
As I said auto-detect just reads the target from the connected board for selection. In this case a new target has been provided and had to be manually selected.

gregoryw3, 2022-09-25:
Sorry, I meant to reply to KarateBrot, I didn't read your comment and thought it was KarateBrot's. Thank you for the clarification reguardless.