betaflight/betaflight #11877: Unified target config files missing in 4.3.1 ZIP download

nmw01223, 2022-10-08:
### Describe the bug

Not sure if this is the right place to report this:

In the 4.3.1 ZIP source tree as downloaded, there are only five .config files in the unified targets config directory.

There are a large number in the online github folder, so not sure if it is an oversight or intended that they are not in the ZIP download?

### To Reproduce

n/a

### Expected behavior

n/a

### Diff configuration

```txt
n/a
```

### Resources configuration

```txt
n/a
```

### Flight controller

n/a

### Other components

n/a

### How are the different components wired up

n/a

### Add any other context about the problem that you think might be relevant here

n/a

SteveCEvans, 2022-10-08:
This is as intended.

closed 2022-10-08 by SteveCEvans

nmw01223, 2022-10-08:
Might be an idea then to put a short text file in the ZIP directory saying so?