betaflight/betaflight #11879: Battery discharge feature

anzejarni, 2022-10-09:
### Is your feature request related to a problem? Please describe

Battery chargers are able to charge at a decent rate of few amps, but discharging is many times limited to 1 amp due to small resistors in such chargers and as such the unused battery packs are always a hassle to discharge if not used in flight and take a lot of time on the charger to discharge. I propose a relatively safe way to discharge the battery on the quad itself.

### Describe the solution you'd like

Betaflight is already able to determine the number of cells in the battery pack and for lipo batteries there is a known storage voltage of 3.8 volts. I propose a feature which would be mapped to the Modes tab called Discharge.

When triggering this mode the fans would spin up slowly to a low RPM value to keep the quad on the ground and use the motors as a discharge load without any gyro balancing. If any of the motor's RPM would fall significantly all motors would stop immediately as this means one of the motors has hit and obstacle. If the accelerometer would recognize a spike, it would also stop all the motors immediately as this means the quad is no longer balanced in some direction. If both conditions (RPM, Accelerometer) would stay stable, the quad would continue to spin the motors until the voltage of 3.8 volts per cell is achieved and then it would stop all motors. It could also warn the user with a beep that it has completed the operation.

### Describe alternatives you've considered

-

KarateBrot, 2022-10-09:
This doesn't work because

1) Betaflight can not detect the number of cells. It only assumes the number of cells.
2) Betaflight cannot monitor each individual cell. One individual cell could be lower than the rest, so discharging can damage your battery this way

You should rather buy a proper battery discharger so that you can quickly discharge batteries accordingly. This also is WAY WAY quicker than trying to discharge by idling the motors.

aQuaplug, 2022-10-10:
I think it's very harmful to FPV equipments.
Many parts of miniquads needs cooling by airflow while flying, especially VTX.

anzejarni, 2022-10-10:
@KarateBrot , @aQuaplug I understand your points and also agree with them, but Betaflight per se is a DIY maker firmware and as such expects the user to have some intelligence and knowledge on the subject.

aQuaplug, 2022-10-10:
@anzejarni Yes, adding hardware switch, or some FC has bulit in power switch for VTX.
But it's still dangerous feature for novices who don't know that.
Sure I think it would be handy if Betaflight discharge and cut off presetted voltage.
Even can handle receiver power, flying second quad while discharging first quad, or you just hands off radio. right?
But, if it can, it's very dangerous something touching quad that prop spinning without rx.

For now, I think proper way to set airmode and throttle cut(-100) by radio. And set telemetry alarm on proper cell voltage.
Without rx power option, radio should'nt be idle for other quad. With rx power off, it's dangerous or too sensitive quad's movement-prop off that makes you hassle.
And I think I don't need discharging on field.

https://www.banggood.com/AOKoda-CellMeter-8-150W-Discharge-Module-Set-with-Battery-Voltage-Tester-Buzzer-p-1196807.html?cur_warehouse=CN

It's discontinued, find similar. It was under $30.
Cheap discharger what I'm using with parallel board.
Set discharge 50W, 100W, 150Watts with three 50W bulb.
Cheaper and better choice than brownout VTX or DVR.

KarateBrot, 2022-10-10:
The ISDT FD100 and FD200 are very nice dischargers, imo.

closed 2022-11-18 by github-actions[bot]