betaflight/betaflight #11893: Led strip battery not works properly

RipperGOT, 2022-10-15:
### Describe the bug

The led strip does not change color when the voltage drops

### To Reproduce

Activate led strip
Assign one or more leds to battery
Varies input voltage
Leds always is in green

### Expected behavior

When the voltage drops, the color of the led should change green-yellow-red-purple

### Diff configuration

```txt
# diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.3.1 Jul 13 2022 / 03:36:10 (8d4f005) MSP API: 1.44
# config: manufacturer_id: MTKS, board_name: MATEKF722HD, version: edcd244b, date: 2022-06-29T02:24:46Z

# start the command batch
batch start

board_name MATEKF722HD
manufacturer_id MTKS

# feature
feature -RX_PARALLEL_PWM
feature GPS
feature LED_STRIP
feature ESC_SENSOR

# serial
serial 2 1024 115200 57600 0 115200
serial 3 2 115200 57600 0 115200
serial 4 8192 115200 57600 0 115200
serial 5 65536 115200 57600 0 115200

# beeper
beeper -ON_USB
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE

# beacon
beacon RX_LOST
beacon RX_SET

# led
led 0 7,0::CB:2
led 1 8,0::CB:2
led 2 9,0::CB:2
led 3 6,2::C:6
led 4 10,2::C:2
led 5 4,10::AB:2
led 6 5,10::AB:2
led 7 6,10::L:0
led 8 7,10::L:0
led 9 8,10::L:0
led 10 9,10::L:0
led 11 10,10::AB:2
led 12 11,10::AB:2

# mode_color
mode_color 6 1 2
mode_color 6 5 6
mode_color 6 7 2

# aux
aux 0 0 0 1775 2100 0 0
aux 1 1 3 900 1150 0 0
aux 2 2 3 1300 1700 0 0
aux 3 46 6 1650 2100 0 0
aux 4 13 7 1750 1875 0 0
aux 5 15 7 1325 1475 0 0
aux 6 19 7 1525 1675 0 0
aux 7 26 0 1750 2100 0 0
aux 8 35 7 1925 2100 0 0
aux 9 36 1 1575 2100 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 5
vtxtable powervalues 25 200 400 600 600
vtxtable powerlabels 25 200 400 800 800

# master
set gyro_lpf1_static_hz = 0
set dyn_notch_count = 1
set dyn_notch_q = 500
set gyro_lpf1_dyn_min_hz = 0
set gyro_lpf1_dyn_max_hz = 0
set acc_calibration = 19,-10,109,1
set align_mag = CW90FLIP
set mag_align_pitch = 1800
set mag_align_yaw = 900
set mag_calibration = 294,-366,-858
set min_check = 1010
set max_check = 1990
set rc_smoothing_auto_factor = 80
set rc_smoothing_setpoint_cutoff = 25
set rc_smoothing_feedforward_cutoff = 25
set serialrx_provider = CRSF
set dshot_idle_value = 700
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set failsafe_delay = 5
set failsafe_procedure = GPS-RESCUE
set bat_capacity = 1800
set ibata_scale = 100
set yaw_motors_reversed = ON
set small_angle = 180
set gps_provider = UBLOX
set gps_sbas_mode = EGNOS
set gps_auto_baud = ON
set gps_ublox_use_galileo = ON
set gps_rescue_angle = 40
set gps_rescue_initial_alt = 70
set gps_rescue_descent_dist = 50
set gps_rescue_throttle_max = 1800
set gps_rescue_descend_rate = 200
set gps_rescue_throttle_hover = 1300
set gps_rescue_sanity_checks = RESCUE_SANITY_FS_ONLY
set gps_rescue_min_dth = 50
set gps_rescue_allow_arming_without_fix = ON
set gps_rescue_alt_mode = FIXED_ALT
set deadband = 5
set yaw_deadband = 5
set report_cell_voltage = ON
set telemetry_disabled_temperature = ON
set osd_warn_link_quality = ON
set osd_warn_rssi_dbm = ON
set osd_rssi_alarm = 100
set osd_link_quality_alarm = 85
set osd_rssi_dbm_alarm = -100
set osd_cap_alarm = 1800
set osd_alt_alarm = 115
set osd_tim1 = 1795
set osd_vbat_pos = 6496
set osd_rssi_pos = 439
set osd_link_quality_pos = 6552
set osd_rssi_dbm_pos = 6585
set osd_tim_1_pos = 6616
set osd_tim_2_pos = 472
set osd_flymode_pos = 6605
set osd_vtx_channel_pos = 6176
set osd_crosshairs_pos = 6382
set osd_ah_sbar_pos = 239
set osd_ah_pos = 6255
set osd_current_pos = 6560
set osd_mah_drawn_pos = 6593
set osd_craft_name_pos = 428
set osd_gps_speed_pos = 6369
set osd_gps_lon_pos = 6161
set osd_gps_lat_pos = 6144
set osd_gps_sats_pos = 6233
set osd_home_dir_pos = 6223
set osd_home_dist_pos = 6265
set osd_altitude_pos = 6391
set osd_warnings_pos = 14762
set osd_avg_cell_voltage_pos = 6528
set osd_disarmed_pos = 6635
set osd_esc_tmp_pos = 2144
set osd_core_temp_pos = 6208
set osd_stat_rtc_date_time = ON
set osd_stat_max_dist = ON
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_max_g_force = ON
set osd_stat_max_esc_temp = ON
set osd_stat_max_esc_rpm = ON
set osd_stat_min_link_quality = ON
set osd_stat_flight_dist = ON
set osd_stat_total_time = ON
set osd_stat_total_dist = ON
set osd_stat_min_rssi_dbm = ON
set debug_mode = GYRO_SCALED
set vtx_band = 5
set vtx_channel = 1
set vtx_power = 5
set vtx_low_power_disarm = ON
set vtx_freq = 5658
set vcd_video_system = PAL
set camera_control_ref_voltage = 324
set camera_control_key_delay = 250
set camera_control_internal_resistance = 300
set camera_control_button_resistance = 460,280,160,80,10

profile 2

# profile 2
set motor_output_limit = 95
set feedforward_smooth_factor = 30
set feedforward_jitter_factor = 10

rateprofile 1

# rateprofile 1
set yaw_rc_rate = 10
set roll_expo = 70
set pitch_expo = 70
set yaw_expo = 70
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70

# end the command batch
batch end
```

### Resources configuration

```txt
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: SERIAL_TX 4
A01: SERIAL_RX 4
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: LED_STRIP
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: LED 2
A14: LED 1
A15: PINIO 1
B00: FREE
B01: FREE
B02: GYRO_CS 1
B03: PINIO 2
B04: MOTOR 3
B05: MOTOR 4
B06: FREE
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: FREE
B11: FREE
B12: FLASH_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: CAMERA_CONTROL
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: SPI_MOSI 2
C04: GYRO_EXTI
C05: FREE
C06: SERIAL_TX 6
C07: SERIAL_RX 6
C08: MOTOR 1
C09: MOTOR 2
C10: FREE
C11: SERIAL_RX 3
C12: SERIAL_TX 5
C13: BEEPER
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1 : LED_STRIP
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH3 : DSHOT_BITBANG 3
    CH4 : DSHOT_BITBANG 2
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12:
    CH2 : CAMERA_CONTROL
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: LED_STRIP
DMA2 Stream 4: DSHOT_BITBANG 3
DMA2 Stream 5: SPI_MOSI 1
DMA2 Stream 6: FREE
DMA2 Stream 7: DSHOT_BITBANG 2
```

### Flight controller

Matek F722 HD and STD / Iflight Succex-E F722

klutvott123, 2022-10-28:
So it works as it should when armed and flying? 
We could probably make it work when not armed too.

RipperGOT, 2022-10-31:
@klutvott123 

Sorry, a lot of work!

I have tried it again. It doesn't matter that this is armed, it works without being armed (my mistake, sry). But the problem continues.
It only blink when it drops below 3.5V, not color changes. Before, the color started to change from green to lime around 4.0V (more or less) and when it went down to 3.5V it was purple and blinking.

klutvott123, 2022-12-02:
@RipperGOT I just tested this on the bench with a matek f722, and it seems to be working as it should. Keep in mind that the color change relies on having a working, calibrated current sensor and `bat_capacity` to be correctly specified by the user. The blinking part relies on voltage.

closed 2023-01-10 by github-actions[bot]