betaflight/betaflight #11894: BMI270 gyro tracking issue

shota3527, 2022-10-15:
### Describe the bug

The quad rotates less than the actual on flight controller with bmi270
I have confirmed the problem on speedybeef7v3, and MATEKF411WTE with inav(bmi270 driver ported from betaflight)
I suspect there might be some problems with bmi270 driver or gyro itself

### To Reproduce

`set imu_dcm_kp = 50 ` To 'disable' the accelerometer to clearly reproduce the problem 
Disarm the quad or fc only, rotate it about 5 loops by hand within 2-3 seconds and put it on the table to level it, 
Check ahi or the animation in the configurator as soon as you finish it, and see if it is level or not

### Expected behavior

The ahi or the animation in the configurator remains level when the quad is leveled after the rotation.
The internal state of the imu tracks the rotation perfectly

### Diff configuration

```txt
# diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.3.1 Jul 13 2022 / 03:36:10 (8d4f005) MSP API: 1.44
# config: manufacturer_id: SPBE, board_name: SPEEDYBEEF7V3, version: 05855be3, date: 2022-09-21T09:30:25Z

# start the command batch
batch start

board_name SPEEDYBEEF7V3
manufacturer_id SPBE

# name: ButterFly

# feature
feature -RX_PARALLEL_PWM
feature -AIRMODE
feature GPS
feature LED_STRIP

# serial
serial 0 1 115200 57600 0 115200
serial 5 2 115200 57600 0 115200

# beacon
beacon RX_LOST
beacon RX_SET

# led
led 0 7,4::AW:12
led 1 7,5::AW:12
led 2 7,6::AW:12
led 3 7,7::AW:12
led 4 7,8::AW:12
led 5 7,9::AW:12

# mode_color
mode_color 6 0 12
mode_color 6 1 7

# aux
aux 0 0 0 1800 2100 0 0
aux 1 1 1 1800 2100 0 0
aux 2 46 3 1700 2100 0 0
aux 3 13 2 1700 2100 0 0
aux 4 26 6 1700 2100 0 0
aux 5 28 1 900 1300 0 0
aux 6 35 2 1300 1700 0 0
aux 7 37 3 1300 1700 0 0

# master
set gyro_lpf2_static_hz = 0
set dyn_notch_count = 1
set dyn_notch_q = 500
set acc_calibration = -5,-2,-43,1
set align_mag = CW90FLIP
set mag_align_pitch = 1800
set mag_align_yaw = 900
set mag_calibration = -170,-656,190
set min_check = 1020
set max_check = 1980
set serialrx_provider = CRSF
set blackbox_sample_rate = 1/8
set dshot_idle_value = 600
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set failsafe_throttle_low_delay = 300
set failsafe_procedure = GPS-RESCUE
set yaw_motors_reversed = ON
set small_angle = 180
set gps_provider = UBLOX
set gps_sbas_mode = AUTO
set gps_auto_baud = ON
set gps_ublox_use_galileo = ON
set gps_set_home_point_once = ON
set gps_rescue_alt_buffer = 100
set gps_rescue_initial_alt = 100
set gps_rescue_descent_dist = 100
set gps_rescue_sanity_checks = RESCUE_SANITY_FS_ONLY
set gps_rescue_min_sats = 5
set gps_rescue_allow_arming_without_fix = ON
set gps_rescue_alt_mode = CURRENT_ALT
set deadband = 6
set yaw_deadband = 6
set osd_warn_rssi = ON
set osd_warn_link_quality = ON
set osd_warn_rssi_dbm = ON
set osd_vbat_pos = 2368
set osd_rssi_pos = 2075
set osd_link_quality_pos = 2104
set osd_tim_2_pos = 2432
set osd_flymode_pos = 2458
set osd_g_force_pos = 2341
set osd_throttle_pos = 2276
set osd_ah_pos = 2126
set osd_current_pos = 2423
set osd_mah_drawn_pos = 2393
set osd_craft_name_pos = 458
set osd_gps_speed_pos = 2244
set osd_gps_lon_pos = 49184
set osd_gps_lat_pos = 51200
set osd_gps_sats_pos = 2071
set osd_home_dir_pos = 2508
set osd_home_dist_pos = 2509
set osd_flight_dist_pos = 2464
set osd_altitude_pos = 2262
set osd_avg_cell_voltage_pos = 2400
set osd_pit_ang_pos = 228
set osd_rol_ang_pos = 196
set osd_disarmed_pos = 2347
set osd_nheading_pos = 2125
set osd_nvario_pos = 2295
set osd_efficiency_pos = 2359
set osd_displayport_device = MSP
set vcd_video_system = PAL
set displayport_msp_serial = 0
set name = ButterFly
set position_alt_gps_min_sats = 8

profile 0

# profile 0
set dterm_lpf2_static_hz = 0
set vbat_sag_compensation = 100
set iterm_relax = RPY_INC
set thrust_linear = 20
set dyn_idle_min_rpm = 20

rateprofile 0

# rateprofile 0
set roll_rc_rate = 10
set pitch_rc_rate = 10
set yaw_rc_rate = 20
set roll_expo = 50
set pitch_expo = 50
set yaw_expo = 50

# end the command batch
batch end
```

### Resources configuration

```txt
# 

# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: CAMERA_CONTROL
A01: FREE
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: LED 2
A14: LED 1
A15: MOTOR 1
B00: FREE
B01: FREE
B02: PREINIT 1
B03: MOTOR 2
B04: MOTOR 3
B05: FREE
B06: MOTOR 4
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: FREE
B11: FREE
B12: PREINIT 5
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: GYRO_EXTI
C04: FREE
C05: FREE
C06: SERIAL_TX 6
C07: SERIAL_RX 6
C08: LED_STRIP
C09: PINIO 1
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: BEEPER
C14: FREE
C15: GYRO_CS 2
D00: FREE
D01: FREE
D02: SDCARD_CS
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1 : DSHOT_BITBANG 1
    CH3 : DSHOT_BITBANG 2
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5:
    CH1 : CAMERA_CONTROL
TIM6: FREE
TIM7: FREE
TIM8:
    CH3 : LED_STRIP
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: DSHOT_BITBANG 1
DMA2 Stream 2: LED_STRIP
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: DSHOT_BITBANG 2
DMA2 Stream 7: FREE

#
```

### Flight controller

SpeedyBee F7 V3

KarateBrot, 2022-10-16:
I wonder if this is caused by aliasing in the accelerometer data. I'll have a look at the gyro driver.

ledvinap, 2022-10-16:
@KarateBrot : I don't think that ACC aliasing is causing this. Also ACC influence is small in this case because of imu_dcm_kp 

- How big is the difference?
- is it related to gyro gain ?
- - try 4 rotation in one direction vs 2 rotations forward then 2 backward

shota3527, 2022-10-17:
@KarateBrot @ledvinap 
i have uploaded a video to demonstrate the problem
https://www.youtube.com/watch?v=hnNc33QCbM8

shota3527, 2022-10-17:
> I wonder if this is caused by aliasing in the accelerometer data, giving you skewed gravity values for the IMU. I'll have a look at the gyro driver.
> 
> Edit: Just confirmed by looking at the driver settings that we indeed might have aliasing in the accelerometer data. I will make a fix for it. I don't know if this fixes your particular problem but I hope so. @shota3527 Would you be so kind and test it if it is ready?

This pr did not fix the issue

shota3527, 2022-10-18:
I am wondering if this problem is caused by Sensitivity Error or scale factor Error in the gyro itself

ledvinap, 2022-10-18:
@shota3527 : It is possible, _try 4 rotation in one direction vs 2 rotations forward then 2 backward_ will test it

shota3527, 2022-10-18:
@ledvinap 
I have did both forward and backward in this video, is it necessary to do strictly 2 forward and 2 backward ?
https://www.youtube.com/watch?v=hnNc33QCbM8

shota3527, 2022-10-18:
No, i haven`t experienced this amount of error in any other gyro.
But it is reproducible in every bmi270 me and my friends own.

With 4deg per revolution, artificial horizon drift will be quite obvious after 2-3 rolls/flips, i have not experienced in any of my previous builds before

ledvinap, 2022-10-19:
@shota3527 : Maybe try finding (any) firmware that can preform calibration and write result into NVM. After calibration, return to BT firmware

shota3527, 2022-10-21:
I did not find any firmware that can perform calibration.
I tried to write my own code to perform CRT but stuck at 7. Set G_TRIG_1.select=1 in the bmi270 datasheet page75 (have already write 0b1 to register 0x2F)

shota3527, 2022-10-21:
According to the MPU6000 datasheet or ICM-20689 datasheet of GYROSCOPE FEATURES.
It is likely that TDK/InvenSense gyro has a factory scale factor calibration
That is why we didn't experience this problem before

shota3527, 2022-10-21:
It might be impossible to apply calibrated gain according to this page
https://community.bosch-sensortec.com/t5/MEMS-sensors-forum/BMI270-remove-unnecessary-data-from-bmi270-config-file/m-p/26224

shota3527, 2022-10-21:
https://github.com/iNavFlight/inav/pull/8487
I have successfully reduced the error to about 0.5% with these codes by performing crt

haslinghuis, 2022-10-22:
@SteveCEvans

haslinghuis, 2022-10-22:
> According to the MPU6000 datasheet or ICM-20689 datasheet of GYROSCOPE FEATURES. It is likely that TDK/InvenSense gyro has a factory scale factor calibration That is why we didn't experience this problem before

@SteveCEvans

Romin0615, 2022-12-08:
Is it possible that this problem is caused by the maximum fifo driver of BMI270? Not just hardware issue. Can we switch to the normal driver(BMI270-Sensor-API/bmi270.c)?

simoneenomis, 2023-01-04:
Betaflight 4.4.0 rc3 does not detect gyro BMI270 ...

KarateBrot, 2023-01-04:
> Betaflight 4.4.0 rc3 does not detect gyro BMI270 ...

You got to be more specific. I suspect your board target doesn't have the needed defines included yet.

simoneenomis, 2023-01-04:
[image]

simoneenomis, 2023-01-04:
[image]

haslinghuis, 2023-01-04:
@simoneenomis looks ok to me:

[image]

The SPEEDYBEEF7V3 does not have a `MAG`netometer.

simoneenomis, 2023-01-04:
thanks i was sure speedybee f7v3 had it integrated thanks again bye

haslinghuis, 2023-01-04:
@KarateBrot closing now as inactive.

ctzsnooze, 2024-01-15:
I can confirm that the BMI270 has issues with gyro calibration, and that integration of the gyro data during rapid movements in level mode results in incorrect assessment of the movement of the quad, which is corrected once the quad slows down to less than 15 deg/s, causing a 'jump' in attitude.
At present Betaflight does not have a solution to write the calibration values back to these gyros.

closed 2024-01-16 by KarateBrot