betaflight/betaflight #11901: osd_aux_channel is set to 0 by default (on a fresh flash). However the valid range is set to 1-18, so a dump/diff shows ###ERROR IN dump: CORRUPTED CONFIG: osd_aux_channel = 0### .

haslinghuis, 2022-10-17:
osd_aux_channel is set to 0 by default (on a fresh flash). However the valid range is set to 1-18, so a dump/diff shows ###ERROR IN dump: CORRUPTED CONFIG: osd_aux_channel = 0### .

Is 0 the proper way to turn it "off" (if it needs to be turned off). If so, range should be 0-18.

_Originally posted by @ZzyzxTek in https://github.com/betaflight/betaflight/issues/10789#issuecomment-1279872204_

haslinghuis, 2022-10-17:
Please check  @ZzyzxTek @alexeystn

alexeystn, 2022-10-18:
Yes, with 1-18 range it returns "CORRUPTED CONFIG" message.
With 0-18 range it is ok on default config.
Thanks for fixing.

haslinghuis, 2022-10-19:
Think we have to keep the range but reset with 1 instead.

ZzyzxTek, 2022-10-19:
Looking at the code for this, it only looks at the aux channel when the OSD element is set to display.
There doesn't seem to be any notion of the 0 value "turning it off" in any way.

But when it does display the OSD element, it accesses the value using `rcData[osdConfig()->aux_channel - 1]` .
Will that cause some bad thing to happen if the aux channel is set to 0, and it's trying to index with -1?

Seems overall better to set the default to 1 and leave the valid range 1-18?

haslinghuis, 2022-10-19:
Yes PR has already been updated to reflect that.

closed 2022-10-21 by blckmn