betaflight/betaflight #11922: Arming possible while throttle is not zeroed

ThisIsJustARandomGuy, 2022-10-23:
### Describe the bug

Today I managed to arm a quad while the throttle was way above zero and it tried to flyaway immediately. I'll try to describe what I did. I'm not absolutely sure on the exact reason, but it can be reproduced, I think. I don't thjnk it's a hardware fault since Imwas able to fix it by way of recalibration of my sticks. 

Hardware:
- Mamba F4 and some Mamba ESC
-  Happymodel EP2 ELRS RX
- Jumper T-Lite with Happymodel TX module

Software:
- Betaflight 4.3
- ELRS RX and TX 3.0
- EdgeTX latest

### To Reproduce

Somehow, I completely lost stick calibration on my TX (after swapping SD cards). Meaning, moving a stick wouldn't change any channel value. I did not notice this at the beginning. The sticks, according to the OpenTX display, were at dead center.

However, when arming without having the sticks calibrated, the quad immediately went batsh*t crazy. Funnily enough, it wasn't a complete flyaway - it just tried to haul ass and then decided to roll to a randdom side. The props were on right and tight. It should *not* have armed. Definitely. According to EdgeTX, it armed at center throttle.

If someone could give me a way to delete my stick calibration on EdgeTX,  I will provide blackbox and video.

### Expected behavior

Betaflight MUST NOT EVER ARM if the throttle is above the min throttle.

### Diff configuration

```txt
Defaults, exept for the modes. After running aa stick calibration on the TX, it's working fine.
```

### Resources configuration

```txt
Defaults
```

### Flight controller

H7 chipset, F4 ESC

### Other components

Jumper T-Lite

### How are the different components wired up

Quad is working perfectly fine when the TX has correctly calibrated sticks

### Add any other context about the problem that you think might be relevant here

Need instructions on how to reset the stick calibration to 0-0 on EdgeTX to reproduce, please.

ThisIsJustARandomGuy, 2022-10-25:
Okay, so apparently, these are valid values to arm the quad. Of course it tries to do whatever it wants, immediately on arming:
[image]

This is a test I did on the bench, and you can see motor 1 immediately goes to 100%. When props are attached, no flyaway or other failsafe kicks in until manually disarming.

[image]

Is there any protection against this available? Say, only allow arming if the throttle is at min (as it is now) and, additionally, the rest of the sticks are centered?

To reliably reproduce the stick positions, I do the following: Calibrate the sticks on an EdgeTX 2.7.1 Jumper T-Pro. Then transfer that SD card to a Jumper T-Lite. It produces the exact same stick output as I've shown.

ctzsnooze, 2022-11-01:
So you want arming prevented unless roll, pitch and yaw are ‘close to center’ *as well as having throttle at zero*?

ThisIsJustARandomGuy, 2022-11-03:
Exactly, or at least having it as a cli option would be great. Maybe a "maximum stick offset for arming" setting could also work and be quite similar to existing threshold settings. I would put it at 1000 by default for now. This would not change the current behavior (i.e. allow arming at full stick deflection), but it would give users the option to choose and even configure to their liking.

 Now that I think of it, it could probably already be done in the TX. Anyways, I'd always rather the quad not moving at all or showing an OSD warning instead of arming to immediate 100% motor output. As it works now, a broken gimbal can lead to quite bad and, at least for me, very unexpected results.

Are there valid cases where one might want to actually arm with the sticks not centered? I can't think of any, but I only fly quads so I may be missing something.

closed 2022-12-11 by github-actions[bot]