betaflight/betaflight #11923: Kurzbefehle Einrichtung

Tom18710, 2022-10-23:
### Is your feature request related to a problem? Please describe

Hab kürzlich eine Drohne mit bestimmten Kurzbefehlen für den Controller bekommen. 
Bin Anfänger und konnte diese leider nirgends finden. Ich bitte um Aufklärung:
-Horizon 
-Blackbox 

Hab eine TX16s und bisher eingestellt: 
-Arm
-Beeper
-Angle
diese sind mir bewusst was die Bedeutung angeht. Noch kurz Disarming findet immer statt, wenn man den Arming Befehl ausstellt und muss nicht extra eine ,,Disarming“ Taste einfügen oder?

Danne für die Hilfe im Voraus!

### Describe the solution you'd like

Aufklärung Kurzbefehle 

### Describe alternatives you've considered

…

sugaarK, 2022-10-23:
Ahhhh the only common language we speak in the project is English.

KarateBrot, 2022-10-23:
[Betaflight 4.3 Complete Walkthrough](https://www.youtube.com/playlist?list=PLwoDb7WF6c8nT4jjsE4VENEmwu9x8zDiE)

haslinghuis, 2022-10-23:
Github is not the place for configuration issues but for reporting bugs. It's mandatory to follow the walkthrough before considering getting help at Discord.

closed 2022-10-23 by haslinghuis