betaflight/betaflight #11948: FUNCTION_MSP_DISPLAYPORT breaks VTX Device over MSP

zandorsp, 2022-11-01:
### Describe the bug

I have a drone with HDzero VTX with VTX over MSP for power and channel control.

In betaflight nightly 31-out-2022 to show the OSD need to enable FUNCTION_MSP_DISPLAYPORT  #11913 , but it effectly disable VTX over MSP control #11705 

### To Reproduce

Use a OpenHD vtx with a VTX over MSP control enabled [https://github.com/hd-zero/hdzero-vtx/actions/runs/3349734108](url)

Try to enable OSD elements with VTXmsp at the same time.

### Expected behavior

OSD elements and VTXmsp working at the same time. 

### Diff configuration

```txt
Entering CLI Mode, type 'exit' to return, or 'help'

# 
# Building AutoComplete Cache ... Done!
# 
# diff

# version
# Betaflight / STM32F411SX1280 (S4SX) 4.4.0 Oct 31 2022 / 21:45:21 (25ff11508) MSP API: 1.45
# config: manufacturer_id: BEFH, board_name: BETAFPVF4SX1280, version: 9d0a4b63, date: 2022-10-12T09:07:03Z

# start the command batch
batch start

board_name BETAFPVF4SX1280
manufacturer_id BEFH

# name: CRUX3 HD

# serial
serial 0 131073 115200 57600 0 115200

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 1 1700 2100 0 0
aux 2 2 1 1300 1700 0 0
aux 3 13 3 1700 2100 0 0
aux 4 35 2 1700 2100 0 0

# adjrange
adjrange 0 0 4 900 2100 29 4 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY    0    0    0    0    0    0    0    0
vtxtable band 2 BOSCAM_B B FACTORY    0    0    0    0    0    0    0    0
vtxtable band 3 BOSCAM_E E FACTORY    0    0    0    0    0    0    0    0
vtxtable band 4 FATSHARK F FACTORY    0 5760    0 5800    0    0    0    0
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I FACTORY    0    0    0    0    0    0    0    0
vtxtable powerlevels 3
vtxtable powervalues 14 23 0
vtxtable powerlabels 25 200 0

# vtx
vtx 0 5 0 0 3 900 1300
vtx 1 5 0 0 1 1300 1700
vtx 2 5 0 0 2 1700 2100

# master
set acc_calibration = 11,53,38,1
set rc_smoothing_auto_factor = 120
set motor_pwm_protocol = DSHOT300
set motor_poles = 12
set report_cell_voltage = ON
set osd_link_quality_pos = 2137
set osd_link_tx_power_pos = 2080
set osd_rssi_dbm_pos = 2105
set osd_throttle_pos = 2073
set osd_vtx_channel_pos = 2048
set osd_craft_name_pos = 2528
set osd_pilot_name_pos = 2496
set osd_avg_cell_voltage_pos = 2552
set osd_ready_mode_pos = 14635
set osd_flip_arrow_pos = 14573
set osd_displayport_device = MSP
set vtx_band = 5
set vtx_channel = 8
set vtx_power = 1
set vtx_freq = 5917
set expresslrs_uid = 226,139,188,184,193,240
set expresslrs_model_id = 2
set craft_name = CRUX3 HD
set pilot_name = ZANDOR

profile 0

# profile 0
set feedforward_averaging = 2_POINT
set feedforward_smooth_factor = 65

rateprofile 0

# end the command batch
batch end
```

### Resources configuration

```txt
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: MOTOR 2
A01: ADC_BATT
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: RX_SPI_EXPRESSLRS_BUSY
A14: BEEPER
A15: RX_SPI_CS
B00: ADC_CURR
B01: LED_STRIP
B02: RX_SPI_BIND
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: GYRO_EXTI
B07: MOTOR 4
B08: MOTOR 1
B09: RX_SPI_EXPRESSLRS_RESET
B10: MOTOR 3
B11: FREE
B12: PREINIT 3
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: RX_SPI_EXTI
C14: LED 1
C15: LED
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH1 : MOTOR 2
    CH3 : MOTOR 3
TIM3:
    CH4 : LED_STRIP
TIM4:
    CH2 : MOTOR 4
    CH3 : MOTOR 1
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: LED_STRIP
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: TIMUP 4
DMA1 Stream 7: TIMUP 2
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Flight controller

BetaFPV F4 1S 12A AIO ELRS 2.4G V2.0

### Other components

HDzero whoop vtx

haslinghuis, 2022-11-01:
Does it work with` serial 0 131072 115200 57600 0 115200 ` ?

zandorsp, 2022-11-01:
> Does it work with`serial 0 131072 115200 57600 0 115200` ?

Doesn't seen to accept that command, on cli:
serial 0 131072 115200 57600 0 115200
save

# serial
serial 20 1 115200 57600 0 115200
serial 0 0 115200 57600 0 115200
serial 1 0 115200 57600 0 115200
serial 30 0 115200 57600 0 115200

haslinghuis, 2022-11-01:
Ignore previous suggestion. You need additional configuration:
```
set displayport_msp_serial = 0
```

haslinghuis, 2022-11-01:
@SteveCEvans you might have hardware to test this. Please check.

SteveCEvans, 2022-11-01:
I think I've missed something here then. Looks like the HDZero VTX use MSP VTX commands AND DisplayPort.

I coded this expecting them to be mutually exclusive. Sounds like I need to support enabling both, concurrently, which, being a bit mask, should be doable.

SteveCEvans, 2022-11-01:
@zandorsp Please test https://github.com/betaflight/betaflight/pull/11950

Use `serial 0 393217 115200 57600 0 115200` to enable both `FUNCTION_VTX_MSP` and `FUNCTION_MSP_DISPLAYPORT`.

zandorsp, 2022-11-02:
> 

It's worked very well. OSD ok and VTX control ok too.
#11950

SteveCEvans, 2022-11-02:
Thanks for testing.

@haslinghuis how should we handle this in the configurator? Embedding knowledge of which options each VTX needs will be easiest for the end user.

haslinghuis, 2022-11-02:
The user should select only protocol - functionNames in configurator are defined in firmware.

PROTOCOL | MSP | MSP_VTX | MSP_DISPLAYPORT
| --- | :---: | :---: | :---: |
| DJI | X |
| HDZero | X | X | X |
| OpenVTX | X | X |  |
| WalkSnail | X | | X |

SteveCEvans, 2022-11-02:
So we could use the presets system to configure for a particular VTX. And just expose the options above in the configurator. How should we select them in combination though?

sugaarK, 2022-11-02:
https://github.com/betaflight/firmware-presets/pull/297

haslinghuis, 2022-11-02:
@sugaarK the preset is not valid anymore after latest changes. Using a preset will not solve the ports tab configuration as saving would reset the function.
```
# set displayport_msp_serial = 0
###ERROR IN set: INVALID NAME###
```

sugaarK, 2022-11-02:
Well we need to fix things and rework the preset then

closed 2022-11-04 by blckmn