betaflight/betaflight #11949: Home Point on OSD Screen

druckgott, 2022-11-01:
### Is your feature request related to a problem? Please describe

I fly often in the mountains and sometimes I have the problem that I lost the orientation or its quide hard to see where I started because I use a area where are many trees or something like this. The arrow on the screen helps quite a lot to find back, but sometimes its hard to realy see where you have to go down for landing where you did start.

### Describe the solution you'd like

My Idea is to have a H-Point moving on the screen like the dji drone to realy see where you did start projected on the screen.

### Describe alternatives you've considered

Landing normaly, which is sometimes quite hard to realy find the starting spot

druckgott, 2022-11-12:
Today I have a situation in fog and therefore I have a second point to have such a future ;-)

waldviertler, 2022-11-28:
I would also like such a feature!

qvasic, 2022-12-23:
The idea is good, but I am quite dubious about the implementation. While the feature itself wouldn't be terribly hard to implement, I see problems with its configuration.
Every camera has different field of view and projection characteristics on the sensor (is it more fish-eye? is it more rectangular?). And also the uptilt.
DJI are in advantage here - each of their drones fly with exactly one camera.

druckgott, 2022-12-23:
Yes I also think on this. I did not find out how to calculate this homepoinr to do a test with excel. Or so and use a picture where I know the distance to find out what values are needed to config the Kamera. Maybe there could be any way to config Kamera vales automatic way with a paper with points on it I a special distance for example 1m oder 10cm ore anything linke this? 

I do not

qvasic, 2022-12-23:
Every single camera will require calibration for this feature to work: we need to know where does a point, located at a given angle deviation from lens axis, lands where on the sensor.

druckgott, 2022-12-23:
Yes I understand. But that's what I mean if it's maybe posible to do a calibration progress but I do not know.

druckgott, 2022-12-23:
I think they did something like this if you take a look on the video in the last post
https://github.com/iNavFlight/inav/pull/5906

qvasic, 2022-12-24:
Adding azimuth would be way easier, since it doesn't require placing home point on the video feed = no camera calibration required. It would also could greatly simplify navigation home in difficult conditions.

BUT both of these features would require compass to be present on the quad. Without it accuracy will be greatly compromised by any side drift - be it slight roll or side wind.

qvasic, 2022-12-24:
As for camera calibration... it would be possible to implement simplified DIY calibration, if Betaflight had access to video feed and could send it to the Configurator... I'll inquire about this.

druckgott, 2022-12-24:
But on vis video it's also a moving home point is visible, so we also can have a look ok this.

druckgott, 2022-12-24:
Here is some information for camera config and homepoint
https://github.com/iNavFlight/inav/wiki/OSD-Hud-and-ESP32-radars
Maybe we can have a look there what's needed.

druckgott, 2022-12-24:
Maybe it´s posible to take some code von Inav?:
https://github.com/iNavFlight/inav/blob/8d70c7a186784fa34c4a421149420cbda9ad5537/src/main/io/osd_hud.c#L123

druckgott, 2022-12-27:
I did a first conzept for it maybe somebody can have a look on it.
https://github.com/betaflight/betaflight/pull/12102

danarrib, 2023-01-06:
> BUT both of these features would require compass to be present on the quad.

Not exactly. To calculate Azimuth, you just need two coordinates: The Home Point Lat/Long and the Aircraft Lat/Lon.

[image]

https://gis.stackexchange.com/questions/108171/calculating-azimuth-from-two-points-both-having-latitude-longtiude

There's also the INAV implementation of that:

https://github.com/iNavFlight/inav/pull/5906

qvasic, 2023-01-06:
> > BUT both of these features would require compass to be present on the quad.
> Not exactly. To calculate Azimuth, you just need two coordinates: The Home Point Lat/Lon and the Aircraft Lat/Lon.
Okay, this statement is true, for only azimuth compass is not needed.

But where we're going to get "aircraft current heading" (as stated on the picture), if there is not compass?

danarrib, 2023-01-06:
On a compass-less aircraft, the only way to know the direction it's pointing to is when it's moving. On quadcopters this is particularly trickier because quadcopters can fly "looking" to any direction (different from airplanes, that always fly looking forward).

But, for navigation purposes, the Azimuth element could be very helpful even on a compass-less quad. If you are flying towards some point of interest (like a mountain, a cave, etc), and you KNOW that this POI is at Azimuth 155º, then during the flight, you can keep watching the azimuth number and if it gets bellow 155, you know you are to the left of the intended POI, and if the number gets above 155, you're on the right, so you can correct your course.

On INAV, the main purpose of this element is to make it easy to point a high gain directional antena to the aircraft without using an automatic antenna tracker. You can set the antenna pointing to an azimuth, and then keep the aircraft flying on this exact azimuth, and you'll know that the aircraft is pointing to the antenna.

qvasic, 2023-01-06:
> On a compass-less aircraft, the only way to know the direction it's pointing to is when it's moving. On quadcopters this is particularly trickier because quadcopters can fly "looking" to any direction (different from airplanes, that always fly looking forward).

Also wind can move the aircraft sideways.

qvasic, 2023-01-06:
Also, the illustration from above has a technical mistake. Azimuth must be calculate from aircraft to home point, not the other way around. This might surprise you, but azimuth from A to B is not equal to azimuth from B to A plus (or minus) 180 degrees.

druckgott, 2023-01-12:
Maybe if anybody want also to test it here ist a first code, I already tested it without compass which was not so exact on yaw
If somebody have a compass and want to test it
https://github.com/betaflight/betaflight/pull/12102#issuecomment-1379458406

It's needed to switch it on in cli to any position and the if gps lock is there and you start you have a moving home point

waldviertler, 2023-01-12:
I would test it, but I don't know where to start....

druckgott, 2023-01-12:
I have fixed some points, we have to take a look when new files are generated. If this is availible you can download the files and flash it nromaly in the configuration the only diff is that you select a local file instead of a online file. Or you can tell me which fc you have then I can compile a file.

druckgott, 2023-01-17:
I now did a test, but I have problems with my gps values, I think if I come loser to the point the horzontal positions is getting better, but vertical is total off because of the wrong altitude value from the gps:
https://youtu.be/D3IC9RrAi94
I opend a Issuse for this:
https://github.com/betaflight/betaflight/issues/12207

Would be cool if somebody else can test it if there is no drift on the persons copter.

AirCombat, 2025-04-15:
I would just like to comment I don't think any complex calibration would be needed. Horizon and FOV as settings for the user would be enough. I have used the Avata, Avata 2 and Neo a lot and DJI's implementation is not even slightly accurate so the bar here shouldn't be set too high

druckgott, 2025-04-16:
Maybe we can Update the pr
https://github.com/betaflight/betaflight/pull/12102

closed 2026-09-26 by blckmn