betaflight/betaflight #11976: G4 ADC2 trouble

David-OConnor, 2022-11-10:
### Describe the bug

I'm having trouble getting a G473 flight controller to read the battery. Specifically, after running `resource show all` in the Configurator CLI, both the pin and DMA channel are showing free. When I use a dummy ADC1 setup, the DMA channel can be selected correctly.

ADC2, CH17, pin PA4. I've verified it's not hardware with a successful test in custom firmware.

### To Reproduce

Relevant config:
```
resource ADC_BATT 1 A04 # ADC2 In 17  # If I use `ADC_BATT 2 A04`, error: ###ERROR IN resource: INDEX NOT BETWEEN 1 AND 1###`
set adc_device = 2

set current_meter = ADC
dma ADC 2 7  # ADC 2 for voltage and current measurement
```

Resources from `resource show all`:
```
A04: FREE
...
Currently active DMA:
--------------------
DMA1 Channel 1: DSHOT_BITBANG 1
DMA1 Channel 2: DSHOT_BITBANG 2
DMA1 Channel 3: SPI_MOSI 1
DMA1 Channel 4: SPI_MISO 1
DMA1 Channel 5: SPI_MOSI 2
DMA1 Channel 6: SPI_MISO 2
DMA1 Channel 7: FREE
// etc FREE
```

### Expected behavior

Resources show the pin and ADC, and a battery voltage is read.

Of note, perhaps `resource ADC_BATT 1 A04` needs to be set to `resource ADC_BATT 2 A04`, but this results in the error described above. (INDEX NOT BETWEEN 1 AND 1)

### Diff configuration

```txt
batch start

board_name MERCURYG4
manufacturer_id ANYL

# resources
resource BEEPER 1 A08
resource MOTOR 1 A00
resource MOTOR 2 A01
resource MOTOR 3 B00
resource MOTOR 4 B01
resource SERIAL_TX 1 B06
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 C10
resource SERIAL_RX 1 B07
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 C11
resource I2C_SCL 1 A15
resource I2C_SCL 2 A09
resource I2C_SDA 1 B09
resource I2C_SDA 2 A10
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 B03
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 B04
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource SPI_MOSI 3 B05
resource ADC_BATT 1 A04
resource ADC_CURR 1 B02
resource FLASH_CS 1 C06
resource RX_SPI_CS 1 C15
resource RX_SPI_EXTI 1 C13
resource RX_SPI_EXPRESSLRS_BUSY 1 C14
resource GYRO_EXTI 1 C04
resource GYRO_CS 1 B12

# timer
timer A00 AF1
# pin A00: TIM2 CH1 (AF1)
timer A01 AF1
# pin A01: TIM2 CH2 (AF1)
timer B00 AF2
# pin B00: TIM3 CH3 (AF2)
timer B01 AF2
# pin B01: TIM3 CH4 (AF2)
timer A08 AF6
# pin A08: TIM1 CH1 (AF6)

# dma
dma ADC 1 8
# ADC 1: DMA2 Channel 1 Request 5
dma ADC 2 8
# ADC 2: DMA2 Channel 1 Request 36
dma TIMUP 2 5
# TIMUP 2: DMA1 Channel 6 Request 60
dma TIMUP 3 6
# TIMUP 3: DMA1 Channel 7 Request 65
dma pin A00 1
# pin A00: DMA1 Channel 2 Request 56
dma pin A01 4
# pin A01: DMA1 Channel 5 Request 57
dma pin B00 3
# pin B00: DMA1 Channel 4 Request 63

# feature
feature -RX_SPI
feature RX_SERIAL
feature TELEMETRY
feature OSD

# serial
serial 1 1 115200 57600 0 115200

# master
set mag_hardware = NONE
set baro_bustype = I2C
set baro_i2c_device = 2
set rssi_channel = 15
set serialrx_provider = CRSF
set rx_spi_bus = 3
set adc_device = 2
set blackbox_device = SPIFLASH
set motor_pwm_protocol = DSHOT300
set current_meter = ADC
set battery_meter = ADC
set vbat_scale = 11
set pid_process_denom = 4
set system_hse_mhz = 16
set flash_spi_bus = 2
set gyro_1_bustype = SPI
set gyro_1_spibus = 1
set gyro_1_sensor_align = CW90
set gyro_1_align_yaw = 900

profile 0

rateprofile 0

# end the command batch
batch end
```

### Resources configuration

```txt
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: MOTOR 1
A01: MOTOR 2
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: BEEPER
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: I2C_SCL 1
B00: MOTOR 3
B01: MOTOR 4
B02: FREE
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: FREE
B07: FREE
B08: FREE
B09: I2C_SDA 1
B10: FREE
B11: FREE
B12: GYRO_CS 1
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: PREINIT 2
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: FREE
C14: FREE
C15: PREINIT 3
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1 : DSHOT_BITBANG 1
    CH2 : DSHOT_BITBANG 2
TIM15: FREE
TIM16: FREE
TIM17: FREE
TIM20: FREE

Currently active DMA:
--------------------
DMA1 Channel 1: DSHOT_BITBANG 1
DMA1 Channel 2: DSHOT_BITBANG 2
DMA1 Channel 3: SPI_MOSI 1
DMA1 Channel 4: SPI_MISO 1
DMA1 Channel 5: SPI_MOSI 2
DMA1 Channel 6: SPI_MISO 2
DMA1 Channel 7: FREE
DMA1 Channel 8: FREE
DMA2 Channel 1: FREE
DMA2 Channel 2: FREE
DMA2 Channel 3: FREE
DMA2 Channel 4: FREE
DMA2 Channel 5: FREE
DMA2 Channel 6: FREE
DMA2 Channel 7: FREE
DMA2 Channel 8: FREE
```

### Flight controller

WIP/custom

### How are the different components wired up

11:1 divider on PA4. (ADC2, CH17)

KarateBrot, 2022-11-10:
I don't know if it's related but check out https://github.com/betaflight/betaflight/pull/11973

David-OConnor, 2022-11-10:
I'm guessing that PR is not directly related, but it might explain some other issues I've been having!

David-OConnor, 2022-11-12:
Solved - needed to assign an ADC 1 DMA resource as well, and leave the values for the `ADC_BATT` and `ADC_CURR` with the `1` following them:

```rust
resource ADC_BATT 1 A04 # ADC2 In 17
resource ADC_CURR 1 B02 # ADC2 In 12
set adc_device = 2

# ADC 2 for voltage and current measurement.
dma ADC 2 8

# We don't use ADC 1, but BF won't let us configure ADC 2 without it.
dma ADC 1 9 
```

Is this the intended behavior?

SteveCEvans, 2022-11-12:
It is rather counter intuitive for sure, but the `ADC_TEMPSENSOR` and `ADC_VREFINT` channels are on ADC 1. So whilst you may not be declaring external pins for them, these internal sources still require a DMA channel to be assigned for ADC 1.

See https://github.com/betaflight/betaflight/blob/931c982a3bdd67714911cad295bd6d0f83446997/src/main/drivers/adc_stm32g4xx.c#L288

If no channel is assigned then the initialisation code bails out at the following line before processing ADC 2.
https://github.com/betaflight/betaflight/blob/931c982a3bdd67714911cad295bd6d0f83446997/src/main/drivers/adc_stm32g4xx.c#L415

David-OConnor, 2022-11-12:
Thanks for the explanation! Closing this. Hopefully it will come up in a search if someone else runs into this later.

closed 2022-11-12 by David-OConnor